Koenkk/zigbee2mqtt #18456: [New device support]: id lock 202 (datek 150 zigbee module)

polyvert, 2023-07-29:
### Link

https://idlock.no/id-lock/

### Database entry

{"id":12,"type":"EndDevice","ieeeAddr":"0x680ae2fffe6bb0eb","nwkAddr":37470,"manufId":4919,"manufName":"Datek","powerSource":"Battery","modelId":"ID Lock 202","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":10,"inClusterList":[0,1,3,257],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"ID Lock 202","manufacturerName":"Datek","powerSource":3,"zclVersion":3,"appVersion":7,"stackVersion":65,"swBuildId":"0.7"}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":7,"stackVersion":65,"swBuildId":"0.7","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690643604707,"defaultSendRequestWhen":"immediate"}

### Comments

The new id lock 202 uses the same zigbee module as previous lock (150). Since it is the same module, would it be possible to add the 202 lock, with the same functionality as the already included 150 lock?

polyvert, 2023-07-30:
I tried to make an external converter file based on the information posted regarding id lock 150 here:
https://github.com/Koenkk/zigbee2mqtt/issues/6976
and combined it with included code for the 150 lock from here:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/datek.ts#L119

--------------------------------------------
`
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const constants = require('zigbee-herdsman-converters/lib/constants');
const ota = require('zigbee-herdsman-converters/lib/ota');
const e = exposes.presets;
const ea = exposes.access;

const fzLocal = {
    lock: {
        cluster: 'closuresDoorLock',
        type: ['attributeReport', 'readResponse'],
        convert: (model, msg, publish, options, meta) => {
            const result = {};
            if (msg.data.hasOwnProperty('lockState')) {
                result.state = msg.data.lockState == 1 ? 'LOCK' : 'UNLOCK';
                const lookup = ['not_fully_locked', 'locked', 'unlocked'];
                result.lock_state = lookup[msg.data['lockState']];
            }

            if (msg.data.hasOwnProperty('soundVolume')) {
                result.sound_volume = constants.lockSoundVolume[msg.data.soundVolume];
            }

            if (msg.data.hasOwnProperty('doorState')) {
                result.door_state = msg.data.doorState;
            }

            if (msg.data.hasOwnProperty('lockMode')) {
                result.lock_mode = msg.data.lockMode;
            }
            return result;
        },
    },
}

const device = {
    zigbeeModel: ['ID Lock 202'],
    model: '0402946',
    vendor: 'Datek',
    description: 'Zigbee module for ID Lock 202',
	fromZigbee: [fz.lock, fz.battery, fz.lock_operation_event, fz.lock_programming_event,fz.idlock, fz.idlock_fw, fz.lock_pin_code_response],
	toZigbee: [tz.lock, tz.lock_sound_volume, tz.idlock_master_pin_mode, tz.idlock_rfid_enable,tz.idlock_service_mode, tz.idlock_lock_mode, tz.idlock_relock_enabled, tz.pincode_lock],
	meta: {pinCodeCount: 109},

	configure: async (device, coordinatorEndpoint, logger) => {
		const endpoint = device.getEndpoint(1);
		const options = {manufacturerCode: 4919};
		await reporting.bind(endpoint, coordinatorEndpoint, ['closuresDoorLock', 'genPowerCfg']);
		await reporting.lockState(endpoint);
		await reporting.batteryPercentageRemaining(endpoint);
		const payload = [{
			attribute: {ID: 0x4000, type: 0x10},
			minimumReportInterval: 0,
			maximumReportInterval: constants.repInterval.HOUR,
			reportableChange: 1,
		},
		{
			attribute: {ID: 0x4001, type: 0x10},
			minimumReportInterval: 0,
			maximumReportInterval: constants.repInterval.HOUR,
			reportableChange: 1,
[... 45 lines cut ...]
		e.enum('lock_mode', ea.ALL, ['auto_off_away_off', 'auto_on_away_off', 'auto_off_away_on',
			'auto_on_away_on']).withDescription('Lock-Mode of the Lock'),
		e.enum('service_mode', ea.ALL, ['deactivated', 'random_pin_1x_use',
			'random_pin_24_hours']).withDescription('Service Mode of the Lock')],	
};

module.exports = device;
`

--------------------------------------------

 
 
I have not been able to test everything, but the "pin_code" refresh gives an error.

--------------------------------------------

`
Publish 'get' 'pin_code' to 'Door' failed: 'Error: Command 0x680ae2fffe6bb0eb/1 closuresDoorLock.getPinCode({"userid":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37470 - 1 - 202 - 257 - 6 after 10000ms)'
Debug 2023-07-30 23:22:30Error: Command 0x680ae2fffe6bb0eb/1 closuresDoorLock.getPinCode({"userid":0}, {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Timeout - 37470 - 1 - 202 - 257 - 6 after 10000ms) at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:64:35) at listOnTimeout (node:internal/timers:569:17) at processTimers (node:internal/timers:512:7)
`

--------------------------------------------

[image]

closed 2023-09-06 by github-actions[bot]

cp1983, 2023-09-29:
Can this be opened again?

Would realy appeciate if anyone could the support for idlock-202, as said before it should be the same as the idlock 150.

terjefl, 2023-10-30:
Same here. I have a ID Lock 202 as well, and would love to get it to work in Home Assistant via Z2M.

christian1986, 2024-01-04:
Ideas on why doorstate is not exposed?

[image]

TheStigh, 2024-01-08:
> Ideas on why doorstate is not exposed?
> 
> ![3ecdee02-5ab5-4b55-b2f7-b1309b5f8903](https://private-user-images.githubusercontent.com/4373980/294163743-11de97ec-7a8d-46c3-b5a2-b34f1dea1921.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ3MTA2MjEsIm5iZiI6MTcwNDcxMDMyMSwicGF0aCI6Ii80MzczOTgwLzI5NDE2Mzc0My0xMWRlOTdlYy03YThkLTQ2YzMtYjVhMi1iMzRmMWRlYTE5MjEuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDEwOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAxMDhUMTAzODQxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ODkxYjRjODJmZGE2YzZkMjBjZjQ2NjA4ZjQwYWU3MWIyNTdmNzAwZWFiZWU2MGM3OTUxMDNlNzJjMjExNWU2NSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.AA-qUI7jnqrgZG1RrMk6AnpGXS2JfpD65aLpizCgFjc)

Hi Christian - you also asked the same question for my integration of the 150 model. Do you have a 150 or 202?

TheStigh, 2024-01-08:
@cp1983 @terjefl 

If any of you are available for doing the testing, I can quickly create a test converter for you to test with. Extremely busy this month but won't take me so much time of either of you can do the testing itself.

terjefl, 2024-01-08:
Hi @TheStigh. 
My ID Lock 202 started working some time ago. I'm not sure what it coincides with, but probably a Z2M update sometime in laste November or early December. I had it paired and just sitting there as a unsupported device and one day I saw it was working. I think everything works. I'm not missing any functionality. That I can think of at least.
[image]

cp1983, 2024-01-08:
Hi

I have the IDLock 202 with Zigbee. However I created my own id202.js file based on the 150 code and just renamed the model to ID Lock 202 to make it work. Door state is not exposed, however I can see in the log file that the door is sending it when opening/closing the door. Lock state is updating fine.

I have been monitoring the Zigbee release notes to see if IDlock 202 support was added however have not noticed anything.

If you have any new code I can test it, however I will not be able to access the lock to see if it reacts on open/close until beginning of February.

Christian