Koenkk/zigbee2mqtt #18457: When group has `optimistic: false`, other groups are not "notified" of change

SHxKM, 2023-07-29:
### What happened?

After the discussion [here](https://github.com/Koenkk/zigbee2mqtt/discussions/10719) and the clarifications by @Koenkk on what `optimistic: false` actually means, I believe what I'm experiencing is a bug.

If you have `big_group` with `light_a` and `light_b` as members and `optimistic: false`, and `small_group` with `light_a` as member, and you turn off `big_group`, `small_group` keeps reporting `ON`.

### What did you expect to happen?

That `small_group`, having the default `optimistic: true` setting, should change state according to its members' state.

### How to reproduce it (minimal and precise)

1. Create two groups, one, `big_group` with two lights, and one, `small_group` with a single light that is a member of `big_group`.
2. Set `big_group`'s `optimistic` flag to `false`.
3. Turn on `small_group`
4. Turn on `big_group`
5. Turn off `big_group`

I would've expected `small_group`'s state to change to `OFF` as well, but it remains `ON`.

Auxiliary YAML:

```yaml
"1000": # large ID numbers to avoid collision
  friendly_name: big_group
  optimistic: false
  devices:
    - light_1 # mutual member
    - light_2
"1001":
  friendly_name: small_group
  devices:
    - light_1 # mutual member
```

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

0x26580700

### Adapter

ConBee2/RaspBee2

Koenkk, 2023-08-05:
When both groups and all lights are `ON` and sending `OFF` to `big_group`, do `light_1` and `light_2` change to state `OFF`? (can you provide the debug of this)

See https://www.zigbee2mqtt.io/guide/usage/debug.html on how to enable debug logging.

RaNo99, 2023-08-05:
Hi @Koenkk , as I'm facing the same issue, let me answer that question. Yes, all involved lights are turned off. The problem is with the group (the small group) that it remains on (although all its lights get off).

SHxKM, 2023-08-05:
> Hi @Koenkk , as I'm facing the same issue, let me answer that question. Yes, all involved lights are turned off. The problem is with the group (the small group) that it remains on (although all its lights get off).

Yes. This is also my experience.

Koenkk, 2023-08-06:
Is the Z2M `state` also `OFF`? Please provide the debug log so I can verify this.

closed 2023-09-13 by github-actions[bot]