Koenkk/zigbee2mqtt #18458: [Wrong device]:

manosioa, 2023-07-29:
### Link

https://www.zigbee2mqtt.io/devices/ZY-M100-S.html

### Model

TuYa ZY-M100-S

### Description

Mini Human Breathe Sensor

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/ZY-M100-S.jpg

### Database entry

{"id":2,"type":"Router","ieeeAddr":"0x70ac08fffe212b6c","nwkAddr":25538,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\b?X,i*@X,i,@X,i-@X,i.@X,i/@X,i0@X,i1@X,i","65506":31,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690655236476,"defaultSendRequestWhen":"immediate"}

### Notes

The device is detected as the TuYa ZY-M100-L which is the ceiling-mounted variation of the sensor.
The actual device is the TuYa ZY-M100-S which is a wall-mounted variation.

I don't know if that changes the way the device works. I'm mentioning this as I have issues with false negatives as well as false positives, after trying every possible placement in the rooms. (I have 2 of them).
I see that many people are very pleased with the performance of this device so I guess that this may have to do with the different model detected by the Z2M

b2un0, 2023-07-31:
Nope.
I have three of them, one wall mounted, two ceiling-mounted.
all of them are detected as ZY-M100-S and all of them are working correctly.

bsaurusrex, 2023-08-23:
I have the same problem.   all 10 devices are ceiling mount m100-L models but 50% are detecting as m100-s_1.  Additionally.  These devices with the wrong model tagging are **extremely** chatty.  1 message/second and they are causing my coordinator to crash while the correctly identified only sends a few message/minute.

**Mis-identified**

{"id":15,"type":"Router","ieeeAddr":"0xa4c138d1e44fe112","nwkAddr":55406,"manufId":4417,"manufName":"_TZE204_sxm7l9xa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":" \u0012P,i","65506":56,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_sxm7l9xa","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690101700545,"defaultSendRequestWhen":"immediate"}

**Correctly identified**

{"id":23,"type":"Router","ieeeAddr":"0x385b44fffe054a07","nwkAddr":8511,"manufId":4098,"manufName":"_TZE204_ztc6ggyl","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"ZeV,iZeV,i[eV,i[eV,i[eV,i\\eV,i\\eV,i]eV,i","65506":31,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE204_ztc6ggyl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690521100571,"defaultSendRequestWhen":"immediate"}

adiov, 2023-09-25:
@bsaurusrex I have the same issue as you. It's not an identification issue. The devices themselves are different, and I suspect they're different hardware revisions inside the same plastic housing. The sellers on AliExpress and other sites are mixing them under the same listings.

closed 2023-11-04 by github-actions[bot]