Koenkk/zigbee2mqtt #18464: Enocean Zigbee Switches PTM 215Z stop working when permit join = false

TjardE, 2023-07-30:
### What happened?

In my zigbee network there are also 8 PTM 215Z switches (greenpower).
They work correctly when Permit join = true in settings of the GUI or when I press the Permit Join in the GUI.
When permit join = false 2 of the 8 stop working. They are also the latest added devices in the zigbee network.

### What did you expect to happen?

I expected them to stay working with permit join = false

### How to reproduce it (minimal and precise)

Untag Permit join in gui/settings and press submit
Use the switches

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20221226

### Adapter

Zigbee 3.0 USB Dongle Plus ZigBee 3.x.0 TI CC2652P coördinator CC2652 from amazon

### Debug log

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/12209810/log.txt)

St0Ma, 2023-08-12:
I can confirm the Problem. 

How can I help to solve this Problem?

TjardE, 2023-09-12:
issue is still there

TjardE, 2023-10-13:
issue still there. Hope this gets fixed somehow. I don't need neighbours Hue devices in my network.

jesperholstjh, 2023-10-30:
I'm facing the same issue.

hart0x, 2023-11-09:
Me to, with permit join enabled everything works fine though.

chris-1243, 2024-02-16:
I am using 7 PTM215Z without any issues and permit join is off.

Have you followed correctly the [pairing](https://www.zigbee2mqtt.io/devices/PTM_215Z.html) method ?

Have a look there as well to get them working perfectly
[](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices)

TjardE, 2024-02-16:
Hello @chris-1243 and thank you for your suggestion. 

What I understand now is that when pairing the switch I have to choose if I use the JOIN ALL or just the JOIN(devicename). 
With this choise I choose broadcast or unicast.
When I choose unicast I must have a device that understands this unicast Green power language and that is called a proxy.

OK I reset the switch by holding all 4 buttons for longer than 7 seconds,
Joined it again but now with the option Join(devicename). And it joined..
I confirmed with the 2 buttons the channel and all should be well.

But still it does not work when on the settings page of zigbee2mqtt the permit join tickbox is empty.
When I tick it and press submit, the switch works as it should.

I have now the question : is the Ikea lamp not able to do the green power translation and when permit join is on, some broadcast happens and the usb stick translates?

Or is it still a bug?

How can I tell if the Ikea lamp understands the green power proxy?

This is all very confusing to me. I am glad that I can get the switch to work. But it is little inconvenient to see neighbours HUE things pop up in my config. (I know.. I just delete them and block them from rejoining but still..)

TjardE, 2024-02-16:
OK I looked further into it. When I switch debug on I found that the working switch with permit join off sends A 
commandNotification

When I permit join on it sends
commandnotification AND commandcommissioningNotification

The NOT WORKING switch sends nothing when permit join OFF and it sends commandcommissioningNotification with permit join ON.

So the question is. How can I make the NOT working switch also send the commandNotification????

TjardE, 2024-02-16:
And ofcourse I tried and tried and reset to factory, changed the channel and reset and factory reset again. Then changed the channel again and joined with Join(device).
And wtf... It sends the commandNotification correctly now. And it also works with the permit join off....
I have to redo this process with all my non working switches and then I will close this ticket if all works, but I have good hopes!. Thank you @chris-1243 for the suggestion

TjardE, 2024-02-19:
@chris-1243 

I am using lots of IKEA :
 
LED2003G10, LED1837R5, LED2102G3, etc.. The one doing the routing now is E1525/E1745 from IKEA.
No HUE in the network just Enocean, SONOFF, Lanesto, Tuya, AwoX.

I understood that more recent a lot of IKEA stuff does the understanding of the Zigbee Green Power frames.

For now the Enocean switches work (yeah!) also with the Permit Join OFF. That is after I re-joined the "before non-working switches" via a specific device (IKEA lamp nearby). I can see in the debug logs the difference : When the switch is joined correctly it sends commandNotification AND commandcommissioningNotification.

I keep monitoring everything for a week and if it still works I will close this ticket.

Thank you again for pointing me in the right direction.

closed 2024-02-24 by TjardE

samuele2723, 2024-08-04:
Hello there, i have same issue with 25x PT 215Z in my environment. if Permit-Join is off, nothing works. when is on they work back. i didn't do debug but i guess i same missing "commandcommissioningNotification."

Majority of lights i have around is Gledopto brand which i unicast-paired to them

i recently repaired all from scratch and i would like to know how to make these unicast greenpower switches the most reliable possible.

thanks for suggestion

chris-1243, 2024-08-04:
@samuele2723 

I would suggest to read this page from [Zigbee2mqtt documentation](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices)

I have no experience with Gledopto products and I can not tell you if they are ZGP capable.
You could try to [sniff](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) zigbee traffic

And have a look at this long thread #22897

rdorys, 2024-09-09:
@samuele2723 I had the issue yesterday. 
I managed to make it work by removing (force = true) the switch from z2m. And then re do the pairing process from scratch. If I don’t remove it from the network, I face the issue. 
good luck

ErikSteiner, 2025-01-08:
Hey there, I bought several Senic Friends of Hue devices, with an enOcean PTM 215 Z. At first I made a mistake, so I didn't notice, that the pairing was success. I noticed my error, only after the second device was successful paired. Long story short, the first device was paired with my network, but I changed the settings of the enOcean somehow and now it is unreachable.

I then removed the switch from my ZigBee network. It is interesting, that I first removed it from the device list of Zigbee2MQTT and afterwards it was still on the device page of Home Assistant.

Now comes the problem: I guess that I need to factory reset the switch, but I am unsure, how to do it. I have a single rocker and a double rocker available. However, it is described to press all four buttons at the same time. Moreover, if I understand it correctly, I also need to press both energy bars in addition to that.

Either I am wrong, or I can not pair the switch afterwards. My ZigBee network is on channel 11, and so I pressed the Top Left button for 7 seconds, while I activated the pairing in my ZigBee network. I tried it to permit join from all devices and only for the coordinator, but I failed.

I would be happy for any helpful information to this issue.

Basti189, 2025-01-24:
Same issue, posted Dez 2023
https://forum.iobroker.net/topic/71463/ptm-215z-funktioniert-nur-w%C3%A4hrend-des-joining

ErikSteiner, 2025-01-25:
Did you join them via `Permit join (DEVICE) `? Of yes, what kind of device?

I've noticed, that I am not able to join them with `Permit join (all)`, nor `Permit join (Coordinator)`. However, I was successfully joining them with a IKEA bulb and with a MOES thermostats (both via `Permit join (DEVICE)`).

What is your ZigBee channel?

Basti189, 2025-01-26:
I have already successfully integrated 6 devices; it’s just two that simply won’t work. I added all of them using "Join All." For these two, neither "Join All" nor "Join (Coordinator)" helps—they only work during the pairing process. It doesn’t matter whether I use ZHA or Z2M; both show the same behavior/problem. My coordinator is a LaunchPad CC2652R1.

chris-1243, 2025-01-26:
@Basti189 

`zstack` does not translate directly ZGP frames. It is a must to translate them via a specific router. Please, read this [link](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html) carefully.

Before doing a new pairing process, factory reset your PTM215Z. In the aera/room where you wish to use the PTM, check if there is a [compatible](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html) router. Choose it and select `permit join this router`. You will ensure a smooth functionning as it will be paired as `unicast`.

Strictly follow the [how to](https://www.zigbee2mqtt.io/devices/PTM_215Z.html) pair your PTM215Z documentation provided on Z2M website.

Basti189, 2025-01-26:
@chris-1243 
allowing new devices to join via 0x60a423fffe0e5200 (Silvercrest smart plug - Router)
z2m: Device '0x0000000001765c98' joined
z2m: Successfully interviewed '0x0000000001765c98', device has successfully been paired
z2m: Device '0x0000000001765c98' is supported, identified as: EnOcean Pushbutton transmitter module (PTM 215Z)
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001765c98/action', payload 'press_1'
z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x0000000001765c98/action', payload 'release_1'
z2m: Zigbee: disabling joining new devices.
After that, I no longer receive any messages from the device. If I start the pairing process again, the button presses are immediately visible again.

chris-1243, 2025-01-26:
@Basti189 

> After 1.5 hours, I actually managed to connect one of the two by resetting, deleting, and reconnecting it. Why or how, I have no idea.

If for any reason the device was not correctly removed from your database, it may cause some troubles.
Force remove your device, factory reset it and strictly follow the pairing process with all the advices given in the previous messages.
Once again, use a specific router while pairing your PTM in order to use `unicast` mode. If paired as `broadcast` (using the `permit join all`), you might be facing some latencies.
From Z2M docs : 
```
Note: Heavy use of broadcasts can negatively impact performance of your network (See Broadcasts).

```

Your `Silvercrest smart plug - Router` is not listed in this router [table](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html). Would you be able to [sniff](https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html) your trafic to check which mode is used while pairing a ZGP device?