Koenkk/zigbee2mqtt #18465: [New device support]: Philips hue elipse 92900031513

ptasigryp, 2023-07-30:
### Link

https://www.philips-hue.com/en-gb/p/bulb-ellipse---e27-smart-bulb/8719514419278#overview

### Database entry

{"id":8,"type":"Router","ieeeAddr":"0x001788010d4836b5","nwkAddr":28633,"manufId":4107,"manufName":"Signify Netherlands B.V.","powerSource":"Mains (single phase)","modelId":"LCZ001","epList":[11,242],"endpoints":{"11":{"profId":260,"epId":11,"devId":269,"inClusterList":[0,3,4,5,6,8,4096,64515,768,64513,64516],"outClusterList":[25],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":1,"dateCode":"20220502","swBuildId":"1.93.11","zclVersion":2,"interviewCompleted":true,"meta":{},"lastSeen":1690743666259,"defaultSendRequestWhen":"immediate"}

### Comments

I tried making myself a converted but couldn't accomplish anything working.

### Supported color modes

color temperature, rgb, dimming

Koenkk, 2023-07-31:
Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

ptasigryp, 2023-07-31:
> Please add the color temperature range as described on https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_3-adding-converter-s-for-your-device

Read result of 'lightingColorCtrl': {"colorTempPhysicalMin":153,"colorTempPhysicalMax":500}

Koenkk, 2023-08-01:
Added!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-01 by Koenkk

ptasigryp, 2023-08-02:
@Koenkk I just tested on the latest dev branch. The configuration works but it doesn't allow to change RGB and also I can see a problem for not exposing it in the entities list, which for example doesn't allow me to add this bulb to Adaptive Lighting add-on.

Koenkk, 2023-08-03:
Added the color support!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)