Koenkk/zigbee2mqtt #18469: EGLO - Remote Control not pairing correctly.

njordan77, 2023-07-31:
### What happened?

I'm on latest stable version of Z2M, have many items paired via ZZH stick. All works fine.
Recently received an EGLO remote control, on the backside its mentioned ModelNr. 31753. pairing starts but ends without success. in your list of EGLO supported devices the model nr is 99099. Is its not clear to me if its a different item. as from the picture it looks totally the same. 

Any idea what i can try to see if its the same model and paring is wrong or the model itself is not yet suppored. maybe 2 models but should work if yoou would add the mdoel nr.

Thanks for your feedback,
Norbert

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

the latest available for ZZH

### Adapter

ZZH!

tonnyongit, 2023-08-18:
I have exactly the same problem as you, only with me it is recognized as an AWOX 33952. I also think that an adjustment needs to be made.

Changing this should work:

```
    {
        zigbeeModel: ['ERCU_3groups_Zm'],
        model: '99099/31753',
        vendor: 'EGLO',
        description: '3 groups remote controller',
        fromZigbee: [fz.command_on, fz.awox_colors, fz.awox_refresh, fz.awox_refreshColored, fz.command_off,
            fz.command_step, fz.command_move, fz.command_move_to_level, fz.command_move_to_color_temp,
            fz.command_stop, fz.command_recall, fz.command_step_color_temperature],
        toZigbee: [],
        exposes: [e.action(['on', 'off', 'red', 'refresh', 'refresh_colored', 'blue', 'yellow',
            'green', 'brightness_step_up', 'brightness_step_down', 'brightness_move_up', 'brightness_move_down', 'brightness_stop',
            'recall_1', 'color_temperature_step_up', 'color_temperature_step_down'])],
    },
```
I have tried to test this but when I add it to the config it is not recognized and it keeps showing up as the AWOX 33952, I don't know if I am doing something wrong or I am missing something

meredrica, 2023-09-03:
> I have exactly the same problem as you, only with me it is recognized as an AWOX 33952

Exactly the same here. I updated the remote through the app, didn't change much.
It's difficult to pair but once it is it does nothing for
I can see the map view registering sends but the other device doesn't react.

If anyone can point me in the correct direction how to fix/diagnose this I can help with implementing the solution

kzajac83, 2023-09-21:
The same issue for AwoX 33952, paring finish error:

`Failed to interview '0xa4c1388118afa5ec', device has not successfully been paired`

Detail:
Have paired already remote controller Avox 33952 few months ago, all button working fine but more than one month the device was not used and toady try use it and do not work. Not sure why remote controller become offline (battery is fine), maybe due to missing availability in my Z2M configuration?   After add availability in Z2M configuration in summary devices appear remote controller as offline (similar offline with aqara cube and door sensor). 

Before repairing AvoX 33952 update FW to newest version and next try paring to Z2M.  Paring start good but cannot finish as you describe , the same problem, now have offline status but when jump into expose then see reaction on pushing buttons in action parameter. So in practical partially has been paired but status still offline and in status tab I see only:

```
{
    "linkquality": 58
}
```

Additional when pushing buttons:
- red
- blue
- yellow
- green
- rounded arrow 

appear error:
`Exception while calling fromZigbee converter: awoxRemoteHelper is not defined}`

My log from start paring to fail status

```
Info 2023-09-21 15:38:44Device '0xa4c1388118afa5ec' joined
Info 2023-09-21 15:38:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388118afa5ec","ieee_address":"0xa4c1388118afa5ec"},"type":"device_joined"}'
Info 2023-09-21 15:38:44MQTT publish: topic 'homeassistant/sensor/0xa4c1388118afa5ec/action/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388118afa5ec/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1388118afa5ec"],"manufacturer":"AwoX","model":"Remote controller (33952)","name":"0xa4c1388118afa5ec","sw_version":"2.4.1_929"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"Action","object_id":"0xa4c1388118afa5ec_action","origin":{"name":"Zigbee2MQTT","sw":"1.33.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c1388118afa5ec","unique_id":"0xa4c1388118afa5ec_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Info 2023-09-21 15:38:44MQTT publish: topic 'homeassistant/sensor/0xa4c1388118afa5ec/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388118afa5ec/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1388118afa5ec"],"manufacturer":"AwoX","model":"Remote controller (33952)","name":"0xa4c1388118afa5ec","sw_version":"2.4.1_929"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c1388118afa5ec_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.33.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1388118afa5ec","unique_id":"0xa4c1388118afa5ec_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Info 2023-09-21 15:38:44Starting interview of '0xa4c1388118afa5ec'
Info 2023-09-21 15:38:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388118afa5ec","ieee_address":"0xa4c1388118afa5ec","status":"started"},"type":"device_interview"}'
Debug 2023-09-21 15:38:44Received MQTT message on 'homeassistant/sensor/0xa4c1388118afa5ec/action/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388118afa5ec/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1388118afa5ec"],"manufacturer":"AwoX","model":"Remote controller (33952)","name":"0xa4c1388118afa5ec","sw_version":"2.4.1_929"},"enabled_by_default":true,"icon":"mdi:gesture-double-tap","name":"Action","object_id":"0xa4c1388118afa5ec_action","origin":{"name":"Zigbee2MQTT","sw":"1.33.0","url":"https://www.zigbee2mqtt.io"},"state_topic":"zigbee2mqtt/0xa4c1388118afa5ec","unique_id":"0xa4c1388118afa5ec_action_zigbee2mqtt","value_template":"{{ value_json.action }}"}'
Debug 2023-09-21 15:38:44Received MQTT message on 'homeassistant/sensor/0xa4c1388118afa5ec/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"},{"topic":"zigbee2mqtt/0xa4c1388118afa5ec/availability","value_template":"{{ value_json.state }}"}],"availability_mode":"all","device":{"identifiers":["zigbee2mqtt_0xa4c1388118afa5ec"],"manufacturer":"AwoX","model":"Remote controller (33952)","name":"0xa4c1388118afa5ec","sw_version":"2.4.1_929"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c1388118afa5ec_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.33.0","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c1388118afa5ec","unique_id":"0xa4c1388118afa5ec_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Debug 2023-09-21 15:38:44Device '0xa4c1388118afa5ec' announced itself
Info 2023-09-21 15:38:44MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388118afa5ec","ieee_address":"0xa4c1388118afa5ec"},"type":"device_announce"}'
Debug 2023-09-21 15:38:46Retrieving state of '0xa4c1388118afa5ec' after reconnect
Error 2023-09-21 15:39:04Failed to interview '0xa4c1388118afa5ec', device has not successfully been paired
Info 2023-09-21 15:39:04MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c1388118afa5ec","ieee_address":"0xa4c1388118afa5ec","status":"failed"},"type":"device_interview"}'
```

kzajac83, 2023-10-22:
Anti-stale comment, issue still exist today check (it was worked correctly but in Jun - Jul functionality has been broken)

njordan77, 2024-04-22:
still does not work. is there any progress? has it been implemented at all? so should it work already? Thanks

closed 2025-01-13 by github-actions[bot]