Koenkk/zigbee2mqtt #18471: [New device support]: lumi.light.acn132 Aqara Smart Light STRIP RGB/W

MiAutomations, 2023-07-31:
### Link

https://www.aqara.cn/LED-Strip-T1_overview

### Database entry

{"id":61,"type":"Router","ieeeAddr":"0x54ef4410008c1c76","nwkAddr":44031,"manufId":4447,"manufName":"Aqara","powerSource":"DC Source","modelId":"lumi.light.acn132","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[5,4,3,0,768,8,6,64704],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"manufacturerName":"Aqara","powerSource":4,"zclVersion":3,"appVersion":26,"stackVersion":27,"hwVersion":1,"dateCode":"20230506","modelId":"lumi.light.acn132"}},"aqaraOpple":{"attributes":{"238":6682,"247":{"type":"Buffer","data":[109,32,0,110,32,2,113,32,1,114,32,10,116,32,2]},"1301":1,"1302":100}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":26,"stackVersion":27,"hwVersion":1,"dateCode":"20230506","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1690810124380,"defaultSendRequestWhen":"immediate"}

### Comments

I'm using the last version of the Z2M and the device is not supported yet 

Can you please add this device to be available in the next releases ? 

I also try to get the Supported color modes and Color temperature range, but without success sorry 

Error 2023-07-31 14:42:46Publish 'set' 'read' to '0x54ef4410008c1c76' failed: 'Error: Read 0x54ef4410008c1c76/1 lightingColorCtrl(["colorTemperature"], {"sendWhen":"immediate","timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'

Thank you for your support and best regards

FabienVINCENT, 2023-08-20:
Has anyone successfully configured it?

markus-barta, 2023-08-24:
I've got my hands on the Aqara Smart Light STRIP as well! 🌈 Would love to see support for this amazing ledstrip!

A massive shoutout to the Zigbee2MQTT-team! 🎉 Your dedication and hard work never cease to inspire 🌟!

FabienVINCENT, 2023-08-24:
This work for me at the moment but not fully compatible
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const ota = require('zigbee-herdsman-converters/lib/ota');

const definition = {
    zigbeeModel: ['lumi.light.acn132'],
    model: 'RLS-K01D',
    vendor: 'Aqara',
    description: 'ZIGBEE 3.0 LED STRIP T1', 
    extend: extend.light_onoff_brightness_colortemp_color({
    disableEffect: true, disablePowerOnBehavior: true, disableColorTempStartup: true, colorTempRange: [153, 370]}),
    ota: ota.zigbeeOTA,
};

module.exports = definition;
```

Koenkk, 2023-08-25:
@FabienVINCENT this looks good, what is not working?

FabienVINCENT, 2023-08-25:
the strip led allows to have several colors at the same time and it is not possible for the moment

markus-barta, 2023-08-25:
@FabienVINCENT thanks for the contribution - I can confirm this external converter works well for the basic stuff 🎉 (FYI: @Koenkk).

Would love to see segment support in the future! 🤗

Koenkk, 2023-08-26:
Thanks @FabienVINCENT, I added the device for out-of-the-box-support.

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

MiAutomations, 2023-08-26:
Thank you all for your work and support

robinbohnen, 2023-09-02:
Great work! Only extensions won't work yet (only the base 2m works) and color-segments don't work yet.

iamandymcinnes, 2023-09-20:
Hi good job getting this added so quickly, really pleased I can at least control my led strip in HA without having to get an Aqara hub! Just wondered do you think it will be possible to add the segment control support, I'm not entirely clued up on how features like that get support in zigbee2mqtt?

ZenoBell, 2023-09-23:
wow fantastic job guys ! i just configured mine as well and works out of the box ! Hoping to see the color segmentation and extension support added in the future! For now you can just use the button in the controller to light up the entire lighstrip along with the extension ! Thanx once more zigbee2mqtt team !

dariuszpaluch, 2023-09-26:
https://www.zigbee2mqtt.io/devices/LGYCDD01LM.html

Finnlife, 2023-10-08:
Did some of you had the issue that the Strip is never being found by z2m. No interview no log nothing. Did yall do something special to pair it? If not, what zigbeechannels and zigbeesticks do you use.

PulsarFX, 2023-10-20:
> Did some of you had the issue that the Strip is never being found by z2m. No interview no log nothing. Did yall do something special to pair it? If not, what zigbeechannels and zigbeesticks do you use.

Try to put z2m into join mode. then on the strip, press and hold the power switch for 4s. It will start blinking and z2m should list it short after that.

Finnlife, 2023-10-21:
> Try to put z2m into join mode. then on the strip, press and hold the power switch for 4s. It will start blinking and z2m should list it short after that.

Thank you, i did that. The issue was my Conbee II Stick. The strip does not  work with it on my end.

pitr12, 2023-11-02:
Is it somehow possible to make also extensions work? I need to make strip at least 5m long, but extensions work only with white color. RGB is not working with added extensions. Seems like we need to define strip length using app (but that requires Aqara hub). Is there some way how to make this work without hub?

robinbohnen, 2023-11-02:
I have a hub; but even if you have configured it with the hub and disconnect and connect again with Zigbee2Mqtt it won't work.

Abadin, 2023-11-10:
Just bought an extension kit only to find this post :-(

Finnlife, 2023-11-10:
Seems like there are a couple of things that are not fully integrated. 
I think there is a way in the app to enable the Music sync, Default Animations and other stuff too.

el-0, 2023-11-15:
Today I added Aqara Zigbee 3.0 LED strip T1 (lumi.light.acn132) and it seems it does not support setting multiple colors in specific zones of the strip. (RGBIC) - I'm running Zigbee-herdsman-converters version 15.116.0

Is it possible to add this like it was added for the HUE gradient strip ? 
I ordered an Aquara HUB to do the sniffing of the required payloads and trying to figure out how it is controlled. (probably vendor specific like for HUE)

I will try to use this for personal use but I was wondering if I could contribute here to make it work generally ?

007Nev, 2023-11-30:
For those who struggling with the extension kit with homeassistant, I’ve got a message from Aqara that you have to change the value of led strip length in your Aqara app(I don’t have one since I don’t have the Aqara hub..).

Do I have to buy the Aqara hub? I’ve been using my Sonoff dongle plus quite nice and doing well with Aqara mostly but this led extension kit makes me sad…

Any thoughts or can anyone chnage the length in your Aqara app if you have the hub for me?
 
Thanks. 

[image]
[image]

007Nev, 2023-11-30:
FYI, here are the pictures of Aqara Led strip setting menu in its app.

Could anyone solve this issue? 

[image]
[image]

Koenkk, 2023-11-30:
More features will be supported in tomorrows new release (thanks to @kirovilya)

007Nev, 2023-11-30:
> More features will be supported in tomorrows new release (thanks to @kirovilya)

Is this issue solved on the next release, sir ?
Thanks for your great work as always.

ZenoBell, 2023-12-01:
Tested as well with 1 m extension, works like charm ! Thanx guys !

JBGITHUB12, 2023-12-04:
> Alright guys, as of 1st.DEC.2023, newly updated Z2M fixed the issue. Make sure to change the length of LED extension part correctly. Mine works great with all mixture of colors.
> 
> Cheers. Thanks for great work, @Koenkk.

Awesome, thanks so much for doing this works great and you get the chase effects

robinbohnen, 2023-12-05:
Yeah great work, it works perfectly fine right now. The only thing I am missing is the gradient option.

Rhysgh, 2023-12-30:
Awesome work here all involved.

I two would like to use segments/gradients without needing a hub if possible 🤞

CrazyThunderbird, 2024-01-02:
Well sorry for digging in this thread but i have the issue since a few days, that it just cant switch colors!

I'm limited to warm or cold white, tried resetting the strip etc. nothing worked so far.

[image]
[image]

Rhysgh, 2024-01-02:
> Well sorry for digging in this thread but i have the issue since a few days, that it just cant switch colors!
> 
> I'm limited to warm or cold white, tried resetting the strip etc. nothing worked so far.
> 
> I think it happened after i made an OTA update atleast it told me one was available and now i tells me allways there's an update available
> 
> ![grafik](https://private-user-images.githubusercontent.com/109760390/293618600-0959a763-a203-4208-8a89-9a6a1505723c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyMTIzMjUsIm5iZiI6MTcwNDIxMjAyNSwicGF0aCI6Ii8xMDk3NjAzOTAvMjkzNjE4NjAwLTA5NTlhNzYzLWEyMDMtNDIwOC04YTg5LTlhNmExNTA1NzIzYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwMlQxNjEzNDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mNjcwMGU4ZDVhZDlkM2U1NGE4NzhhOTA3YzU4NjFjNzg0Njc5NGE4ZDE0NGYwMDhlYTk0ZWNkOWYxMzlhNzJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.59WH3NRC4S8H0e_izetA974o4h_9TkpDPCjqvQ1aphs) ![grafik](https://private-user-images.githubusercontent.com/109760390/293618634-e6c3ca58-00d6-4026-b60d-dec8b9874205.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQyMTIzMjUsIm5iZiI6MTcwNDIxMjAyNSwicGF0aCI6Ii8xMDk3NjAzOTAvMjkzNjE4NjM0LWU2YzNjYTU4LTAwZDYtNDAyNi1iNjBkLWRlYzhiOTg3NDIwNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDEwMlQxNjEzNDVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZWExNzA0MWJlMjdjZDEwYjg4ZTI3MDMzNTFlZDAzOThkNjM4MDE0MTJjNDc5MmU0OGI0YTk5NTliZDk3YWFhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.2tARg3PWK-UmxVO2oPoV9Xrkh5bDqkIFwEYrJdGKRUk)

I also have this i set up new devices a few days ago and all was fine then suddenly the colour options vanished, i have been unable to manually change them either  ( with call services)

3x1at1on, 2024-01-05:
> Well sorry for digging in this thread but i have the issue since a few days, that it just cant switch colors!
>
> I'm limited to warm or cold white, tried resetting the strip etc. nothing worked so [...]

Same issue here with the Aqara T1 LED strip LGYCDD01LM . Only warmth colour control available and no RGB control. New install HA 2024.01 Z2M 1.35.0-1

FabPara64, 2024-01-05:
Hi,
Same issue. RGB control has left

Spitefulboy07, 2024-01-06:
I have the same problem! Only white light!

CrazyThunderbird, 2024-01-07:
Update to the newest version of Z2M and it's working again.

TehloWasTaken, 2024-01-14:
Have people here been able to set multiple colors on a 2M ledstrip? I can only select one for the entire strip, not use the zones..

Rhysgh, 2024-01-16:
> Have people here been able to set multiple colors on a 2M ledstrip? I can only select one for the entire strip, not use the zones..

Im hoping this comes soon

chrisb86, 2024-05-06:
I got my Light strip today and was able to pair it without problems. RGB control works fine but in white mode the color temperature range is inverted. When I set it cooler, the light gets warmer and vice verca. I made a firmware update, switched to z2m edge and repaired the strip several times but nothing helps.

namardnA, 2024-09-15:
Awesome work mate! This is fantastic work!
I was wondering if by any chance it might be possible to invert the length?
So that you could change lenghts to per example the first meter from the beginning, or the first meter from the end?

Would be amazing if that could be achieved, or is this something that will only ever work with an aqara hub?

Matthe95, 2024-10-06:
Hello, I bought the LED and installed it via Zigbee2MQTT yesterday, but I have the problem that as soon as I put the LED in RGB, only half of the LED lights up. However, I indicated the right size in the application, but it doesn't change anything. Does anyone have a solution?
No problem as soon as I leave it white though.

closed 2025-01-13 by github-actions[bot]

david2242, 2025-11-02:
I’m running Zigbee2MQTT 2.6.2-dev with herdsman-converters 25.35.1,
the lumi.light.acn132 definition is up-to-date, but color control still behaves incorrectly (color doesn’t match the payload).

I've just gained so much switching from ZHA to Z2M, but I also lost the LED strip working fine.