Koenkk/zigbee2mqtt #18478: TS011F_TZ3000_hdopuwv6 not updating power

BenPortner, 2023-07-31:
### What happened?

I have a lamp plugged into the Zigbee socket. When switching on the lamp, z2m correctly shows a consumption of 10-11 W. However, after switching off the lamp, the power reading stays at this value for minutes, even hours. Current seems to update correctly meanwhile.

The issue occurs on 1.32.1 and 1.32.1-dev. I also tried installing the [ext_converter](https://github.com/Koenkk/zigbee2mqtt/issues/16709#issuecomment-1509617792) without success.

### What did you expect to happen?

I expect the value to drop to zero after switching off the lamp.

### How to reproduce it (minimal and precise)

Connect lamp, switch on lamp, wait a few seconds, switch off lamp.

### Zigbee2MQTT version

1.32.1(-dev)

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

BenPortner, 2023-08-16:
@Koenkk 
Here is the log from the pairing. For me as a novice, this line sticks out: `No converter available for 'TS011F_plug_1' with cluster 'manuSpecificBosch' and type 'raw' and data '{"data":[8,4,10,1,208,72,2,0,6],"type":"Buffer"}'`. Could this be the cause of the issue?

```
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:18: Device '0xa4c138b1705b434e' joined
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b1705b434e","ieee_address":"0xa4c138b1705b434e"},"type":"device_joined"}'
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:18: Starting interview of '0xa4c138b1705b434e'
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b1705b434e","ieee_address":"0xa4c138b1705b434e","status":"started"},"type":"device_interview"}'
z2mtest        | Zigbee2MQTT:debug 2023-08-16 23:35:18: Device '0xa4c138b1705b434e' announced itself
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:18: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c138b1705b434e","ieee_address":"0xa4c138b1705b434e"},"type":"device_announce"}'
z2mtest        | Zigbee2MQTT:debug 2023-08-16 23:35:19: Received Zigbee message from '0xa4c138b1705b434e', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS011F"}' from endpoint 1 with groupID 0
z2mtest        | Zigbee2MQTT:debug 2023-08-16 23:35:19: Skipping message, definition is undefined and still interviewing
z2mtest        | Zigbee2MQTT:debug 2023-08-16 23:35:19: Received Zigbee message from '0xa4c138b1705b434e', type 'attributeReport', cluster 'manuSpecificTuya_3', data '{"53248":1,"53249":0,"53250":2854840,"53251":122,"53252":10912,"53253":1261,"powerOnBehavior":2,"switchType":0}' from endpoint 1 with groupID 0
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:19: Successfully interviewed '0xa4c138b1705b434e', device has successfully been paired
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:19: Device '0xa4c138b1705b434e' is supported, identified as: TuYa Smart plug (with power monitoring) (TS011F_plug_1)
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:19: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"Smart plug (with power monitoring)","exposes":[{"features":[{"access":7,"description":"On/off state of the switch","name":"state","property":"state","type":"binary","value_off":"OFF","value_on":"ON","value_toggle":"TOGGLE"}],"type":"switch"},{"access":7,"description":"Recover state after power outage","name":"power_outage_memory","property":"power_outage_memory","type":"enum","values":["on","off","restore"]},{"access":7,"description":"LED indicator mode","name":"indicator_mode","property":"indicator_mode","type":"enum","values":["off","off/on","on/off","on"]},{"access":1,"description":"Instantaneous measured power","name":"power","property":"power","type":"numeric","unit":"W"},{"access":1,"description":"Instantaneous measured electrical current","name":"current","property":"current","type":"numeric","unit":"A"},{"access":1,"description":"Measured electrical potential value","name":"voltage","property":"voltage","type":"numeric","unit":"V"},{"access":1,"description":"Sum of consumed energy","name":"energy","property":"energy","type":"numeric","unit":"kWh"},{"features":[{"access":3,"description":"Enables/disables physical input on the device","name":"state","property":"child_lock","type":"binary","value_off":"UNLOCK","value_on":"LOCK"}],"type":"lock"},{"access":1,"description":"Link quality (signal strength)","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS011F_plug_1","options":[{"access":2,"description":"State actions will also be published as 'action' when true (default false).","name":"state_action","property":"state_action","type":"binary","value_off":false,"value_on":true},{"access":2,"description":"Calibrates the power value (percentual offset), takes into effect on next report of device.","name":"power_calibration","property":"power_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for power, takes into effect on next report of device.","name":"power_precision","property":"power_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the current value (percentual offset), takes into effect on next report of device.","name":"current_calibration","property":"current_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for current, takes into effect on next report of device.","name":"current_precision","property":"current_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Calibrates the voltage value (percentual offset), takes into effect on next report of device.","name":"voltage_calibration","property":"voltage_calibration","type":"numeric"},{"access":2,"description":"Number of digits after decimal point for voltage, takes into effect on next report of device.","name":"voltage_precision","property":"voltage_precision","type":"numeric","value_max":3,"value_min":0},{"access":2,"description":"Number of digits after decimal point for energy, takes into effect on next report of device.","name":"energy_precision","property":"energy_precision","type":"numeric"," [... cut ...]
z2mtest        | Zigbee2MQTT:info  2023-08-16 23:35:35: MQTT publish: topic 'zigbee2mqtt/0xa4c138b1705b434e', payload '{"child_lock":"UNLOCK","current":0,"energy":0.08,"indicator_mode":"off/on","linkquality":211,"power":0,"power_outage_memory":"restore","state":"ON","voltage":228}'
z2mtest        | Zigbee2MQTT:debug 2023-08-16 23:37:03: Received Zigbee message from '0xa4c138b1705b434e', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0011\u0000\u0000\u0000\u0012\u0016\u0000\u0000\u0000\u0016"}' from endpoint 1 with groupID 0
```

BenPortner, 2023-08-16:
Going further in the tests, I see the following message when switching off the lamp:

```
zigbee2mqtt      | Zigbee2MQTT:debug 2023-08-17 00:27:27: Received Zigbee message from '0xa4c138b1705b434e', type 'attributeReport', cluster 'haElectricalMeasurement', data '{"activePower":0}' from endpoint 1 with groupID 0
```

So it seems that the correct message is sent and received. However, the corresponding MQTT publish event is missing. Anyone know how to fix this?

BenPortner, 2023-08-17:
I realized that my problem is caused by the "fix" of https://github.com/Koenkk/zigbee2mqtt/issues/16709. Reverting to Z2M v.1.30.3 resolves the issue. However, I would rather have a working up-to-date version. Is there anything I can do to help implement this?

Koenkk, 2023-08-17:
Please provide the stdout when using the ext converter mentioned here https://github.com/Koenkk/zigbee2mqtt/issues/16709

closed 2023-08-17 by Koenkk