Koenkk/zigbee2mqtt #18481: Zigbee scenes are not exposed to Home Assistant.

erik3rik, 2023-08-01:
### What happened?

1. I create click "Groups" in the main menu.
2. I create a group with the name "my_group"
3. I add some lights to that group
4. I create 4 different scenes for this group, one where all lights are on (name: all_on), one where all lights are off (name: all_off), one where all lights are 50% brightness (name: all_50), and one where some lights are on and some are off (name: mixed).

What happens:
One switch is visible in Home Assistant through the MQTT-integration. The `switch.test_group_2` entity has two states, `on` or `off`. Turning it on turns on all lights in the group to 100%, turning it off turns all lights in the group off.

There are no traces of any scenes (all_on, all_off, all_50, etc.)

### What did you expect to happen?

I expected that each of those 4 zigbee group (my_group) scenes (all_on, all_off, all_50, etc.) should correspond to 4 `scene` entities in Home Assitant. 

I.e. (in Home Assitant) be able to do:
`scene.turn_on` with something like `scene.all_50` as argument. And then I would expect that scene to be recalled.

Only having a switch for the entire group on/off is only usefull if you want to group e.g. wall socket plugs that are only on/off capable and should all have the same settings.

Scenes for lights include things like brightness and color temperature.

This is the way the deCONZ integration works with the Conbee 2 stick. All scenes created in that addon (i.e. the deCONZ addon) also creates a scene in Home Assistant.

Scenes created in the zigbee2mqtt addon (or a standalone installation, it does not matter) are nowhere to be seen in Hoem Assistant. Despite all discovery parameters are set and e.g. the switch was added automatically.

### How to reproduce it (minimal and precise)

1. Create click "Groups" in the main menu.
2. Create a group with the name "my_group"
3. Add some lights to that group
4. Create a scene where some lights are on and some are off, and name that scene "my_scene"
5. Turn all lights off manually (using e.g. Home Assistant, NOT pulling the plug)
6. Try to activate that scene (my_scene) in Home Assistant. 

Step 6 will fail as there is not trace of my_scene in Home Assistant, but I expected it to exist.

### Zigbee2MQTT version

1.32.1 

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2

katiuskt, 2023-08-01:
+1

I expected the same behaviour (ie: group scenes to be automagically imported into HA as scenes; also coming from DECONZ so I expected similar behaviour) but apparently, for Z2M you should create the scenes manually.

In case it helps, here is one excerp from my configuration:
```
mqtt:
  scene:
    - unique_id: kitchen_scene_standard
      name: Kitchen_Scene_Standard
      icon: mdi:palette
      command_topic: 'zigbee2mqtt/Kitchen/set'
      payload_on: '{"scene_recall": 0}'
    - unique_id: kitchen_scene_visit
      name: Kitchen_Scene_Visit
      icon: mdi:account-group
      command_topic: 'zigbee2mqtt/Kitchen/set'
      payload_on: '{"scene_recall": 1}'
```
Where 'Kitchen' is the Group Name and scene Standard has the Scene ID=0 and Visit has the Scene ID=1

I don't know if it is planned to support the automatically discover of Scenes from HA but in the meantime, this approach does the trick.

References:
https://www.zigbee2mqtt.io/guide/usage/scenes.html#using-transitions-in-scenes
https://www.home-assistant.io/integrations/scene.mqtt/

erik3rik, 2023-08-01:
Thanks for responding and confirming that my experience of zigbee2mqtt is not simply a mistake from my side.
I know about the workaround you mentioned. But it is quite a big effort when updating / changing. I mean to manually have to keep all of this in sync when you update a name or id or add a scene.

As you are, I'm coming from a deCONZ setup where this automatically happens. And as other things (e.g. temperature sensors, lights, switches, power monitors etc) are created "automagically" by the zigbee2mqtt software I think (hope) the effort in "fixing" also the scenes in the same way is quite low, know the code base, language, etc.

So perhaps this is more of a feature request than a bug, then. Not sure. Depends on if skipping scene support was deliberate or simply a bug somewehere.

katiuskt, 2023-09-01:
No stale. It still be useful if implemented

mundschenk-at, 2023-09-17:
@gdgib and I recently shared some thoughts on this in https://github.com/Koenkk/zigbee2mqtt/issues/13937#issuecomment-1722257166.

erik3rik, 2023-10-18:
Not stale

mundschenk-at, 2024-01-22:
@Koenkk: I think this issue can be closed (auto-closing with "fixes" mentions does not appear to work with `dev` branch).

closed 2024-01-24 by Koenkk

giga8yte, 2024-01-27:
I got scenes to show up in homeassistant by restarting the z2m addon.