Koenkk/zigbee2mqtt #18483: Aqara H1M - 2 and 3 Gang incomplete info

osamaabdo, 2023-08-01:
### What happened?

Hi,
I have 3 Aqara H1M switches
H1M - 1 Gang identified as ZNQBKG24LM
H1M - 2 Gang identified as ZNQBKG25LM
H1M - 3 Gang identified as ZNQBKG25LM

ZNQBKG24LM exoses all info - such as energy, voltage and power_loss_recovery

the problem is that ZNQBKG25LM and ZNQBKG26LM do not expose the above. Any way to fix this

### What did you expect to happen?

to expose all info

### Zigbee2MQTT version

1.32.1

### Adapter firmware version

20230507

### Adapter

Sonoff USB Dongle P

Rumlik, 2023-08-04:
I have similar problem
I have 2gang ZNQBKG25LM switch and apparently there is some error in the code. In relay mode, everything works fine. But if you switch to decoupled switching mode, the action sensor then doesn't distinguish between pressing the button right or left. Thus, the right and left button report the same.
Would anyone know how to fix this? Thank you very much.

osamaabdo, 2023-08-08:
I created an external converter, but still unable to get power recovery and turn off LED at night 

`zigbeeModel: ['lumi.switch.acn030'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZNQBKG25LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara H1M Doubel', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.on_off, fz.xiaomi_power, fz.xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [
        tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_flip_indicator_light,
    ],
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
    },
    meta: {multiEndpoint: true},
    exposes: [
        e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'),
        e.power(), e.energy(), e.voltage(),
     e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
        .withDescription('Decoupled mode for left button')
        .withEndpoint('left'),
     e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
        .withDescription('Decoupled mode for right button')
        .withEndpoint('right'),
     e.action(['single_left', 'double_left', 'single_right', 'double_right',
        'single_left_right', 'double_left_right', 'single_all', 'double_all']),
        e.device_temperature(), e.flip_indicator_light(),
    ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend`

osamaabdo, 2023-08-13:
@Rumlik Did you ever figure out the action issue ?

osamaabdo, 2023-08-14:
I discovered a solution for the ZNQBKG26LM switch to add power_loss_recovery, and led_disabled_at_night
The model number (ZNQBKG26LM) needs to be added to toZigbee
`
 xiaomi_switch_power_outage_memory: {
        key: ['power_outage_memory'],
        convertSet: async (entity, key, value, meta) => {
            if (['SP-EUC01', 'ZNCZ04LM', 'ZNCZ15LM', 'QBCZ14LM', 'QBCZ15LM', 'SSM-U01', 'SSM-U02', 'DLKZMK11LM', 'DLKZMK12LM',
                'WS-EUK01', 'WS-EUK02', 'WS-EUK03', 'WS-EUK04', 'QBKG19LM', 'QBKG18LM', 'QBKG20LM', 'QBKG25LM', 'QBKG26LM', 'QBKG28LM', 'QBKG29LM',
                'QBKG31LM', 'QBKG34LM', 'QBKG38LM', 'QBKG39LM', 'QBKG40LM', 'QBKG41LM', 'ZNDDMK11LM', 'ZNLDP13LM', 'ZNQBKG31LM',
                'WS-USC02', 'WS-USC03', 'WS-USC04', 'ZNQBKG24LM', 'ZNQBKG25LM',
`

And in   xiaomi_led_disabled_night:

but i am still struggling to get actions in ZNQBKG25LM, as mentioned, it only has single_action (no single_left_action) and single_right

osamaabdo, 2023-08-15:
i finally figured it out with the following converter - hopfully it is added to the official release
it is basically adding ZNQBKG25LM to the multistate action

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const {
    hasAlreadyProcessedMessage
} = require('zigbee-herdsman-converters/lib/utils');

const new_xiaomi_multistate_action= {
    cluster: 'genMultistateInput',
    type: ['attributeReport'],
    convert: (model, msg, publish, options, meta) => {
        //console.log('Received presentValue:', msg.data['presentValue'], 'Endpoint ID:', msg.endpoint.ID);

        if (hasAlreadyProcessedMessage(msg, model)) return;
        let actionLookup = {0: 'hold', 1: 'single', 2: 'double', 3: 'triple', 255: 'release'};
        if (model.model === 'WXKG12LM') {
            actionLookup = {...actionLookup, 16: 'hold', 17: 'release', 18: 'shake'};
        }

        let buttonLookup = null;
        if (['WXKG02LM_rev2', 'WXKG07LM', 'WXKG15LM', 'WXKG17LM'].includes(model.model)) {
            buttonLookup = {1: 'left', 2: 'right', 3: 'both'};
        }
        if (['QBKG12LM', 'QBKG24LM'].includes(model.model)) buttonLookup = {5: 'left', 6: 'right', 7: 'both'};
        if (['QBKG39LM', 'QBKG41LM', 'WS-EUK02', 'WS-EUK04', 'QBKG20LM', 'QBKG28LM', 'QBKG31LM','ZNQBKG25LM'].includes(model.model)) {
            buttonLookup = {41: 'left', 42: 'right', 51: 'both'};
        }
        if (['QBKG25LM', 'QBKG26LM', 'QBKG29LM', 'QBKG34LM', 'ZNQBKG31LM', 'ZNQBKG26LM'].includes(model.model)) {
            buttonLookup = {
                41: 'left', 42: 'center', 43: 'right',
                51: 'left_center', 52: 'left_right', 53: 'center_right',
                61: 'all',
            };
        }
        if (['WS-USC02', 'WS-USC04'].includes(model.model)) {
            buttonLookup = {41: 'top', 42: 'bottom', 51: 'both'};
        }

        const action = actionLookup[msg.data['presentValue']];
        if (buttonLookup) {
            const button = buttonLookup[msg.endpoint.ID];
            if (button) {
                return {action: `${action}_${button}`};
            }
        } else {
            return {action};
        }
    },
}
const definition = {
    zigbeeModel: ['lumi.switch.acn030'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'ZNQBKG25LM', // Vendor model number, look on the device for a model number
    vendor: 'Xiaomi', // Vendor of the device (only used for documentation and startup logging)
    description: 'Aqara H1M Doubel', // Description of the device, copy from vendor site. (only used for documentation and startup logging)

    fromZigbee: [fz.on_off, fz.xiaomi_power, new_xiaomi_multistate_action, fz.aqara_opple],
    toZigbee: [
        tz.on_off, tz.xiaomi_switch_operation_mode_opple, tz.xiaomi_flip_indicator_light,tz.xiaomi_led_disabled_night,
        tz.xiaomi_switch_power_outage_memory,
    ],
    meta: {multiEndpoint: true},
    endpoint: (device) => {
        return {'left': 1, 'right': 2};
[... 113 lines cut ...]
    endpoint: (device) => {
        return {'left': 1, 'center': 2, 'right': 3};
    },

    exposes: [
        e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right'),
        e.power_outage_memory(),e.power(), e.energy(), e.voltage(),e.led_disabled_night(),
        e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
            .withDescription('Decoupled mode for left button')
            .withEndpoint('left'),
        e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
            .withDescription('Decoupled mode for center button')
            .withEndpoint('center'),
        e.enum('operation_mode', ea.ALL, ['control_relay', 'decoupled'])
            .withDescription('Decoupled mode for right button')
            .withEndpoint('right'),
        e.action(['single_left', 'double_left', 'single_center', 'double_center', 'single_right', 'double_right',
            'single_left_center', 'double_left_center', 'single_left_right', 'double_left_right',
            'single_center_right', 'double_center_right', 'single_all', 'double_all']),
        e.device_temperature(), e.flip_indicator_light(),
    ], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
    
};

module.exports = definition;

Koenkk, 2023-08-15:
Could you make a pull request for this?

Rumlik, 2023-08-15:
> Could you make a pull request for this?

I will try this tomorrow or next day. To check this.

Rumlik, 2023-08-17:
> Could you make a pull request for this?

For 2 gang everything good now with this converter 👌

Its-Manzzi, 2023-09-08:
@osamaabdo did you manage to solve for 3 gang for the decoupled switching and the led light? Thank you

Whipsmith, 2023-09-22:
@Rumlik did you make a PR with this fix?

Also, do you update your switches to the latest firmware?

closed 2023-11-02 by github-actions[bot]

Rumlik, 2023-12-12:
Hi there,
Since the last Z2M update, right and left button press on ZNQBKG25LM - 2 gang logging no longer works in operation mode. It only logs as a single press without indicating whether right or left.
Can someone please help with a solution?
Thank you very much.

Its-Manzzi, 2024-01-01:
Hey, did you manage to resolve?

Rumlik, 2024-01-01:
No

Its-Manzzi, 2024-01-01:
So you aren't able to use the switch effecticely in HA?

Rumlik, 2024-01-01:
> So you aren't able to use the switch effecticely in HA? 

Now i cant with decouple status. Only relay switch.
Nobody seems to care.

Its-Manzzi, 2024-01-01:
@osamaabdo Aslm, hoping you can assist here, you were really helpful in solving the previous issue. It would be appreciated if possible. جزاك الله خير

utegental, 2024-01-02:
> > @Rumlik did you make a PR with this fix?
> > Also, do you update your switches to the latest firmware?
> 
> Me? I thought it was said to @osamaabdo
> 
> I cant update firmware. I click update and nothing happens. You can update?

I tried, and it successfully updated, but it changed nothing :-)

mdworton, 2024-03-07:
I also have the same issue with the firmware. Says it has been successful but doesn't update. I have ZNQBKG26LM, ZNQBKG25LM, ZNQBKG24LM in my network but none can perform update. I have some older Aqara H1 with Neutrals and they work fine it just seems to be the H1M models that don't work. Maybe the firmware update could be related to the above issue and could fix that too. No obvious errors in the logs either. Happy to diagnose where i can if needed.

mdworton, 2024-03-07:
Just to add to this if you look under the device and select state it shows the new firmware as if it is loaded once you run the update, so in my case:
"update": {
        "installed_version": 1380147,
        "latest_version": 1380147,
but if you hit the check for firmware updates it reverts back to installed version as before the update:
"update": {
        "installed_version": 1376310,
        "latest_version": 1380147,

Its-Manzzi, 2024-03-07:
Still no fixes it seems:(

Its-Manzzi, 2025-06-15:
Any update on this please?

Its-Manzzi, 2025-09-22:
Any update? Has anybody tried adding their hub via matter to HA and controlling the devices through that route?