Koenkk/zigbee2mqtt #18493: version 1.32.2-1: can not start zigbee2mqtt addon due to error "friendly_name is not allowed to end with: '/l23"

bartplessers, 2023-08-01:
### What happened?

when starting the addon, I receive following message
```
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
            READ THIS CAREFULLY
Refusing to start because configuration is not valid, found the following errors:
- friendly_name is not allowed to end with: '/l23'
- friendly_name is not allowed to end with: '/l24'
If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```

### What did you expect to happen?

normal start, this did not happen with previous version of Z2M

### How to reproduce it (minimal and precise)

This is (a part of) devices.yaml

```
'0x842e14fffe1f0f71':
  friendly_name: my/L23
  description: |-
    marokkaans sfeerlampje
    00.01 sfeerlampje living (marokko)
    02.01 lampje staat nu op kamer 2de verdieping
  disabled: false
  homeassistant: {}
  legacy: false
  optimistic: true
'0x842e14fffe2216e6':
  friendly_name: my/L24
  description: 00.01 living tafellampje

```

if I just add "/light" to the friendly_name, no problem anymore
```
'0x842e14fffe1f0f71':
  friendly_name: my/L23/light
  description: |-
    marokkaans sfeerlampje
    00.01 sfeerlampje living (marokko)
    02.01 lampje staat nu op kamer 2de verdieping
  disabled: false
  homeassistant: {}
  legacy: false
  optimistic: true
'0x842e14fffe2216e6':
  friendly_name: my/L24/light
  description: 00.01 living tafellampje

```

strange this is that I have other lights with same naming convention that do not cause any trouble
```
'0x14b457fffe4f7695':
  friendly_name: my/L07
  disabled: true
  homeassistant: {}
  legacy: false
  optimistic: true
'0x842e14fffe2c36fc':
  friendly_name: my/L28
  disabled: true
  homeassistant: {}
  legacy: false
  optimistic: true

```

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

CC2652R_coordinator_20221226

### Adapter

CC2652 (Electrolama zig-a-zig-ah! (zzh!))

antsmartti, 2023-08-02:
Similar issue starting from this morning. Log is the following: 
```
[12:33:15] INFO: Preparing to start... [12:33:15] INFO: Socat not enabled [12:33:15] INFO: Zigbee Herdsman debug logging enabled [12:33:16] INFO: Starting Zigbee2MQTT... !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! READ THIS CAREFULLY Refusing to start because configuration is not valid, found the following errors: - Following friendly_name are not allowed: 'left,right,center,bottom_left,bottom_right,default,top_left,top_right,white,rgb,cct,system,top,bottom,center_left,center_right,ep1,ep2,row_1,row_2,row_3,row_4,relay,usb,l1,l2,l3,l4,l5,l6,l7,l8,l9,l10,l11,l12,l13,l14,l15,l16,l17,l18,l19,l20,l21,l22,l23,l24,th1,th2,th3,th4,th5,th6,th7,th8,th9,th10,button_1,button_2,button_3,button_4,button_5,button_6,button_7,button_8,button_9,button_10,button_11,button_12,button_13,button_14,button_15,button_16,button_17,button_18,button_19,button_20,button_light,button_fan_high,button_fan_med,button_fan_low,heat,cool,water,meter,wifi,no_occupancy_since' - Following friendly_name are not allowed: 'left,right,center,bottom_left,bottom_right,default,top_left,top_right,white,rgb,cct,system,top,bottom,center_left,center_right,ep1,ep2,row_1,row_2,row_3,row_4,relay,usb,l1,l2,l3,l4,l5,l6,l7,l8,l9,l10,l11,l12,l13,l14,l15,l16,l17,l18,l19,l20,l21,l22,l23,l24,th1,th2,th3,th4,th5,th6,th7,th8,th9,th10,button_1,button_2,button_3,button_4,button_5,button_6,button_7,button_8,button_9,button_10,button_11,button_12,button_13,button_14,button_15,button_16,button_17,button_18,button_19,button_20,button_light,button_fan_high,button_fan_med,button_fan_low,heat,cool,water,meter,wifi,no_occupancy_since' - Following friendly_name are not allowed: 'left,right,center,bottom_left,bottom_right,default,top_left,top_right,white,rgb,cct,system,top,bottom,center_left,center_right,ep1,ep2,row_1,row_2,row_3,row_4,relay,usb,l1,l2,l3,l4,l5,l6,l7,l8,l9,l10,l11,l12,l13,l14,l15,l16,l17,l18,l19,l20,l21,l22,l23,l24,th1,th2,th3,th4,th5,th6,th7,th8,th9,th10,button_1,button_2,button_3,button_4,button_5,button_6,button_7,button_8,button_9,button_10,button_11,button_12,button_13,button_14,button_15,button_16,button_17,button_18,button_19,button_20,button_light,button_fan_high,button_fan_med,button_fan_low,heat,cool,water,meter,wifi,no_occupancy_since' - Following friendly_name are not allowed: 'left,right,center,bottom_left,bottom_right,default,top_left,top_right,white,rgb,cct,system,top,bottom,center_left,center_right,ep1,ep2,row_1,row_2,row_3,row_4,relay,usb,l1,l2,l3,l4,l5,l6,l7,l8,l9,l10,l11,l12,l13,l14,l15,l16,l17,l18,l19,l20,l21,l22,l23,l24,th1,th2,th3,th4,th5,th6,th7,th8,th9,th10,button_1,button_2,button_3,button_4,button_5,button_6,button_7,button_8,button_9,button_10,button_11,button_12,button_13,button_14,button_15,button_16,button_17,button_18,button_19,button_20,button_light,button_fan_high,button_fan_med,button_fan_low,heat,cool,water,meter,wifi,no_occupancy_since' If you don't know how to solve this, read https://www.zigbee2mqtt.io/guide/configuration !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
```

Koenkk, 2023-08-02:
The eps are indeed not allowed anymore in 1.32.2. `l1` - `l17` wasn't allowed already, I didn't expect `l18` - `l24` would break for anyone. So please update your friendly names.

bartplessers, 2023-08-02:
Hi @Koenkk ,
Thanx for your answer.

Can you elaborate this a bit? What do you mean with 'eps'?

And why are there some *reserved* (friendly)names? I thought that the "friendly name" was a possibility to assign, well .... a friendly name without much limitations. 

And what if the IEEE address accidentally ends with "l24" and you didn't change the friendly name, will it work?

It's not very transparent to me...

It's not that big deal, but the most of my lights do have this name convention: "my/Lxx",and they all work, except for the above mentioned. I.e. L06 and L07 still work, even with this version of Z2M. 

So if it could be possible to allow any reasonable txt for friendly name (or at least lxx), I would be grateful 

Kind regards,

Bart

Koenkk, 2023-08-03:
@bartplessers because of ambiguity friendly names are not allowed to end with [`/ANY_OF_EP_NAMES`](https://github.com/Koenkk/zigbee2mqtt/blob/1ec1e57244834cc6b9dea6dbd3612b01cd21fa65/lib/util/utils.ts#L32). `l06` works indeed, but `l6` doesn't.

Koenkk, 2023-08-04:
ep = endpoint, this is applicable for devices with multiple endpoints (e.g. a multi gang switch)

closed 2023-09-11 by github-actions[bot]