Koenkk/zigbee2mqtt #18494: Tuya fingerbot plus _TZ3210_j4pdtz9v recognized as standard switch

bbcorp, 2023-08-02:
### What happened?

I've bought a Tuya Fingerbot and it got recognized only as a switch. Not all parameters are here, only state, 
power_on_behavior and linkquality
I've also tried with z2m Edge 1.32.1-dev commit: ba0cc61

Details:
Device type
EndDevice
Zigbee Model
TS0001
Zigbee Manufacturer
_TZ3210_j4pdtz9v
Description
1 gang switch
Support status
Supported

IEEE Address
0xa4c138314067ddc3
Network address
0x6041
Manufacturer
TuYa
Model
TS0001

### What did you expect to happen?

It should be recognized as [TS0001_fingerbot](https://www.zigbee2mqtt.io/device/TS0001_fingerbot.html) with other options like switch mode, upper and lower position.

### How to reproduce it (minimal and precise)

Include a Tuya Fingerbot plus _TZ3210_j4pdtz9v in z2m

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20220219

### Adapter

sonoff ZigBee 3.0 cc2652p

bbcorp, 2023-08-02:
Feel free if you need anything more

Koenkk, 2023-08-02:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-02 by Koenkk