Koenkk/zigbee2mqtt #18498: Legrand netatmo 600086 ( 067776 on Z2M) reports 'OPEN" when closed and vice versa #15101

stefanuytterhoeven, 2023-08-02:
### What happened?

Like in #15101 , the error comes back.
Please have a look at it ?
according Zigbee2MQTT:
Brand : Legrand netatmo , model: 067776_inverted

### What did you expect to happen?

right status op the cover

### How to reproduce it (minimal and precise)

always : it reports 'OPEN" when closed and vice versa

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20220219

### Adapter

Sonoff zigbee dongle P

Koenkk, 2023-08-02:
- Can you provide the data/databse.db entry of your device?
- What z2m version were you running previously?

stefanuytterhoeven, 2023-08-03:
Thanks for your quick reply.

Yesterday evening I upgraded to ha 2023.8. I got those messages about the naming of mqtt entities. I tried to restore my latest full backup (from a day before), but it failed all evening. 
This morning started again trying to restore the backup. Finally, at some moment, I succeeded to restore the backup using the cli method. But, serveral addons and integrations were not restored. So, had some work to do. At the moment all seems (kind of) working.

dababase.db entry: (hope I selected enough)
"manufId":4129,"manufName":" Legrand\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","powerSource":"Mains (single phase)","modelId":" Shutter switch with neutral\u0000\u0000\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":514,"inClusterList":[0,3,258,4,5,15,64513],"outClusterList":[0,64513,25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"zclVersion":2}},"genIdentify":{"attributes":{"identifyTime":60}},"genBinaryInput":{"attributes":{"presentValue":0}},"closuresWindowCovering":{"attributes":{"currentPositionLiftPercentage":0,"tuyaMovingState":0,"moesCalibrationTime":0,"currentPositionTiltPercentage":0}}},"binds":[{"cluster":15,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9b4af","endpointID":1},{"cluster":258,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9b4af","endpointID":1},{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0024c9b4af","endpointID":1}],"configuredReportings":[{"cluster":258,"attrId":8,"minRepIntval":1,"maxRepIntval":62000,"repChange":1}],"meta":{}}},"appVersion":0,"stackVersion":65,"hwVersion":1,"dateCode":" \u000020191105\u0000\u0000\u0000\u0000\u0000","swBuildId":"001a","zclVersion":2,"interviewCompleted":true,"meta":{"configured":-904845555},"lastSeen":1691053574456,"defaultSendRequestWhen":"immediate"}
{"id":6,"type":"EndDevice","ieeeAddr":"0x001788010bce52d7","nwkAddr":31351,

the z2M version previously: Can't tell the exact version. It was the latest version available, before the current version. Hope this is enough info for you. :)

Thanks for your time !

ka0r5, 2023-08-08:
Hi, I'm currently observing similar issue. Since release 1.32.2 the OTA update service is able to push a firmware update for Legrand 067776.
I have 4 devices Legrand 067776, before the update the firmware version was 001a and all of them were identified by zigbe2mqtt as 067776_inverted (despite nothing was inverted in the use, status OPEN and CLOSE were accurate)
I proceed to the update for only two devices, the new firmware version is 0037. The devices were still identified as 067776_inverted, but that time the status OPEN and CLOSE were inverted, and changing the invert_cover parameter had no effect at all. 
On the day after, while my window covers were already opened, I clicked on OPEN and the status "miraculously" switched from CLOSE to OPEN. I did the same on the second device that were updated to 0037 and the same thing happened. Also I noticed that the two of them are now identified as 067776 (and not 067776_inverted).
So I decided to update the 2 other devices, but so far (it has been 2 days) they are still identified as 067776_inverted and the status OPEN/CLOSE are actually inverted.

stefanuytterhoeven, 2023-08-08:
Exactly what happened here

stefanuytterhoeven, 2023-08-09:
Re-pair? 
It was working before. After the latest update, it didn't work anymore. 
Hopefully someone can test. I need to break out the switch, to repair.... I'll wait. Sorry

Koenkk, 2023-08-09:
Did you do an OTA update of the device?

Koenkk, 2023-08-09:
What if in the z2m frontend, click on device, go to dev console and read de swbuildid:

[image]

Can you post a screenshot of this?

stefanuytterhoeven, 2023-08-09:
So I went to the DEV console.
at first nothing is selected (Cluster)
So, from the screenshot you sent, I selected genBasic.
But, nothing happens. The read and write buttons are not active
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]
[image: image.png]

instellingen tab geeft niks speciaal

nu verder:
[image: image.png]
[image: image.png]
BTW, momenteel is het rolluik (fysiek) OPEN.

[image: image.png]

Voila, nu heb je het meeste.
Now, you have almost everything :)

stefan

stefanuytterhoeven, 2023-08-09:
ok, when selecting tab devconsole; I can select genBasic. Nothing happened. So, was looking around. Seems that I have to select "swBuildID" in the "attributes" ?
Well tried that too now. after choosing swBuildID, the attribute Box is cleared. Only difference is now, is , that Read and Write are active.
After pressing Read I get:
Info 2023-08-09 20:19:4 5Read result of 'genBasic': {"swBuildId":"0037"}
Guess that's what you were looking for :)

Koenkk, 2023-08-09:
what if you now restart z2m, does it detect it as `067776` instead of `067776_inverted`?

stefanuytterhoeven, 2023-08-09:
Restarted:
while starting: in the log i see this;
Warning 2023-08-09 20:26:00Firmware on '0x0004740000a167b4' is newer than latest firmware online. (no device is mentioned, but it's new to me)

And directly after that:
Info 2023-08-09 20:26:00MQTT publish: topic 'zigbee2mqtt/Woonkamer Rolluik', payload '{"led_in_dark":null,"linkquality":105,"position":0,"state":"CLOSE","tilt":0,"update":{"installed_version":3621887,"latest_version":3621379,"state":"idle"},"update_available":null}'

at little later:
Info 2023-08-09 20:26:46MQTT publish: topic 'zigbee2mqtt/Woonkamer Rolluik', payload '{"led_in_dark":null,"linkquality":109,"position":0,"state":"CLOSE","tilt":100,"update":{"installed_version":3621887,"latest_version":3621379,"state":"idle"},"update_available":null}'

ka0r5, 2023-08-09:
Hello, on my side everything is now ok also for the two remaining devices that were previously recognized as 067776_inverted. I just restarted zigbee2mqtt and they appeared as 067776 (status OPEN/CLOSE is now correct)

stefanuytterhoeven, 2023-08-09:
@ka0r5 you only restarted Z2M? I did a full restart of HA (the option that also restarts all addons)., but that didn't help. 
Aparently, a full restart of HA and it's addons, doesn't restart z2m the right way in this case. A restart in Z2M itself is needed?

anyway, problem solved. Now I know a few things to try, before turning to @koenkk :)

Thanks all of you!

stefanuytterhoeven, 2023-08-10:
@ka0r5 as said before: it looks like a compleet restart of HA was not enough. Probably the "state" of z2m is "saved". After restarting zigbee2mqtt (from inside the Z2M webpage) seems to do some "refresh" and made it working ok again.
Glad it's ok now. Was annoying  :)

Koenkk, 2023-08-10:
Normally a simple restart should be enough as z2m reads the swBuildId after update (but this can sometime fail if z2m can't connect to the device after ota update). 

Assuming I can close this issue now.

closed 2023-08-10 by Koenkk