Koenkk/zigbee2mqtt #18499: Light Entities Fail - 1.32.2-dev commit: f619610

beasthouse-au, 2023-08-02:
### What happened?

Updated to `1.32.2-dev commit: f619610` and all light entities stopped reporting to Home Assistant with the error:

`This entity is no longer being provided by the mqtt integration. If the entity is no longer in use, delete it in settings.` 

All other entities worked fine, such as sensor and switch entities. 

Rolling back to 1.32.1 commit: 4ccfdfc resolved the issue.

### What did you expect to happen?

Light entities to be exposed to Home Assistant

### How to reproduce it (minimal and precise)

Reinstall 1.32.2-dev produced the same error.

### Zigbee2MQTT version

1.32.2-dev commit: f619610

### Adapter firmware version

20210120

### Adapter

Electrolama

closed 2023-09-09 by github-actions[bot]

Sommerzeit, 2023-09-19:
I just had the exact same issue with 1.33.0 running in Docker after migrating from 1.29.0 (or so?) on a native Raspbian installation. First everything was working fine (or I did not see the lights beeing unavailable because I was testing a new presence detector which was not supported in 1.29). But ever since restarting Home Assistant it became obvious that the lights are not available. Downgrading to 1.32.1 fixed the problem. Thanks to beasthouse for posting this issue, it was a huge help :)

beasthouse-au, 2023-09-19:
A shame nothing was done about it!