Koenkk/zigbee2mqtt #18502: Home Assistant 2023.8.0 introduced some naming change requirements for MQTT entitites

xconverge, 2023-08-02:
### What happened?

"Discovered entities with a name that starts with the device name"

[image]

I think this is the relevant change, will edit it if this isn't right: https://github.com/home-assistant/core/pull/95159

Apologies if this is not a zigbee2mqtt issue I will close it if so, thanks

### What did you expect to happen?

No warning

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

7.3.1.0 build 176

### Adapter

skyconnect

olafz, 2023-08-02:
> I think this is the relevant change, will edit it if this isn't right: https://github.com/home-assistant/core/pull/95159

I think it is. I run into the same issue, and the [changelog for version 2023.8.0](https://www.home-assistant.io/blog/2023/08/02/release-20238/) refers to PR mentioned before and the following explanation:

[image]

xconverge, 2023-08-02:
Closing now that dupe was identified

closed 2023-08-02 by xconverge

KaportsevIA, 2023-08-17:
I have Zigbee2MQTT and 2 gateway SLS (slsys.io) in Direct2MQTT Mode (without Z2M). Devices which connected to SLS-MQTT nothave problem with Naming. Only one device connected to Z2M - have problem

> Обнаруженные объекты с именем, начинающимся с имени устройства
Это приведет к неисправностям в версии 2024.2.0. Пожалуйста, устраните эту проблему перед обновлением.
Некоторые объекты MQTT имеют имя, начинающееся с имени устройства. Это не предусмотрено. Во избежание задвоения имени префикс имени устройства удаляется из имени объекта. Пожалуйста, сообщите разработчику программного приложения, поставляющего соответствующие объекты, о необходимости устранения этой проблемы.

Список затронутых объектов:

sensor.0x00158d0006ba8b3e_power_outage_count
switch.0x00158d0006ba8b3e_l1
switch.0x00158d0006ba8b3e_interlock
switch.0x00158d0006ba8b3e_power_outage_memory
sensor.0x00158d0006ba8b3e_device_temperature
sensor.0x00158d0006ba8b3e_current
switch.0x00158d0006ba8b3e_l2
sensor.0x00158d0006ba8b3e_energy
sensor.0x00158d0006ba8b3e_power
sensor.0x00158d0006ba8b3e_voltage