Koenkk/zigbee2mqtt #18503: problem with bind Ikea Dimmer to a Group or device

wwyyrrmm, 2023-08-02:
### What happened?

1. a Ikea tradfri E1743 dimmer switch is working with zigbee2mqtt  Firmware version 2.2.010 from 20190723
2. a Ikea tradfri 30 W driver is working with zigbee2mqtt     ICPSHC24-30EU-IL-1 Firmware version 1.0.002 from 20181203 the driver is brand new (01082023). OTA does not work.

I select the E1742 device change to tab "bind" select the "endpoint" 1 select the  ICPSHC24-30EU-IL-1 as destination and 1 as destination endpoint, chek OnOff. Wake up the E1743 and bind it. SUCCESS

But it does not work.
On the dashboard the dimmer action is visible, the ICPSHC24-30EU-IL-1 can be switched on or off in the dashboard,
But the bind does not work.

### What did you expect to happen?

I expected, that the dimmer switch can switch the driver on or off

### How to reproduce it (minimal and precise)

I select the E1742 device change to tab "bind" select the "endpoint" 1 select the  ICPSHC24-30EU-IL-1 as destination and 1 as destination endpoint, chek OnOff. Wake up the E1743 and bind it. SUCCESS

But it does not work.
On the dashboard the dimmer action is visible, the ICPSHC24-30EU-IL-1 can be switched on or off in the dashboard,
But the bind does not work.

### Zigbee2MQTT version

    1.32.2

### Adapter firmware version

    20221226

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus,TI CC2652P + CP2102(N) 

### Debug log

no errors shown after acting the dimmer

closed 2023-09-10 by github-actions[bot]

mistergyle, 2024-01-20:
same issue here...

FilippoGhisio, 2024-09-18:
Same for me, any update?