Koenkk/zigbee2mqtt #18507: Can't remove device that failed to interview

sining951, 2023-08-03:
### What happened?

I was using the webgui adding devices (ikea bulbs) to my network and I had a weird error where a device that wasn't supposed to be added got pinged somehow by my mesh, failed to interview and now I can't remove it. 

When I try to do so, I get this error

2023-08-03 14:55:50Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device '0x001c55000001d0d8' (block: false, force: false) (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)'

How do I solve this? I just flashed my firmware with the latest one but I'm still getting this error

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff zigbee dongle plus

ssnaveen10, 2023-08-31:
I also have the same issue with one of my Aqara p1 motion sensors. When I try to remove the following error:  Request 'zigbee2mqtt/bridge/request/device/remove' failed with error: 'Failed to remove device 'MS1' (block: false, force: false) (Error: SRSP - ZDO - mgmtLeaveReq after 6000ms)'

Version running is : [1.32.2] commit: [1ec1e57]

czachor, 2023-10-08:
Same here with Sonoff SNZB-01.

Koenkk, 2023-10-09:
Force remove it instead

closed 2024-05-07 by github-actions[bot]