Koenkk/zigbee2mqtt #18508: Cant add Nimly Connect module. / easyCodeTouch_v1

bergetun, 2023-08-03:
### What happened?

Have used several of these zigbee modules for Nimly smart door locks earlier in z2m with no problem. 
The new one i just bought doesnt seem to work. The only difference is that this one has a newer firmware: 20230210

Log entry : 
 Warning 2023-08-03 09:14:48Device '0xf4ce368708ef39b2' with Zigbee model 'NimlyIn' and manufacturer name 'Onesti Products AS' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

### What did you expect to happen?

To be paired

### How to reproduce it (minimal and precise)

Update the module to the latest firmware and pair

### Zigbee2MQTT version

 1.28.0-dev commit: a426ee1

### Adapter firmware version

0x26720700

### Adapter

ConBee2/RaspBee2

bergetun, 2023-08-03:
The module indentify as NimlyIn and not NimlyPRO in z2m when installed in a Nimly Indoor lock.    Functionality is the same on  Nimly indoor and Nimly Touch Pro, so it shouldnt matter.

Kaptensanders, 2023-08-29:
Same problem, also firmware 20230210

closed 2023-10-07 by github-actions[bot]