Koenkk/zigbee2mqtt #18509: [New device support]: add feature to Bosch BSD-2 smoke alarm detector to use it as burglar alarm

burpy, 2023-08-03:
### Link

https://www.zigbee2mqtt.io/devices/BSD-2.html

### Database entry

{"id":29,"type":"EndDevice","ieeeAddr":"0x04cd15fffefe8b0b","nwkAddr":28488,"manufId":4617,"manufName":"Bosch","powerSource":"Battery","modelId":"RBSH-SD-ZB-EU","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,1,3,32,1280,2821],"outClusterList":[25],"clusters":{"ssIasZone":{"attributes":{"iasCieAddr":"0xe0798dfffeb7883a","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb7883a","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb7883a","endpointID":1},{"cluster":64684,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffeb7883a","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":62000,"repChange":0}],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":1,"dateCode":"20220222","swBuildId":"1.10","zclVersion":8,"interviewCompleted":true,"meta":{"configured":1394888596},"lastSeen":1691049819635,"defaultSendRequestWhen":"active","checkinInterval":1620}

### Comments

This smoke detector already works fine with Z2M 1.32.2-dev but does not expose the alarm as a switch.
Via the Bosch gateway it is possible to e.g. trigger the smoke alarm when a door is opened or similar triggers are fired.

Can this attribute be made available as a switch also in Z2M?

N3rdix, 2023-08-05:
Requested this already a while ago in #17229, would love to get this working

closed 2023-09-06 by burpy

vjonas, 2024-01-15:
Merged in master. You can pull the the branch or wait for the docker release.
Can be closed.