Koenkk/zigbee2mqtt #18510: MQTT device naming issues with home assisstant

thomasdelaet, 2023-08-03:
### What happened?

With the latest release of home assistant (2023.8), I'm getting a lot of warnings around MQTT entity naming:
- "Discovered MQTT entities with a name that is equal to the device name" => for all my lights
- "Discovered entities with a name that starts with the device name" => for all _update_available, an _power_on_behavior entities.

Suggested fix: just add _device as suffix to the device name field in MQTT config entry.

wdyt?

### How to reproduce it (minimal and precise)

Upgrade to home assistant 2023.8 and check repairs list (and log)

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

not relevant for this issue

### Adapter

not relevant for this issue

stefanuytterhoeven, 2023-08-03:
From what I read on the HA website, you can ignore these messages. In september there should be a solution. (correct me if i'm wrong :) )

Gorouk, 2023-08-03:
https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

juahan, 2023-08-03:
It is known and looks like it will be fixed in https://github.com/Koenkk/zigbee2mqtt/pull/18490.

Koenkk, 2023-08-03:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18445

closed 2023-08-03 by Koenkk