Koenkk/zigbee2mqtt #18511: Some PowerTag sensors renamed to "MQTT Sensor"

tkarlsson, 2023-08-03:
### What happened?

Some of the PowerTag sensors (specifically it looks like energy sensors) have been renamed to the generic "MQTT Sensor" in Home Assistant and not containing the PowerTag device name making them indistinguishable in HA.

### What did you expect to happen?

Either be named a long the lines "power phase a" as before.

### Zigbee2MQTT version

1.32.2-dev commit

### Adapter firmware version

20221226

### Adapter

Sonoff

tkarlsson, 2023-08-03:
Seems to have fixed itself by updating to latest dev and latest stable HA version

closed 2023-08-03 by tkarlsson

senna1992, 2023-11-10:
please post how you got them working! =)