Koenkk/zigbee2mqtt #18513: TS0201 humidity is factor 10 smaller

Miesje, 2023-08-03:
### What happened?

This is a know problem and should be solved in the prevouis version but for me it is not working.

The problem does exist in the new version 1.32.2 as well.
I have bought 6 new devices TS0201 from factorynumber _TZ3210_ncw88jfq.
Al looks great exept the humidity, that is a factor 10 smaller.

Log:
Debug 2023-08-03 19:41:56Received Zigbee message from 'ZT02 - Zolder', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":605}' from endpoint 1 with groupID 0
Info 2023-08-03 19:41:56MQTT publish: topic 'zigbee/ZT02 - Zolder', payload '{"battery":100,"humidity":6.05,"linkquality":180,"temperature":24.8,"voltage":3000}'
Debug 2023-08-03 19:42:14Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2440}' from endpoint 1 with groupID 0
Info 2023-08-03 19:42:14MQTT publish: topic 'zigbee/ZT01 - Serre', payload '{"battery":100,"humidity":6.24,"linkquality":172,"temperature":24.4,"voltage":3000}'

[image]

Is it possible to solve the problem.

With kind regards,
Miesje

### What did you expect to happen?

Should be increased by 10 times 

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

Sonoff 

### Debug log

2023-08-03 19:20:25Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
2023-08-03 19:20:25Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":626}' from endpoint 1 with groupID 0
2023-08-03 19:20:25Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2440}' from endpoint 1 with groupID 0
2023-08-03 19:21:40Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
2023-08-03 19:21:40Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":628}' from endpoint 1 with groupID 0
2023-08-03 19:21:41Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2490}' from endpoint 1 with groupID 0

Koenkk, 2023-08-03:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-03 by Koenkk

Miesje, 2023-08-05:
Dear Koenkk,

Thanks very much for the quick response.
I do have a additional question regading the dev branch.
My Zigbee is changed to the develepment branch do I need to implement the zigbee-herdsman converter as well.
Unfortunately only with the latest-dev it is still not working.

1.32.2-dev commit: [f08cd81]
Debug 2023-08-05 07:53:20Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":580}' from endpoint 1 with groupID 0
Debug 2023-08-05 07:53:21Received Zigbee message from 'ZT01 - Serre', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2280}' from endpoint 1 with groupID 0
Info 2023-08-05 07:53:21MQTT publish: topic 'zigbee/ZT01 - Serre', payload '{"battery":100,"humidity"**:6.24,**"linkquality":252,"temperature":22.9,"voltage":3000}'

vxav, 2023-08-17:
Any chance you could cut a release to bump the converter so it includes this fix?