Koenkk/zigbee2mqtt #18515: Can't update Ikea Styrbar 1.0.024 to latest version

crispy78, 2023-08-03:
### What happened?

When using the OTA functionality Zigbee2MQTT concludes that the version 1.0.024 is newer than the firmware available online 2.4.5, so it won't update the device. 

I've tried forcing the update, but that didn't work. To accomplish that I've renamed the index.json from pull request https://github.com/Koenkk/zigbee-OTA/pull/340 to my_index.json and put it in \config\zigbee2mqtt in Home Assistant and entered /config/zigbee2mqtt/my_index.json in Zigbee2MQTT as the OTA index override file name.

### What did you expect to happen?

I would have liked the device to update to the latests firmware.

### How to reproduce it (minimal and precise)

Press "Check for new updates" on a Ikea Styrbar in the OTA-menu.

### Zigbee2MQTT version

1.32.2-dev commit: 68ec507

### Adapter firmware version

20220928

### Adapter

CC2652RB

### Debug log

info  2023-08-03 22:04:53: Checking if update available for 'Drukknop voordeur'
info  2023-08-03 22:04:53: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Checking if update available for 'Drukknop voordeur'","meta":{"device":"Drukknop voordeur","status":"checking_if_available"},"type":"ota_update"}'
debug 2023-08-03 22:04:53: Check if update available for '0x540f57fffe8f81f0' (Remote Control N2)
debug 2023-08-03 22:04:53: Using endpoint '1'
debug 2023-08-03 22:04:54: Received Zigbee message from 'Drukknop voordeur', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:54: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"off","battery":100,"last_seen":"2023-08-03T22:04:54+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:54: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"","battery":100,"last_seen":"2023-08-03T22:04:54+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:54: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur/action', payload 'off'
debug 2023-08-03 22:04:54: Received Zigbee message from 'Drukknop voordeur', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:54: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":null,"battery":100,"last_seen":"2023-08-03T22:04:54+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
debug 2023-08-03 22:04:56: Received Zigbee message from 'Drukknop voordeur', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:56: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"arrow_right_click","battery":100,"last_seen":"2023-08-03T22:04:56+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:56: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"","battery":100,"last_seen":"2023-08-03T22:04:56+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:56: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur/action', payload 'arrow_right_click'
debug 2023-08-03 22:04:56: Received Zigbee message from 'Drukknop voordeur', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:56: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":null,"battery":100,"last_seen":"2023-08-03T22:04:56+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
debug 2023-08-03 22:04:57: Received Zigbee message from 'Drukknop voordeur', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:57: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"arrow_left_click","battery":100,"last_seen":"2023-08-03T22:04:57+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:57: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"","battery":100,"last_seen":"2023-08-03T22:04:57+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:57: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur/action', payload 'arrow_left_click'
debug 2023-08-03 22:04:57: Received Zigbee message from 'Drukknop voordeur', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:57: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":null,"battery":100,"last_seen":"2023-08-03T22:04:57+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
debug 2023-08-03 22:04:57: Received Zigbee message from 'Drukknop voordeur', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":65572,"imageType":4555,"manufacturerCode":4476}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:57: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":null,"battery":100,"last_seen":"2023-08-03T22:04:57+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
debug 2023-08-03 22:04:57: Got OTA request '{"fieldControl":1,"manufacturerCode":4476,"imageType":4555,"fileVersion":65572}'
debug 2023-08-03 22:04:58: Received Zigbee message from 'Drukknop voordeur', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:58: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"on","battery":100,"last_seen":"2023-08-03T22:04:58+02:00","linkquality":32,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:58: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":"","battery":100,"last_seen":"2023-08-03T22:04:58+02:00","linkquality":32,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
info  2023-08-03 22:04:58: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur/action', payload 'on'
debug 2023-08-03 22:04:58: Received Zigbee message from 'Drukknop voordeur', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0
info  2023-08-03 22:04:58: MQTT publish: topic 'zigbee2mqtt/Drukknop voordeur', payload '{"action":null,"battery":100,"last_seen":"2023-08-03T22:04:58+02:00","linkquality":36,"update":{"installed_version":65572,"latest_version":581,"state":"idle"},"update_available":false}'
debug 2023-08-03 22:04:58: Is new image available for '0x540f57fffe8f81f0', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":4555,"fileVersion":65572}', latest meta '{"fileVersion":581,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed"}'
debug 2023-08-03 22:04:58: Update available for '0x540f57fffe8f81f0': NO
warn  2023-08-03 22:04:58: Firmware on '0x540f57fffe8f81f0' is newer than latest firmware online.
info  2023-08-03 22:04:58: No update available for 'Drukknop voordeur'

crispy78, 2023-08-04:
I've found someone having the same issues as me.
Someone has altered the header in the update: https://github.com/zigpy/zigpy/issues/1153#issuecomment-1416078425
I've published the file on my server, created a my_index.json with only that file:
```
[
    {
        "fileVersion": 33816581,
        "fileSize": 218698,
        "manufacturerCode": 4476,
        "imageType": 4555,
        "sha512": "0e94264bd853d41cc5ee7caa8336b310b60ea8503d56f05ccfb42cb9ec8b03a1aa0cdac27bb6314c3cf9abbd55c9862cc6127bea46abedb8067bea07ffb8a8b1",
        "url": "http://__myserver__/10078031-zingo_kt_styrbar_remote-2.4.5_modded_header.ota.ota.signed",
        "force": true
    }
]
```
I've entered `/config/zigbee2mqtt/my_index.json` in "OTA index override file name", I've also tried `my_index.json`, but it still fails to update.

malaakso, 2023-08-25:
I've also tried to force update Styrbar by downloading the file locally and using local index.json. It seems that zigbee2mqtt ignores local index.json for those devices which are found in online feed, like IKEA. @Koenkk can you confirm this?

Koenkk, 2023-08-25:
It's only used when `ota.zigbeeOTA` is used (for this device `ota.tradfri` is used). You can override it with an ext converter: https://gist.github.com/Koenkk/fafd62c76aee9d331f70aa4ff4cf12fd

malaakso, 2023-08-25:
Thanks! I was able to update to 2.4.5 with the external converter.

I opened a pull request to get this information to the official docs.

Koenkk, 2023-08-26:
Great, assuming this can be closed now.

closed 2023-08-26 by Koenkk

crispy78, 2023-08-26:
I've tried it in several ways and couldn't get it to work. I've created the .js-file, I've changed the config, I've restarted Z2M from within Home Assistant numerous times and I can't get the button to update. I've added my my_index.json to the config and then I got the message that 1.0.024 is newer than 2.4.5.

malaakso, 2023-08-27:
@crispy78 this is the index.json I used for successful OTA (I downloaded the file from IKEA feed to the /app/data directory):
`[
    {
        "manufacturerCode": 4476,
        "imageType": 4555,
        "fileVersion": 581,
        "url": "10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed",
        "force": true
    }
]`

crispy78, 2023-08-27:
I'm about to quit this struggle. I can't find the app/data directory, so I've placed the file in the Zigbee2MQTT config-directory. I've changed the my_index.json-file, changed the config-file and still no luck. I'm thinking of changing to ZHA temporarily, the device seems to be updating on that platform with less hassle. I might be wrong.

tishcufksips, 2023-08-27:
> I'm about to quit this struggle. I can't find the app/data directory, so I've placed the file in the Zigbee2MQTT config-directory. I've changed the my_index.json-file, changed the config-file and [...]

I've just managed to start the update, maybe I can help...
[image]

crispy78, 2023-08-27:
@tishcufksips thanks for your instructions, it finally got to the point that it found an update and it's currently updating.
For some reason the config wanted to use the test server and putting the my_index.json and update on my own server was a very helpfull comment.

jkl1337, 2023-10-01:
There is no need to run a server or anything like that. zigbee2mqtt is perfectly capable of serving the file. The firmware file and my_index.json can both go in the data directory, which is the same directory that zigbee2mqtt configuration.yaml is located.

brainhig, 2023-10-05:
Hi y'all:

I'm pretty sure I've done all the steps here correctly, and have tried both using a web server and putting the necessary files on the HA instance directly, but I'm getting errors (see below) that seem to point to another problem. The buttons are working correctly, so I'm not sure these errors really mean what they say. Any help or pointers is appreciated!

Info 2023-10-04 18:30:39MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Failed to check if update available for 'IKEAButton2' (Device didn't respond to OTA request)","meta":{"device":"IKEAButton2","status":"check_failed"},"type":"ota_update"}'
Info 2023-10-04 18:30:39MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/check', payload '{"data":{"id":"IKEAButton2"},"error":"Failed to check if update available for 'IKEAButton2' (Device didn't respond to OTA request)","status":"error","transaction":"9rsko-15"}'

malaakso, 2023-10-05:
Did you press button on the remote to wake it up after clicking Check for updates?

rbalaev, 2023-10-09:
After flashing the firmware, the buttons stopped working, action commands are no longer sent, 3 buttons were flashed and they all stopped working. What have I done wrong?

malaakso, 2023-10-09:
You have to remove the external converter and restart zigbee2mqtt to get all functionality back.

rbalaev, 2023-10-09:
> You have to remove the external converter and restart zigbee2mqtt to get all functionality back.

Thank you, everything returned after completely stopping the container and starting it, restarting did not help.

evilmumi, 2023-11-03:
Hi, i did the update from https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1742177146 and now only got device_announce messages when i insert battery.
Restart Z2M restart Docker container - no other information from device
Debug says: Retrieving state of "STYRBAR01' after reconnect.

DrLeurs, 2023-11-05:
> > You have to remove the external converter and restart zigbee2mqtt to get all functionality back.
> 
> Thank you, everything returned after completely stopping the container and starting it, restarting did not help.

I alas seem to experience the same problem, even after restarting z2m/the whole system: both of my buttons stopped working after flashing the new firmware. No action commands are sent from them and I also can't seem to remove them.

They do however seem to pair perfectly again when factory resetting them, but the result is the same: no functionality whatsoever.

evilmumi, 2023-11-06:
I think something worked here.
Reset them
delete from Zigbee2MQTT
Pair with a bulb
pair with Zigbee2MQTT
reset the bulb

DrLeurs, 2023-11-06:
> I think something worked here. Reset them delete from Zigbee2MQTT Pair with a bulb pair with Zigbee2MQTT reset the bulb

I was about to do this, when I suddenly realized on of the two buttons worked. The only thing I had been doing (repeatedly) was removing the device, pairing it again and rebooting my entire system, to no avail. After leaving the button many hours, it suddenly worked. The second button I had still didn't, but worked immediately after re-pairing it. I can't really pinpoint what suddenly changed...

MacJawa, 2023-11-07:
> > I think something worked here. Reset them delete from Zigbee2MQTT Pair with a bulb pair with Zigbee2MQTT reset the bulb
> 
> I was about to do this, when I suddenly realized one of the two buttons worked. The only thing I had been doing (repeatedly) was removing the device, pairing it again and rebooting my entire system, to no avail. After leaving the button many hours, it suddenly worked. The second button I had still didn't work, but worked immediately after re-pairing it. I can't really pinpoint what suddenly changed...

I've seen this behavior before. Mostly with IKEA shortcut button. It paired well, but sent no actions. Sometimes it started to work after multiple factory reset, sometimes I left it on the table and started to work after some hours by itself. 
Hope this will happen with Styrbar, as both of my remotes now do not communicate as well after firmware update

bub4, 2023-11-08:
same here - after the FW update it did not work anymore. Pairing was fine, just didn't trigger any action. I took out the batteries and replaced the remote with an old FW-level. After two days I wanted to do further tests and after inserting the batteries, the Styrbar worked just normal. So maybe it helps to take out the batteries over night...

thanks @jkl1337 for the above manual!!

hazemhelmy, 2023-11-09:
same issue , tried to pair, it pairs ok ( long process of interview but finally pairs) but no actions triggered , also You can force a update for the firmware by pairing with hub and letting it update and then unpairing , but still when i connect to HA there is no trigger or binds working

malaakso, 2023-11-09:
Since it is fixed (in container installation) by completely removing the container, it must be that the external converter gets somehow included in some other directory than data. Is it somehow cached or compiled in under files in dist?

Redominus, 2023-11-19:
> ...
> I still have to find the fix for this 1% brightness..

Same here :disappointed:

Redominus, 2023-11-23:
Thank you @SnakeZi ! It worked and fixed the brightness issue!

boozeman, 2023-12-02:
> There is no need to run a server or anything like that. zigbee2mqtt is perfectly capable of serving the file. The firmware file and my_index.json can both go in the data directory, which is the _same_ directory that zigbee2mqtt configuration.yaml is located. The steps that work are as follows:
> 
> 1. Copy https://gist.github.com/Koenkk/fafd62c76aee9d331f70aa4ff4cf12fd to ext_converter.js (in the zigbee2mqtt data directory).
> 2. Download http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed and place the file into the zigbee2mqtt data directory.
> 3. Add the following to a file called my_index.json in the data directory
> 
> ```json
> [ { "manufacturerCode": 4476,
> "imageType": 4555, 
> "fileVersion": 581,
> "url": "10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed",
> "force": true } ]
> ```
> 
> 4. Add/merge the following configuration items to configuration.yaml
> 
> ```yaml
> external_converters:
>   - ext_converter.js
> ota:
>   zigbee_ota_override_index_location: my_index.json
> ```
> 
> 5. Restart zigbee2mqtt and go through the OTA update process as normal. After completion should then remove the ext_converter.js from the configuration and restart.

My 2 cents with this instruction:

6. After removing the ext_converter.js from the configuration and Before restarting the Z2M, remove device from Z2M. After thar reboot Z2M and repair the remote. All buttons working.

After the successfull update the only problem I have  will be this:

`{
    "linkquality": 171,
    "update": {
        "installed_version": 33816581,
        "latest_version": 581,
        "state": "idle"
    },
    "update_available": false
}`

Installed Version has that whole Ikea number but latest_version is just last 3 digits --> Installed firmware is newer than available.

victorhooi, 2023-12-03:
I have several IKEA Styrbar remotes (E2002), which seemed to be stuck on version 1.0.024, and wouldn't update to the newer version 2.4.5.

I was able to follow the instructions in [the comment above](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1742177146) to update them.

One small issue I hit - at first, when I tried to do the update, it errored out like so:

[image]

If I check the docker logs for my zigbee2mqtt container, I see:

```
Zigbee2MQTT:info  2023-12-03 17:10:44: Updating 'Downstairs Bunk Bed Switch' to latest firmware
Zigbee2MQTT:info  2023-12-03 17:12:07: MQTT publish: topic 'zigbee2mqtt/Downstairs Bunk Bed Switch', payload '{"battery":80,"linkquality":25,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
Zigbee2MQTT:info  2023-12-03 17:12:07: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Downstairs Bunk Bed Switch"},"error":"Update of 'Downstairs Bunk Bed Switch' failed (Device didn't respond to OTA request)","status":"error","transaction":"ajzql-1"}'
Zigbee2MQTT:error 2023-12-03 17:12:07: Update of 'Downstairs Bunk Bed Switch' failed (Device didn't respond to OTA request)
```

However, I then saw the comment elsewhere about pressing the button to wake it up - to be honest, it had probably been less than a minute or so since I'd used the remote, but either way, I pressed a button on the remote, then immediately clicked the update button - it seemed to work after that - although it took around 3 hours to push the update...

[image]

It might be worth summarising all of the steps that are known to work, directly into the [zigbee2mqtt entry for the IKEA Styrbar remote](https://www.zigbee2mqtt.io/devices/E2001_E2002.html). (At the moment, it's just a link back to this issue - which means people have to wade through text). What do people think?

Also - is there a reason this update workaround can't be rolled directly into a zigbee2mqtt fix? Or is there some technical reason, that we have to do the workaround?

Finally, after I updated my remotes, I rolled back the changes from [the comment above](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1742177146) and restarted zigbee2mqtt. 

The Styrbar remotes didn't work anymore - so I unpaired them (four presses of the pairing button), and they automatically repaired, and then they worked after that.

There were some error messages about Bind and Timeouts - but it seemed to work after a while

[image]

Log entries are below:

```
Zigbee2MQTT:warn  2023-12-03 21:27:49: Device 'Upstairs Bunk Switch' left the network
Zigbee2MQTT:info  2023-12-03 21:27:49: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Upstairs Bunk Switch","ieee_address":"0x9035eafffe9d463f"},"type":"device_leave"}'
Zigbee2MQTT:info  2023-12-03 21:27:55: Device 'Upstairs Bunk Switch' joined
Zigbee2MQTT:info  2023-12-03 21:27:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Upstairs Bunk Switch","ieee_address":"0x9035eafffe9d463f"},"type":"device_joined"}'
Zigbee2MQTT:info  2023-12-03 21:27:55: Starting interview of 'Upstairs Bunk Switch'
Zigbee2MQTT:info  2023-12-03 21:27:55: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Upstairs Bunk Switch","ieee_address":"0x9035eafffe9d463f","status":"started"},"type":"device_interview"}'
Zigbee2MQTT:info  2023-12-03 21:27:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Upstairs Bunk Switch","ieee_address":"0x9035eafffe9d463f"},"type":"device_announce"}'
Zigbee2MQTT:info  2023-12-03 21:28:03: Successfully interviewed 'Upstairs Bunk Switch', device has successfully been paired
Zigbee2MQTT:info  2023-12-03 21:28:03: Device 'Upstairs Bunk Switch' is supported, identified as: IKEA STYRBAR remote control (E2001/E2002)
Zigbee2MQTT:info  2023-12-03 21:28:03: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"STYRBAR remote control","exposes":[{"access":5,"description":"Remaining battery in %, can take up to 24 hours before reported.","label":"Battery","name":"battery","property":"battery","type":"numeric","unit":"%","value_max":100,"value_min":0},{"access":1,"description":"Triggered action (e.g. a button click)","label":"Action","name":"action","property":"action","type":"enum","values":["on","off","brightness_move_up","brightness_move_down","brightness_stop","arrow_left_click","arrow_right_click","arrow_left_hold","arrow_right_hold","arrow_left_release","arrow_right_release"]},{"access":1,"description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"E2001/E2002","options":[{"access":2,"description":"Simulate a brightness value. If this device provides a brightness_move_up or brightness_move_down action it is possible to specify the update interval and delta. The action_brightness_delta indicates the delta for each interval. ","features":[{"access":2,"description":"Delta per interval, 20 by default","label":"Delta","name":"delta","property":"delta","type":"numeric","value_min":0},{"access":2,"description":"Interval duration","label":"Interval","name":"interval","property":"interval","type":"numeric","unit":"ms","value_min":0}],"label":"Simulated brightness","name":"simulated_brightness","property" [... cut ...]
    at listOnTimeout (node:internal/timers:569:17)
    at processTimers (node:internal/timers:512:7))
Zigbee2MQTT:info  2023-12-03 21:28:46: Configuring 'Upstairs Bunk Switch'
Zigbee2MQTT:warn  2023-12-03 21:28:47: Firmware on '0x9035eafffe9d463f' is newer than latest firmware online.
Zigbee2MQTT:info  2023-12-03 21:28:47: MQTT publish: topic 'zigbee2mqtt/Upstairs Bunk Switch', payload '{"action":null,"battery":75,"linkquality":153,"update":{"installed_version":33816581,"latest_version":581,"state":"idle"},"update_available":null}'
```

rajp1503, 2024-02-14:
> I have several IKEA Styrbar remotes (E2002), which seemed to be stuck on version 1.0.024, and wouldn't update to the newer version 2.4.5.
> 
> I was able to follow the instructions in [the comment above](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1742177146) to update them.
> 
> One small issue I hit - at first, when I tried to do the update, it errored out like so:
> <img alt="Screenshot 2023-12-03 at 17 12 26" width="349" src="https://private-user-images.githubusercontent.com/243499/287500383-3ed0f7a5-932d-4233-aa23-415168b78650.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5MjkwOTMsIm5iZiI6MTcwNzkyODc5MywicGF0aCI6Ii8yNDM0OTkvMjg3NTAwMzgzLTNlZDBmN2E1LTkzMmQtNDIzMy1hYTIzLTQxNTE2OGI3ODY1MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNFQxNjM5NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MDUyZDg3MTdlMGEyYjQ3NmEyYmM1MTE0YWJiYWEzMDFkOWQ5ZWQwZTg2ZTQ5ZWNlNTVlNDlkMzU3ZGMwOTE2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.3N_msRLsqTsoKGoaOnokQ4MS4OgaSoGNoN-JaoRDY-I">
> 
> If I check the docker logs for my zigbee2mqtt container, I see:
> 
> ```
> Zigbee2MQTT:info  2023-12-03 17:10:44: Updating 'Downstairs Bunk Bed Switch' to latest firmware
> Zigbee2MQTT:info  2023-12-03 17:12:07: MQTT publish: topic 'zigbee2mqtt/Downstairs Bunk Bed Switch', payload '{"battery":80,"linkquality":25,"update":{"installed_version":65572,"latest_version":581,"state":"available"},"update_available":null}'
> Zigbee2MQTT:info  2023-12-03 17:12:07: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/ota_update/update', payload '{"data":{"id":"Downstairs Bunk Bed Switch"},"error":"Update of 'Downstairs Bunk Bed Switch' failed (Device didn't respond to OTA request)","status":"error","transaction":"ajzql-1"}'
> Zigbee2MQTT:error 2023-12-03 17:12:07: Update of 'Downstairs Bunk Bed Switch' failed (Device didn't respond to OTA request)
> ```
> 
> However, I then saw the comment elsewhere about pressing the button to wake it up - to be honest, it had probably been less than a minute or so since I'd used the remote, but either way, I pressed a button on the remote, then immediately clicked the update button - it seemed to work after that - although it took around 3 hours to push the update...
> <img alt="Screenshot 2023-12-03 at 18 39 05" width="1708" src="https://private-user-images.githubusercontent.com/243499/287510679-ea023d5d-9bf1-43d6-bf02-48ebbfd56037.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5MjkwOTMsIm5iZiI6MTcwNzkyODc5MywicGF0aCI6Ii8yNDM0OTkvMjg3NTEwNjc5LWVhMDIzZDVkLTliZjEtNDNkNi1iZjAyLTQ4ZWJiZmQ1NjAzNy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNFQxNjM5NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MmRjNjI4ZGU3NzQ5N2JlMTMzNTNlNDFiNDk4MzFiMzFjMDBkNTZiOTVkMDZhNGU5MjBjOTBjMTYxYzc0MjYzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.4u6Tz_O5EfJQygzV1n9gUt9GVkGxMGFrf6FvWZ_JR84">
> 
> It might be worth summarising all of the steps that are known to work, directly into the [zigbee2mqtt entry for the IKEA Styrbar remote](https://www.zigbee2mqtt.io/devices/E2001_E2002.html). (At the moment, it's just a link back to this issue - which means people have to wade through text). What do people think?
> 
> Also - is there a reason this update workaround can't be rolled directly into a zigbee2mqtt fix? Or is there some technical reason, that we have to do the workaround?
> 
> Finally, after I updated my remotes, I rolled back the changes from [the comment above](https://github.com/Koenkk/zigbee2mqtt/issues/18515#issuecomment-1742177146) and restarted zigbee2mqtt.
> 
> The Styrbar remotes didn't work anymore - so I unpaired them (four presses of the pairing button), and they automatically repaired, and then they worked after that.
> 
> There were some error messages about Bind and Timeouts - but it seemed to work after a while
> <img alt="Screenshot 2023-12-03 at 21 28 17" width="352" src="https://private-user-images.githubusercontent.com/243499/287513758-af2c674b-46d4-43b7-acc6-f63cb02d358b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5MjkwOTMsIm5iZiI6MTcwNzkyODc5MywicGF0aCI6Ii8yNDM0OTkvMjg3NTEzNzU4LWFmMmM2NzRiLTQ2ZDQtNDNiNy1hY2M2LWY2M2NiMDJkMzU4Yi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIxNFQxNjM5NTNaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jM2UwZDk3MDgyYWU0MzQ0MTE0MTY4YzMyOGRlZGY0YWE0OGE4Zjc4YjYxNWQyMzg2N2QxMDdmZjI2ZjYzN2ZjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.FN9mvjWLPYzTqmIG080sYW-E-pd3aPqyHja6BzkO8UM">
> 
> Log entries are below:
> 
> ```
> Zigbee2MQTT:warn  2023-12-03 21:27:49: Device 'Upstairs Bunk Switch' left the network
[... cut ...]
> Zigbee2MQTT:info  2023-12-03 21:28:46: Configuring 'Upstairs Bunk Switch'
> Zigbee2MQTT:warn  2023-12-03 21:28:47: Firmware on '0x9035eafffe9d463f' is newer than latest firmware online.
> Zigbee2MQTT:info  2023-12-03 21:28:47: MQTT publish: topic 'zigbee2mqtt/Upstairs Bunk Switch', payload '{"action":null,"battery":75,"linkquality":153,"update":{"installed_version":33816581,"latest_version":581,"state":"idle"},"update_available":null}'
> ```

Thanks for this.. I was able to get the update started.. it shows approx 2 hours to complete.. did yours take this long???

MilesTEG1, 2024-03-12:
> There is no need to run a server or anything like that. zigbee2mqtt is perfectly capable of serving the file. The firmware file and my_index.json can both go in the data directory, which is the [...]

Hello,
This method is very usefull, and should be explained in plain in the product page on Z2M doc.

Can someone explain what did all the modifications?
I'm not a developer and very new to Home Assistant and Z2M, so I don't understand very much what does all this modifications.

However, thank you for your answer.

HotelAlpha4, 2024-04-07:
Is this no longer working in 2024? I have tried all of the above steps, attempting to edit the json files, also using the GUI to change config, restarting zigbee2mqtt many many times, even restarting HA itself in case that did it. The js file is there, and I even changed it to execute permissions in case that was missed in the instructions. I have tried referencing all files with /homeassistant/zigbee2mqtt/... paths, and tried /config/zigbee2mqtt/.... paths, and just relative paths. The firmware patch file is in the zigbee2mqtt folder.  Maybe this does not work any more? Is there a better way to update firmware? Anything I should check? 

Thank you

boozeman, 2024-04-07:
> Is this no longer working in 2024? I have tried all of the above steps, attempting to edit the json files, also using the GUI to change config, restarting zigbee2mqtt many many times, even restarting HA itself in case that did it. The js file is there, and I even changed it to execute permissions in case that was missed in the instructions. I have tried referencing all files with /homeassistant/zigbee2mqtt/... paths, and tried /config/zigbee2mqtt/.... paths, and just relative paths. The firmware patch file is in the zigbee2mqtt folder. Maybe this does not work any more? Is there a better way to update firmware? Anything I should check?
> 
> It always says that the online version is later than the one on the device.
> 
> Thank you

When you hit check updates on z2m OTA and press one of the styrbar buttons (This Need to be done!), What happen? What messages you got?

HotelAlpha4, 2024-04-07:
I move the button right next to the zigbee router and push the button on the remote immediately before requesting update,  so i have no issue initiating the firmware update.

But even after making all of the changes above i get this:

No updata available for <name>
OTA: Firmware on <id> (remote control N2) is newer than layest firmware online

bztes, 2024-04-12:
Updating my first Styrbar Device worked. Now one week later with my second device I get "_Firmware on (remote control N2) is newer than latest firmware online_"

I found this workaround which works for me: https://github.com/Koenkk/zigbee2mqtt/issues/18023#issuecomment-1624316644

- ext_converters definition in step 1 and 4 can be skipped
- Open: `/.../zigbee2mqtt/node_modules/zigbee-herdsman-converters/devices/ikea.js`
- Find the Styrbar device and replace `ota: ota.tradfri` with `ota: ota.zigbeeOTA` (around line 754)
- after update and before restart undo changes in `configuration.yaml` and `ikea.js`

Lorthiz, 2024-04-22:
> There is no need to run a server or anything like that. zigbee2mqtt is perfectly capable of serving the file. The firmware file and my_index.json can both go in the data directory, which is the [...]

I also followed the steps, restarted zigbee2mqtt and I am getting `Firmware on (remote control N2) is newer than latest firmware online`
My z2m version is 1.36.1-1
I don't have node_modules inside zigbee2mqtt folder so I can't test newer workaround ;/

zoomtronicOR, 2024-05-04:
I had no trouble updating remote to latest firmware, but I regret doing that. After update 2.4.11 arrow buttons stopped sending event action in zigbee2mqtt. other two buttons worked just fine. After binding one ikea lightbulb to remote via Z2mqtt short press on arrows started working, but longer press is not registering in event action, (not triggering).
When I press  and hold any of the arrow keys  its starts to dim lightbulb I just have bind.

boozeman, 2024-05-04:
> I had no trouble updating remote to latest firmware, but I regret doing that. After update 2.4.11 arrow buttons stopped sending event action in zigbee2mqtt. other two buttons worked just fine. After [...]

Hmm, I updated also 2.4.11 and my Styrbar works as intented. I use it with HA and appdaemon contollerX component. Try to remove it to your zigbee network (force remove) and then pair it back and see if it will help. I was forced to do that after 1.0.24 to 2.4.5.

Hammerhand17, 2024-05-04:
> Hmm, I updated also 2.4.11 and my Styrbar works as intented. I use it with HA and appdaemon contollerX component. Try to remove it to your zigbee network (force remove) and then pair it back and see [...]

Trying to update right now, could you please tell me the link for the update? In the guide is pointed the 2.4.5 update, but don't know if I need to download the 2.4.11 one

http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed

boozeman, 2024-05-04:
> Trying to update right now, could you please tell me the link for the update? In the guide is pointed the 2.4.5 update, but don't know if I need to download the 2.4.11 one
> 
> http://fw.ota.homesmart.ikea.net/global/GW1.0/01.20.065/bin/10078031-zingo_kt_styrbar_remote-2.4.5.ota.ota.signed

If you are success to updata 1.0.24 to 2.4.5 manually, then 2.4.11 will be updated automatically, like Z2M Ikea Ota devices do.

Hammerhand17, 2024-05-04:
> If you are success to updata 1.0.24 to 2.4.5 manually, then 2.4.11 will be updated automatically, like Z2M Ikea Ota devices do.

Ok, doing right now, 1 hour left yet. But should I delete the external converter before of after this 2.4.11 update?

boozeman, 2024-05-04:
> Ok, doing right now, 1 hour left yet. But should I delete the external converter before of after this 2.4.11 update?

Seems like I removed it from the configuration.yaml after successfull 2.4.5 update. So you can remove it from the configuration.yaml and leave converter file if you want. 

Let's hope all goes fine. My 2.4.5 to 2.4.11 update took 3 hours and 38 minutes :-)

m10x, 2024-05-05:
Is there any reason to upgrade from the 1.0.024 version? I didnt experience any issue with that version

JohnBull-1975, 2024-05-05:
Updated from 1.0.024 to 2.4.11 and now only up and down works. Sideway clics does not get registered at all!
(Tested in HA with Node Red and compared to another E2001/E2002 still on 1.0.024)

Tried to brows through the thread in a hurry...hope I didn't missed if it was allready reported. Saw mainly regarding error messages during update and firmware numbers... My update whent smothly with no errors.

Any ideas?

Hammerhand17, 2024-05-05:
> Updated from 1.0.024 to 2.4.11 and now only up and down works. Sideway clics does not get registered at all! (Tested in HA with Node Red and compared to another E2001/E2002 still on 1.0.024)
> 
> Tried to brows through the thread in a hurry...hope I didn't missed if it was allready reported. Saw mainly regarding error messages during update and firmware numbers... My update whent smothly with no errors.
> 
> Excluded it and included again and now it does not register any clics at all!
> 
> Any ideas?

In my case, in the latest firmware they simply changed the name of the lateral buttons, so you should go to the device and see the new one. Now it's called "arrow_right_click" and "arrow_left_click. Changing the name in the automations you set should solve it.

JohnBull-1975, 2024-05-05:
> > Updated from 1.0.024 to 2.4.11 and now only up and down works. Sideway clics does not get registered at all! (Tested in HA with Node Red and compared to another E2001/E2002 still on 1.0.024)
> > Tried to brows through the thread in a hurry...hope I didn't missed if it was allready reported. Saw mainly regarding error messages during update and firmware numbers... My update whent smothly with no errors.
> > Excluded it and included again and now it does not register any clics at all!
> > Any ideas?
> 
> In my case, in the latest firmware they simply changed the name of the lateral buttons, so you should go to the device and see the new one. Now it's called "arrow_right_click" and "arrow_left_click. Changing the name in the automations you set should solve it.
> 
> Edit: just saw you are using node red, don't know if this is the same behaviour.

Thanks for quick reply

Damn...I had the switch node sorting commands when I troubleshooted...should have connected directly to a debug node of cause.

After excluding and including the E2001/E2002 from the network it looks to be completely dead! Can't see any repsonse in "About" or "Exposes tabs in MQTT console! But can exclude and include again...

JohnBull-1975, 2024-05-05:
> When you say include and exclude, what you mean? I got my remote work when updating 1.1.24 > 2.4.5 just removing the device from Z2M (Force Remove) and then pairing it back (press pairing button 4 times a row within 5 seconds)

I have tested both excluding by pressing the pairing button - then it leaves the network. When I permint join it comes back.
I have also tested to take out batteries, pressing delete in MQTT and selected force remote. When inserting batteries and hit permit join it comes back.
Still no response at all when I hit any button and after delete-force exclude it says "offline" in flashing red.

boozeman, 2024-05-05:
And you removed that external converter and restart Z2M also?

JohnBull-1975, 2024-05-05:
> And you removed that external converter and restart Z2M also?

external converter?

boozeman, 2024-05-05:
> > And you removed that external converter and restart Z2M also?
> 
> external converter?
> 
> If I remove and re-insert batteries, last seen gets updated. But it still says offline and no response from buttons

And you restarted Z2M also? If so, I am clueless here.  :(

JohnBull-1975, 2024-05-05:
Just tried this:

- Removed batteries
- Deleted with Force Exclude
- Reboot Home Assistant (full reboot)
- Inserted batteries - device came back without activating Permit Join

Stil "Offline" and no response

JohnBull-1975, 2024-05-05:
Read further up in this thread...some buttons looks to have started working after just leaving them for a day or two...
I'll see if my works tomorrow - otherwise I try to take it to IKEA and see if they replace it...

JohnBull-1975, 2024-05-05:
that other thread was ZHA...so not applicable...other than it looks to be problems there as well

JohnBull-1975, 2024-05-05:
> I had no trouble updating remote to latest firmware, but I regret doing that. After update 2.4.11 arrow buttons stopped sending event action in zigbee2mqtt. other two buttons worked just fine. After [...]

My updated styrbar does not react on any button after update to 2.4.11 and excluded and re-joined to the network. Tried to bind it to a bulb according to your post above, but it fails to bind.

zoomtronicOR, 2024-05-05:
> > I had no trouble updating remote to latest firmware, but I regret doing that. After update 2.4.11 arrow buttons stopped sending event action in zigbee2mqtt. other two buttons worked just fine. After binding one ikea lightbulb to remote via Z2mqtt short press on arrows started working, but longer press is not registering in event action, (not triggering). When I press and hold any of the arrow keys its starts to dim lightbulb I just have bind.
> 
> My updated styrbar does not react on any button after update to 2.4.11 and excluded and re-joined to the network. Tried to bind it to a bulb according to your post above, but it fails to bind:
> 
> Error 2024-05-05 23:08:53Frame changeSourceRouteHandler parsing error: RangeError: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readUInt16LE (node:internal/buffer:245:5) at Buffer.readUIntLE (node:internal/buffer:182:17) at Function.deserialize (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/types/basic.ts:19:67) at new EZSPFrameData (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:180:54) at /app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:154:23 at Array.every () at Function.createFrame (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:148:15) at Ezsp.onFrameReceived (/app/node_modules/zigbee-herdsman/src/adapter/ezsp/driver/ezsp.ts:439:35) Error 2024-05-05 23:08:53Unparsed frame 0xc4. Skipped Error 2024-05-05 23:08:55Failed to bind cluster 'genOnOff' from 'Styrbar' to 'Leos Byrå' (Error: Bind 0x287681fffeecf0d8/1 genOnOff from '0x5cc7c1fffe9f59fe/1' failed ({"address":35917,"clusterId":32801,"sequence":216} after 10000ms)) Error 2024-05-05 23:08:55Failed to bind

Did You checked following checkboxes on  bulbs? OnOff, genPollCtrl
do that and try binding again

JohnBull-1975, 2024-05-05:
> Did You checked following checkboxes on bulbs? OnOff, genPollCtrl do that and try binding again

I tried checking OnOff, but there were not genPollCtrl for that bulb:
[image]

JohnBull-1975, 2024-05-05:
For the STYRBAR on 1.0.024, genPollCtrl and PowerCfg is bound to coordinator, but for the one on 2.4.11 it's only bound to genPollCtrl....and I can't get it to bind to PowerCfg

GermanDZ, 2024-05-06:
My control doesn't report `arrow_right_hold`, `arrow_right_release`, `arrow_left_hold`, `arrow_left_release` anymore. But that was happening even before the firmware upgrade (but after upgrading zigbee2mqtt in the recent days).

JohnBull-1975, 2024-05-06:
> My control doesn't report `arrow_right_hold`, `arrow_right_release`, `arrow_left_hold`, `arrow_left_release` anymore. But that was happening even before the firmware upgrade (but after upgrading zigbee2mqtt in the recent days).

I planned taking my STYRBAR back to IKEA this afternoon and see if they accept to swap it under warranty as it stopped working...they use to have a very friendly customer service, but than that might not help. I don't think I tested it after updating Z2MQTT the other day before I updated the firmware of the STYRBAR.

I really hope some skilled enthusiast here can manage to update some driver so we can get our STYRBARs to work again!!!

bchevreau, 2024-05-06:
> For the STYRBAR on 1.0.024, genPollCtrl and PowerCfg is bound to coordinator, but for the one on 2.4.11 it's only bound to genPollCtrl....and I can't get it to bind to PowerCfg
> 
> When it's possible to bind both functions for up down to go directly via coordinator, but Zigbee2MQTT does not react at all for those buttons....is there something incompatible with this new firmware...?
> 
> I don't think I will wait for to long...think I'll try to take the stuff to IKEA and see if they exchange it. It's just half a year old They use to have a very customer friendly approach...

Have you tried to upgrade to 2.4.5 then to 2.4.11 though ? I was in the same boat, then I went thru the update process described above (for 2.4.5) then updated thru regular OTA.

Since then, only the 2 arrows were not working same as you, for which I described the fix above (right and left arrow are color temperature controls):

> If you are having issues with the color temperature controls on this remote, do the following:
> 
> This answer should be added to Styrbar remotes in zigbee2mqtt documentation as well imo, as it just worked for me FINALLY !!!
> 
> 1- Create a group called ikea_lights (or any name) with ID 65289 (the ID is important)
> 2- Add lights AND remote to this "magic" group 65289
> 3- Go to remote and bind the light including feature/cluster 'Scenes' (my binds are
> LevelCtrl, OnOff, PowerCfg, Scenes)
> 4- Add the scenes to the bulb/group via MQTT json in your Home Assistant by clicking on MQTT>Configure with those options:
> 
> Topic:
> zigbee2mqtt/ikea_lights/set
> Payload:
> (add each line one by one before you publish for safety)
> 
> {"scene_add":{"ID":1,"name":"Cold temp","transition":1,"color_temp":250}}
> {"scene_add":{"ID":2,"name":"Neutral temp","transition":1,"color_temp":370}}
> {"scene_add":{"ID":3,"name":"Warm temp","transition":1,"color_temp":454}}
> 
> Then you should be able to switch through the scenes via the 'left/right click' on the remote !!!

JohnBull-1975, 2024-05-06:
> Have you tried to upgrade to 2.4.5 then to 2.4.11 though ? I was in the same boat, then I went thru the update process described above (for 2.4.5) then updated thru regular OTA.
>
> Since then, only [...]

I updated directly to 2.4.11, so I can't try via 2.4.5
I want left/right to be read by the Z2MQTT controller as I don't use the standard IKEA colors/functions for left/right

[24 more replies]