Koenkk/zigbee2mqtt #18516: [Wrong device]: Lidl 399629_2110 recognized as TuYa TS0505B_1

Gnarfoz, 2023-08-03:
### Link

https://www.lidl.de/p/livarno-home-led-deckenleuchte-zigbee-smart-home-16-millionen-farben-38-w/p100357329

### Model

399629_2110

### Description

Livarno Lux Ceiling Panel RGB+CCT

### Vendor

Lidl

### Picture (link)

https://www.lidl.de/assets/gcp65c6ab555c1449879c2e06a470bd945c.jpeg

### Database entry

{"id":14,"type":"Router","ieeeAddr":"0x847127fffe9b551c","nwkAddr":57886,"manufId":4098,"manufName":"_TZ3210_x13bu7za","powerSource":"Mains (single phase)","modelId":"TS0505B","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,4096,8,768,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0505B","manufacturerName":"_TZ3210_x13bu7za","powerSource":1,"zclVersion":3,"appVersion":107,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"colorCapabilities":29,"colorMode":2,"currentX":0,"currentY":0,"currentHue":195,"currentSaturation":253,"colorTemperature":500}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":107,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1842547789},"lastSeen":1691086649415,"defaultSendRequestWhen":"immediate"}

### Notes

My recently bought Lidl ceiling lights show up as the generic TS0505B_1 device.
Seems like it's just another new revision of the same device. (_TZ3210_x13bu7za)
So far, they seem to be working correctly regardless, but maybe some things are wrong in non-obvious ways.

Koenkk, 2023-08-04:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-04 by Koenkk

Gnarfoz, 2023-08-04:
That was quick, thanks!