Koenkk/zigbee2mqtt #18517: Latest version of HA

MYeager1967, 2023-08-03:
### What happened?

Upon updating everything today, I get the following notice from Home Assistant:

Discovered entities with a name that starts with the device name
This stops working in version 2024.2.0. Please address before upgrading.
Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

sensor.front_door_sensor_battery
binary_sensor.patio_door_sensor_battery_low
binary_sensor.front_door_sensor_battery_low
binary_sensor.patio_door_contact
binary_sensor.front_door_contact
sensor.patio_door_sensor_battery

Is this something I can fix on my end or is this a change in code for ZigBee2 MQTT? 

### What did you expect to happen?

Unsure

### Zigbee2MQTT version

1.32.2 commit: 1ec1e57

### Adapter firmware version

20210708

### Adapter

Sonoff

oyvindwe, 2023-08-04:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18445

MYeager1967, 2023-08-04:
Sorry... 😐

oyvindwe, 2023-08-04:
The warning in HA is pretty insistent. You're not the first to report a duplicate…

Please close this issue.

closed 2023-08-04 by Koenkk

MYeager1967, 2023-08-04:
I'm seeing that in the Facebook group. I also read up and realized that it should have been worded better. Since it's something that's being addressed and there's nothing the end user can do, it should have been stated as such. Oh well....