Koenkk/zigbee2mqtt #18519: [New device support]: EGLO MotionSensor Item no: 99106

SJTSmokey, 2023-08-04:
### Link

https://www.amazon.nl/connect-z-Smart-Home-bewegingsmelder-batterijen-app-bediening/dp/B09FSXL927/ref=pd_lpo_sccl_1/260-7114404-8087053?pd_rd_w=BG5bc&content-id=amzn1.sym.76c8aba9-d157-472f-a2eb-7ecb2f29aa15&pf_rd_p=76c8aba9-d157-472f-a2eb-7ecb2f29aa15&pf_rd_r=QYMC809JVFTYPSDBY0EN&pd_rd_wg=wmKR6&pd_rd_r=08415e22-cef9-4671-a577-c6bca871b427&pd_rd_i=B09FSXL927&psc=1

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c13844c5fd37e7","nwkAddr":25871,"manufId":4417,"manufName":"AwoX","powerSource":"Battery","modelId":"EPIR_Zm","epList":[1,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,3,4,4096],"outClusterList":[0,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"EPIR_Zm","manufacturerName":"AwoX","powerSource":3,"zclVersion":3,"appVersion":2,"stackVersion":1,"hwVersion":64,"swBuildId":"2.5.0_999"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":263,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":263,"inClusterList":[0,3,4,1024,1030,4096,32],"outClusterList":[0,3,4,1024,1030,4096,0,3],"clusters":{"genBasic":{"attributes":{}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0029de1d42","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":1,"hwVersion":64,"swBuildId":"2.5.0_999","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691133286979,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

I tried to follow the instructions but i don't have the files you are refering to in the documentation. 
Besides that the intructions are unclear to me, i can't figure out where to place what and what to rewrite and what not. 
In the logs of  MQTT the following comes up: 

Info 2023-08-04 09:14:38Device '0xa4c13844c5fd37e7' joined
Info 2023-08-04 09:14:38MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13844c5fd37e7","ieee_address":"0xa4c13844c5fd37e7"},"type":"device_joined"}'
Info 2023-08-04 09:14:38Starting interview of '0xa4c13844c5fd37e7'
Info 2023-08-04 09:14:38MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13844c5fd37e7","ieee_address":"0xa4c13844c5fd37e7","status":"started"},"type":"device_interview"}'
Info 2023-08-04 09:14:38MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0xa4c13844c5fd37e7","ieee_address":"0xa4c13844c5fd37e7"},"type":"device_announce"}'
Info 2023-08-04 09:14:46Successfully interviewed '0xa4c13844c5fd37e7', device has successfully been paired
Warning 2023-08-04 09:14:46Device '0xa4c13844c5fd37e7' with Zigbee model 'EPIR_Zm' and manufacturer name 'AwoX' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Info 2023-08-04 09:14:46MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0xa4c13844c5fd37e7","ieee_address":"0xa4c13844c5fd37e7","status":"successful","supported":false},"type":"device_interview"}'
Warning 2023-08-04 09:14:46Received message from unsupported device with Zigbee model 'EPIR_Zm' and manufacturer name 'AwoX'
Warning 2023-08-04 09:14:46Please see: https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

The motion sensor shows as a AWOX motion sensor while the product should be an EGLO motion sensor. 
Please add the device to be supported.
Thanks in advance for adding the device.

closed 2023-09-11 by github-actions[bot]

Johannest85, 2024-03-08:
As these sensors was only $2 each I bought a bunch and now I have started looking at it. I have a working external converter, but as a total noob in zigbee i can't get it to finish the interview. This sensor works better then the 31747 as it is a PIR sensor.

I also need to setup a fingerprint for the sensor to not have it mixed up with AwoX lights. After I get the interview working and manage to setup the fingerprint I will make a pull request

https://github.com/Johannest85/zigbee-herdsman-converters/blob/master/src/devices/eglo.ts

feel free to help me out.

R00S, 2024-03-13:
99106 seams to work well in zha over skyconnect. It exposes  Illuminance, occupancy and Open. So I guess we should be able to make it work in z2m as well.

Johannest85, 2024-03-13:
i get it to work with just some actions but at soon as i try to add occupancy it identifies as a lightbulb.

FredrikM97, 2024-08-14:
Hi! I can see that the device I want exists in https://github.com/Koenkk/zigbee-herdsman-converters/blob/b6e0677e8ea146d9b2872301d41508c078eb5d73/src/devices/eglo.ts#L33 but z2m configures it as a[AwoX](https://www.zigbee2mqtt.io/supported-devices/#v=AwoX)
Model[33951/33948](https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948)