Koenkk/zigbee2mqtt #18520: Hive DWS003

spasypaddy, 2023-08-04:
### What happened?

I have 3 of these and they all fail to complete their interview

### What did you expect to happen?

Them to connect and complete their interview so they can be used

### How to reproduce it (minimal and precise)

1. set in pairing mode
2. start join on z2m
3. press button on back every 2-3 seconds for 5mins 
4. interview fails
5. repair again
6. repeat above

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20221226

### Adapter

SLZB-06

closed 2023-09-11 by github-actions[bot]

Puntoboy, 2024-05-17:
Same issue here. They worked fine with Deconz and now I'm migrating to Z2M I've having issues with them.

gogades, 2024-08-22:
I just acquired a few of these and I'm having the same problem.....

Chrisni2, 2024-11-09:
Another case here, is fine in ZHA but will not complete pairing in Z2MQQT