Koenkk/zigbee2mqtt #18526: Warnings to repair mames for 2024.2.0

Mab306, 2023-08-04:
### What happened?

After Update homeassistant core now i have Warnings with same name as entity....
I attched a pic. What i should do or could some one help?
[image]

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

7.1.4.0 build 389

### Adapter

Ezsp v9

fumantsu, 2023-08-04:
Little search doesn't hurt. https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

Koenkk, 2023-08-05:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18445

closed 2023-08-05 by Koenkk