Koenkk/zigbee2mqtt #18531: No OTA endpoint available for Xiaomi GZCGQ11LM (Aqara T1 light intensity sensor)

leroyloren, 2023-08-05:
### What happened?

[image]

### What did you expect to happen?

Xiaomi endpoint linked as expected in order to upgrade the firmware successfully.

### How to reproduce it (minimal and precise)

Check OTA update

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Sonoff

### Debug log

Failed to check if update available for 'Intenzita osvetleni 2' (Failed to find endpoint which support OTA cluster)

`{"id":20,"type":"EndDevice","ieeeAddr":"0x04cf8cdf3c8f3b4c","nwkAddr":54254,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sen_ill.agl01","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":262,"inClusterList":[0,1024,3,1],"outClusterList":[3],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":27,"stackVersion":2,"hwVersion":1,"dateCode":"20200312","swBuildId":"2019\u0000www.","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691248860585,"defaultSendRequestWhen":"immediate"}`

Koenkk, 2023-08-06:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/1cf564faeb1d26554dbb04a885dfcfc5

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

leroyloren, 2023-08-06:
Added an external converter, tried the standard procedures of pairing/deleting/reconfiguring/restarting the device, the problem still remains.

Koenkk, 2023-08-06:
Great, pushed a fix!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-06 by Koenkk

Jinra, 2023-11-03:
Sorry to piggyback on this issue, but I'm having the same issue, but when I try to restart HA after adding the file. I'm getting a config error in `configuration.yaml`

[image]

error
[image]