Koenkk/zigbee2mqtt #18534: FP1 sensor --> Region configuration not working after update

xX-Nexus-Xx, 2023-08-06:
### What happened?

Hi had setup the FP1 with version 1.30.3 (on RPI) and was able to configure the different regions of the FP1. I could create and delete the different regions and confirm with APPLY button (i explained [here](https://discourse.nodered.org/u/xx_Nexus_xx)). There was no dashboard view available in this version.

I upgraded to 1.32.2 and the regiond functionality is not working anymore (same was reported by another user in above link..working with 1.32.1). The APPLY button is not visible anymore and below.
I was able to delete regions but unable to create new ones

Please can you correct the issue as the sensor is not usable at moment

### What did you expect to happen?

Same configuration capability as in version 1.30.3

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20221226

### Adapter

Sonoff

jaqb, 2023-08-15:
I'm experiencing the same issue.

gledsonrabelo, 2023-08-18:
Same issue for me

Bolli1983, 2023-08-30:
Haven't tested it in previous versions, but I can confirm it doesn't work in the current version (also not in edge-version of HA-addon)

debug 2023-08-30 21:40:01Received MQTT message on 'zigbee2mqtt/Anwesenheitssensor Flur/set' with data '{"region_upsert":{"region_id":1}}'
debug 2023-08-30 21:40:01Publishing 'set' 'region_upsert' to 'Anwesenheitssensor Flur'
warning 2023-08-30 21:40:01zigbee-herdsman-converters:xiaomi:aqara_fp1:region_upsert: encountered an error (ZONES_LIST_EMPTY) while parsing configuration commands (input: {"region_id":1})
debug 2023-08-30 21:40:04Received MQTT message on 'zigbee2mqtt/Anwesenheitssensor Flur/set' with data '{"region_upsert":{"zones":[{"x":1}]}}'

piffio, 2023-09-04:
Hello,
just confirming that the issue is still present in the newly released `1.33.0` version

DellBoy4, 2023-09-22:
Same issue. Is there a workaround or fix in the works?

legolas0802, 2023-09-24:
same problem both from ui and sending commands via mqtt

xX-Nexus-Xx, 2023-09-24:
@legolas0802 have you deleted all regions 1-10 before setting new ones ... this is important ... i tested with [mqtt command ](https://discourse.nodered.org/t/aqara-fp1-with-zigbee2mqtt/79596/12?u=xx_nexus_xx)over the weekend and it still works

legolas0802, 2023-09-30:
> @legolas0802 have you deleted all regions 1-10 before setting new ones ... this is important ... i tested with [mqtt command ](https://discourse.nodered.org/t/aqara-fp1-with-zigbee2mqtt/79596/12?u=xx_nexus_xx)over the weekend and it still works

I have never defined regions

Koenkk, 2023-10-15:
This seems to be a frontend issue, so let's continue in https://github.com/nurikk/zigbee2mqtt-frontend/issues/1759

closed 2023-10-15 by Koenkk