Koenkk/zigbee2mqtt #18535: TS0052 _TZ3000_zjtxnoft 2 gang dimmer discovered as 1 gang dimmer

tlovitz, 2023-08-06:
### What happened?

I paired a Tuya 2 gang dimmer switch to Z2M in Home Assistant. It was found and added as a 1 gang dimmer switch and only shows properties for the first channel. I have searched the Issues and Discussions forums and there are three entries that come up under problems when I search the Issues page:

[[New device support]: TS0052 _TZ3000_vluyhemf](https://github.com/Koenkk/zigbee2mqtt/issues/17921)
[Tuya TS0052 unsupported](https://github.com/Koenkk/zigbee2mqtt/issues/17239)
[TUYA TS110E Device type detected incorrectly](https://github.com/Koenkk/zigbee2mqtt/issues/15939)

The final entry seems to be closest to my issue. I attempted to add the external converter that sisalik wrote, but that seemed to cause Z2M to be unable to start. It seems to reference the correct device by Zigbee Model and Zigbee Manufacturer.

This is the database entry for the device:

{"id":12,"type":"Router","ieeeAddr":"0xa4c1384cdbebe52f","nwkAddr":57074,"manufId":4417,"manufName":"_TZ3000_zjtxnoft","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0052","manufacturerName":"_TZ3000_zjtxnoft","powerSource":1,"zclVersion":3,"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":""}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":0}},"manuSpecificTuya_3":{"attributes":{"switchType":0}},"genLevelCtrl":{"attributes":{"currentLevel":254,"minLevel":2,"maxLevel":254,"onOffTransitionTime":50}}},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345,8],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"attributes":{"currentLevel":254,"minLevel":2,"maxLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1691291317436,"defaultSendRequestWhen":"immediate"}

### What did you expect to happen?

I expected to add the device and it would show two dimmer switches.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Nabu Casa SkyConnect

Koenkk, 2023-08-06:
Could you check if the issue is fixed with the following external converter: https://gist.github.com/Koenkk/7461baadc1aaadb35bd2eaefcebc3b97

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, check if issue is fixed

tlovitz, 2023-08-06:
I added the ext.js file next to configuration.yaml and added it to Z2M and restarted. I added the device again (it seems it disconnected on its own overnight). It showed up with the description of being a 2 channel dimmer switch, but in the Z2M dashboard only l1 state and brightness show up. The Home Assistant dashboard only sees l1.

If I go into MQTT settings and publish a packet with topic `zigbee2mqtt/Bathroom_1_lights_dimmer/set` and payload `{"brightness_l2": "75"}` it dims the light. The payload `{"state_l2": "OFF"}` will turn the switch off.

For the sake of completeness, I removed ext.js as an external converter, restarted, and re-published the same payloads and nothing changed. I added the external converter back, restarted, and re-published the payloads and I was able to change the state and brightness.

While there is nothing showing up graphically, the external converter is at least creating access to the second channel.

Thank you for your help.

tlovitz, 2023-08-07:
Here is the screenshot. I'm currently taking the device off of the lights. I don't know if it's the physical switches or the module itself, but the lights randomly come on and go off and it activates the pairing mode and gets lost from Z2M.  I will keep it available for testing for now in a non-use application where it can do no harm. Until I can diagnose if it's the switch or the module, I will keep the device away from any live applications.

[image]

Koenkk, 2023-08-08:
Can you scroll down the screen and make another screenshot?

tlovitz, 2023-08-08:
Here you go. There isn't anything below what was shown before.

[image]

tlovitz, 2023-08-09:
Koen, looks like this finds both the state and brightness for each channel. Your help is very much appreciated! Thank you.

[image]

tlovitz, 2023-08-10:
Since it isn't actually attached to any lights, I cannot confirm operation of dimming lights. It did work when only channel l1 was listed. And it did work when I published a payload to the second channel, in the absence of the graphical display. It will probably be at least two weeks before I'm able to reinstall connected to actual lights that I could test with.

hACKrus, 2023-08-14:
Sorry for my absence in past issue. Now I'm in new home with that dimmer and ready to test.

> I updated https://gist.github.com/Koenkk/7461baadc1aaadb35bd2eaefcebc3b97 , can you try again?
I checked it. Both switches works OK, but dimming doesn't react on long press.

Koenkk, 2023-08-14:
> but dimming doesn't react on long press.

Can you explain this further? Does physically pressing the button not dim the lights? (if yes, is this working with the original hub?)

hACKrus, 2023-08-14:
> > but dimming doesn't react on long press.
> 
> Can you explain this further? Does physically pressing the button not dim the lights? (if yes, is this working with the original hub?)

Short press on physical button react as on/off in state_l1 or state_l2 in correspondense to button. Long press on button doesn't change brightness_l1 or brightness_l2 as expected.

I'm using this dimmer as additional control and doesn't have directly connected lights.

hACKrus, 2023-08-16:
> Can you try with https://gist.github.com/Koenkk/7461baadc1aaadb35bd2eaefcebc3b97

Hi. Same, brightness doesn't change. I noticed that nothing changes when you hold it for a long time, but when you release the button, the on/off state changes.

Koenkk, 2023-08-17:
Can you confirm that when connecting a light to it, it dims when you hold the button?

hACKrus, 2023-08-19:
Hi. I'm onsite. Checked, dimming on physical outputs works from both physical buttons and z2m.

But there is huge problem, dimming started only after several seconds - that was a reason of my previous message, I just didn't wait for a long.

Behaviour is - you hold button, after several seconds dimmer start to change level on physical outputs, but new brightness state reported to z2m only then you leave button. In case if you hold button not enough time - it just change on/off state, on physical outputs as well.

So, brightness state can be readed and I did my idea with additional control. But it's impossible to use in practice, you have to catch light's state.

Do_not_disturb state holded with provided converter. It's OK for me, but may be someone want to have such switch, could you add it please.

Koenkk, 2023-08-20:
This sounds like a firmware issue of the device itself, do you maybe have a TuYa hub to check what is the behaviour there?

main--, 2023-08-21:
I have this device. Regarding the dimming issue: I have it wired up to a regular old rocker (?) light switch, i.e., a switch that *doesn't* snap back when you turn it on. Interestingly, this just works (with no dimming functionality, as expected). This was surprising to me, since I had read reports from other users about dimmer switches which expect short-press/long-press controls and won't (properly) work with rocker switches at all.

Here is the config I ended up with: https://gist.github.com/main--/1426ba558e8a2e0eaea6ff19d86c14d9
This exposes all the functional options I could discover by trial+error through the z2m console, except for genLevelCtrl->onOffTransitionTime, though that one doesn't appear to do anything unfortunately. No matter what I set this to (0, 1, 10, 100, 1000, 10000), the device will always perform a slight (~1s) fade-in+fade-out transition, which is really annoying, because I currently have one non-dimmable bulb connected and it is flickering during those transitions.

Note that configuring a transition time in z2m settings for this device is not advisable, because the device will then always remain ON with a brightness value of 2 instead of turning off. I had thought that the `disableTransition: true` flag might take care of this, but it does not. Though I'm wondering if there is potentially a bug [here](https://github.com/Koenkk/zigbee-herdsman-converters/blob/05c8c047029c7e4d0ed073373428f4e7947d34d0/src/lib/extend.ts#L29) because if I read this code correctly, then `disableTransition: true` would actually *prevent* the `ignore_transition` converter from being added - but I might be misunderstanding how transitions are implemented in general.

hACKrus, 2023-09-22:
> Here is the config I ended up with: https://gist.github.com/main--/1426ba558e8a2e0eaea6ff19d86c14d9 This exposes all the functional options I could discover by trial+error through the z2m console, [...]

Unfortuinately, your config doesn't expose second channel in my case...
[image]

hACKrus, 2023-09-25:
@main-- Thank you for converter. I adopted it by changing model ID. Will test tomorrow.

closed 2023-11-02 by github-actions[bot]

KipK, 2023-12-03:
@Koenkk , I've tested your [ext.js ](https://gist.github.com/Koenkk/7461baadc1aaadb35bd2eaefcebc3b97), this works ok, but some attributes are missing ( same for 1 gang version too ). 

Using booth those converters here, brings Power-on behavior, Switch type and min/max brightness fully working here

[TS005-2ch.js](https://gist.github.com/KipK/c76fc1999f2bb8da83a49643d5094bb7)
[TS005-1ch.js](https://gist.github.com/KipK/43c5b9bbcb66571a2a686f49aa12a595)

KipK, 2023-12-04:
after some test the poweron_behavior doesn't work for the 2ch version. It's ok for the 1ch. 
When I was testing first I was switching the main power a bit too fast. And the light was waking up, but not when I wait few seconds. 

That's exactly the same config than for the 1ch version so I'm a bit lost here

Martijndutch, 2024-02-02:
For some reason, both my TS0052 switched decided to ignore the extention as of yesterday. Could the latest update of z2m have broken the extention somehow? It's back to the old behaviour, even my old scripts work again when I was only using the first gang

KipK, 2024-02-02:
Same problem here, after latest update, my 2 channels TS0052 are only exposing 1 channel. 
It seems z2m now ignore the external converters as I also have some issues with other devices now.

Martijndutch, 2024-02-02:
This is the error from the log:
`error 2024-02-02 03:16:16: Failed to load external converter file 'ext_converter.js' (tuya.extend.light_onoff_brightness is not a function)
error 2024-02-02 03:16:16: Probably there is a syntax error in the file or the external converter is not compatible with the current Zigbee2MQTT version
error 2024-02-02 03:16:16: Note that external converters are not meant for long term usage, it's meant for local testing after which a pull request should be created to add out-of-the-box support for the device
`

Martijndutch, 2024-02-02:
The code as I use it:
`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0052'],
    model: 'TS0052',
    vendor: 'TuYa',
    description: 'Zigbee dimmer module 2 channel',
    extend: tuya.extend.light_onoff_brightness({endpoints: ['l1', 'l2'], noConfigure: true}),
    endpoint: (device) => {
        return {'l1': 1, 'l2': 2};
    },
    exposes: [e.light_brightness().withEndpoint('l1'), e.light_brightness().withEndpoint('l2')],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await tuya.extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;

`

eduardorgos, 2024-02-02:
Problem seems to be in the tuya library, cause `tuya.extend.light_onoff_brightness` method it's not available anymore or maybe has been refactored/moved to another place.

As many of you I've experienced the same problem and after some trial and error I've ended up taking the TS1101 code and use it in the converter to make it works again. If you want to try, replace the old converter code with this one:

[TS0052_as_TS1101_converter.zip](https://github.com/Koenkk/zigbee2mqtt/files/14143911/TS0052_as_TS1101_converter.zip)

Maybe the `whiteLabel` property it's not correct or perhaps unnecessary, but it works like that

Martijndutch, 2024-02-03:
> Problem seems to be in the tuya library, cause `tuya.extend.light_onoff_brightness` method it's not available anymore or maybe has been refactored/moved to another place.
>
> As many of you I've [...]

Thanks, it works again.

Another issue with this switch is that, apparently random, the light goes to 1%. Pretty annoying as my LEDs are not dimabble.

boojew, 2024-02-04:
This works to discover the 2nd channel, but they report 1 or 2% when turned off.

tjallingk, 2024-02-04:
> This works to discover the 2nd channel, but they report 1 or 2% when turned off.

i noticed this to is it only reporting or also doing 1% as i have a non dimmable led in channel 1 that i wouldn't want to be 1% dimmed. BTW you can manually set it to 0% (don't know if it does anything) and the Min brightness doesn't seem to be supported(i just installed the dimmer so don't know if it was in the original from koen). but thanks [eduardorgos](https://github.com/eduardorgos) for this quick fix.

Martijndutch, 2024-02-15:
This apparently random switching to 1% starts to get annoying. I can't turn off the light anymore, not even with the physical switch. Any suggestions how to overcome this with an automation or so?

Martijndutch, 2024-02-15:
[image]
This is also interesting. At first I try to set the brightness using the homeassistant card, which results in an error. Seconds later I have the brightness set by an automation (linked to a motion sensor), and things do work. Puzzles me!

Martijndutch, 2024-02-15:
[image]
Even when you try to switch the device off, it expects to receive a transition number. Weird?

spanizo, 2024-02-15:
Hi, this converter is succesfully working for me, exposing both gangs with no errors dimming or switching them. Hope It helps you.

const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const legacy = require('zigbee-herdsman-converters/lib/legacy');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const {tuyaLight} = tuya.modernExtend;

const definition = {
    zigbeeModel: ['TS0052'],
    model: 'TS0052',
    vendor: 'TuYa',
    description: 'Zigbee dimmer module 2 channel',
    extend: [tuyaLight({minBrightness: true, endpoints: {l1: 1, l2: 2}, configureReporting: true})],
    exposes: [e.light_brightness().withEndpoint('l1'), e.light_brightness().withEndpoint('l2')],
    meta: {multiEndpoint: true},
    configure: async (device, coordinatorEndpoint, logger) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        await tuya.extend.light_onoff_brightness().configure(device, coordinatorEndpoint, logger);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff', 'genLevelCtrl']);
    },
};

module.exports = definition;

KipK, 2024-02-18:
Previous converters were not working with latest z2m updates because of some change in tuya functions.

Using converter from @eduardorgos , I've modified it to add the switchType and PowerOnBehavior settings, things are working again now.

For those who have the 1% brightness when powering off, try to disable effect in device settings by setting it to 0 again. 

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const e = exposes.presets;
const ea = exposes.access;

const {tuyaLight} = tuya.modernExtend;

const definition = {
	fingerprint: [{modelID: 'TS0052', manufacturerName: '_TZ3000_zjtxnoft'}],
	model: 'TS0052_dimmer_module_2ch',
	vendor: 'TuYa',
	description: 'Zigbee dimmer module 2 channel',
	whiteLabel: [{vendor: 'OXT', model: 'SWTZ25'}],
	extend: [
		tuyaLight(
		{minBrightness: true,
		endpoints: {l1: 1, l2: 2},
		switchType: true,
		powerOnBehavior: true,
		configureReporting: true}),
		],
	configure: async (device, coordinatorEndpoint, logger) => {
		await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
	},
};

module.exports = definition;

`

Koenkk, 2024-02-18:
Added the `_TZ3000_zjtxnoft`

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

boojew, 2024-02-19:
The device appears to work normally with this converter, but continues to report 1-2% when powered off - even even setting transition to 0

KipK, 2024-02-19:
That's strange, I got this issue once with previous converters but not anymore since.

KipK, 2024-02-19:
@Martijndutch logs shows an interesting  error: "transition is not a number ( got string at ....) " 
I'm pretty sure we have a bug somewhere on how we handle transition settings in tuya lib.  As transitions never worked on all tuya device I've tried with z2m, that would make sense.

Martijndutch, 2024-02-20:
I'll try to dive in a blt deeper once I'm back from holidays, but now I think of it, I've two of these dimmers, same batch, but they behave slightly different.

raarts, 2024-02-24:
I just bought this device. It is recognised as a 1-ch dimmer switch. Some things also don't work:

> Error 2024-02-24 19:43:35Publish 'set' 'min_brightness' to '0xa4c1384af5049184' failed: 'Error: Write 0xa4c1384af5049184/1 genLevelCtrl({"64512":{"value":6399,"type":33}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'

Selecting an effect did nothing, no errors.  Here's my exposes screen,, sorry Dutch. 

[image]

Note I am a complete beginner and just started playing with HA. I _**am**_ an experienced programmer though.

If you need anything, just ask, I'm pretty motivated for this device to work.

raarts, 2024-02-26:
Fixed it by editing `zigbee-herdsman-converters/devices/tuya.js`

and changing this line:

```
fingerprint: tuya.fingerprint('TS0052', ['_TZ3000_zjtxnoft']),
```

into:
```
fingerprint: tuya.fingerprint('TS0052', ['_TZ3000_zjtxnoft', '_TZ3000_kvwrdf47']),
```

Don't know if this is the correct approach though...

Martijndutch, 2024-02-26:
> Fixed it by editing `zigbee-herdsman-converters/devices/tuya.js`
> 
Without using a converter file?

raarts, 2024-02-26:
Just changed it by hand in that file in the docker container and restarted the container.
I found the `zigbee-herdsman-converters` package on github, so it would be an easy PR, but I first need to be sure if this is the correct approach.

raarts, 2024-02-26:
Funny enough the brightness info is not published to MQTT, so no sliders in the GUI.
Maybe it's because it's missing here in the extend array, there are other definitions which *do* reference 'brightness'. 
 
```
    {
        fingerprint: tuya.fingerprint('TS0052', ['_TZ3000_zjtxnoft', '_TZ3000_kvwrdf47']),
        model: 'TS0052_2',
        vendor: 'TuYa',
        description: 'Zigbee dimmer module 2 channel',
        extend: [
            (0, modernExtend_1.deviceEndpoints)({ endpoints: { 'l1': 1, 'l2': 2 } }),
            tuyaLight({ powerOnBehavior: true, configureReporting: true, switchType: true, minBrightness: true, endpointNames: ['l1', 'l2'] }),
        ],
        configure: async (device, coordinatorEndpoint, logger) => {
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        },
    },
```

Koenkk, 2024-02-27:
@raarts added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Martijndutch, 2024-02-27:
How long does it typically take before this is the stable release? I don't
want to risk messing up my ha installation..

Koenkk, 2024-02-27:
next release = 1 March

KipK, 2024-03-02:
Doesn't work with 1.36.0, now only 1 channel is detected.  Seems using previous converter dosen't works too now.

tjallingk, 2024-03-02:
i have not yet upgraded (will not for now seeing this) but it seems its native supported now? atleast looking at the release notes so maybe the converter is breaking stuff and removing the device and pairing it without the converter solves the problem now?(its just an idea i have no clue what exactly is happening and if this even could solve it)

Koenkk, 2024-03-02:
@KipK could you provide the data/database.db entry of this device?

KipK, 2024-03-03:
@koenkk, here is the device db before and after the update. Device was removed and paired inbetween.

Before:

```
{"id":29,"type":"Router","ieeeAddr":"0xa4c138607d6fcc9f","nwkAddr":21003,"manufId":4417,"manufName":"_TZ3000_zjtxnoft","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\f\u0000\u0000\u0000\u0012\b\u0000\u0000\u0000\u0012]\u0000\u0000\u0000\u0012*��,\u0012","65506":54,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","zclVersion":3,"manufacturerName":"_TZ3000_zjtxnoft","appVersion":68,"modelId":"TS0052","powerSource":1}},"manuSpecificBosch":{"attributes":{"53251":"AAAAAgAA"}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"moesStartUpOnOff":1}},"manuSpecificTuya_3":{"attributes":{"switchType":2,"powerOnBehavior":2}},"genLevelCtrl":{"attributes":{"currentLevel":152,"minLevel":40,"maxLevel":212,"onOffTransitionTime":50}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{"onLevelSupported":false,"scenes":{"0_2":{"name":"cozy","state":{"state":"ON","brightness":254,"color_mode":"xy"}},"1_2":{"name":"apero","state":{"state":"ON","brightness":254,"color_mode":"xy"}},"2_2":{"name":"full","state":{"state":"ON","brightness":254,"color_mode":"xy"}},"3_2":{"name":"night","state":{"state":"ON","brightness":254,"color_mode":"xy"}}}}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345,8],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}},"genLevelCtrl":{"attributes":{"currentLevel":212,"minLevel":40,"maxLevel":212}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1709459908408}
{"id":55,"type":"Group","groupID":2,"members":[{"deviceIeeeAddr":"0xa4c138f43f251d6d","endpointID":1},{"deviceIeeeAddr":"0xa4c138607d6fcc9f","endpointID":1}],"meta":{}}
```

After:

`{"id":29,"type":"Router","ieeeAddr":"0xa4c138607d6fcc9f","nwkAddr":23914,"manufId":4417,"manufName":"_TZ3000_zjtxnoft","powerSource":"Mains (single phase)","modelId":"TS0052","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[3,4,5,6,57344,57345,8,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"stackVersion":0,"dateCode":"","manufacturerName":"_TZ3000_zjtxnoft","zclVersion":3,"appVersion":68,"modelId":"TS0052","powerSource":1}},"genOnOff":{"attributes":{"onOff":1}},"genLevelCtrl":{"attributes":{"currentLevel":152}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6,57345,8],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":212}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea81c6e","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null},{"cluster":8,"attrId":0,"minRepIntval":10,"maxRepIntval":65000,"repChange":1,"manufacturerCode":null}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1324213189},"lastSeen":1709463338205}`

Koenkk, 2024-03-04:
@KipK can you provide a screenshot of the frontend: z2m -> settings -> about page and device -> about page?

KipK, 2024-03-05:
@Koenkk, here is it
[image]
[image]

Martijndutch, 2024-03-14:
What would be the order? Remove the converter and upgrade, or the other way around?

hfolkers, 2024-03-31:
@KipK are you able to control both channels?
I tried everything but only one channel is reported.

KipK, 2024-04-01:
Yes booth channels are on here

hfolkers, 2024-04-02:
Thanks for the answer.
Today i totally removed the external converters from the system (instead of benaming) and installed the latest update. More it is working correct!
So Thanks to all who are working on this integration!

grant17, 2024-12-08:
Hello is any one having an issue with this device and the latest zigbee update 1.42.0-2? I am finding if the dimmer is added to a zigbee group that the group cant be controlled from home assistant and the log message reports "z2m: No converter available for 'state' ("ON")"