Koenkk/zigbee2mqtt #18536: SP02-ZB001 iAlarm Motion Sensor

ADavidBarros, 2023-08-06:
### Link

https://www.amazon.com/iAlarm-Motion-Sensor-Corridors-Bathrooms/dp/B0BYRM6BGT/ref=sr_1_11_sspa?keywords=zigbee+motion+sensor&qid=1691297668&sr=8-11-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9tdGY&psc=1

### Database entry

{"id":2,"type":"EndDevice","ieeeAddr":"0xa4c138e6144fef5d","nwkAddr":34513,"manufId":4417,"manufName":"_TZE200_mgxy2d9f","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691295961574,"defaultSendRequestWhen":"immediate"}

### Comments

I couldn't get it to work with the proper sensors. It is a motion sensor per this guy's review in  Amazon. 

" set this up with home assistant using the zigbee integration. Using Z2M you'll need to add a custom js file to it as there is not native support for this device yet. Though it may be added soon.
The device itself exposes four entities. Tampered which is a tilt sensor. Remaining battery percentage. Motion sensor, which is listed as occupancy. And finally Link quality which is your signal strength.

The battery is a C123A and it's included. The motion sensor mount attaches to the wall with a screw and has an included drywall anchor. The mount has a magnet on the end which allows you to attach the motion sensor and adjust the angle. When the battery cover is remove there are dip switches that allow you to configure the wait time between triggering, the sensitivity, and to turn off or on the LED when motion is detected. By default it's on high sensitivity, 60s timeout and LED On. The sensor works well and has integrated well into my home automation platform."

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: 'SP02-ZB', // Vendor model number, look on the device for a model number
    vendor: 'iAlarm', // Vendor of the device (only used for documentation and startup logging)
    description: 'Motion Sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.presence(), e.illuminance_lux()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

Turgo82, 2023-08-09:
I think there is something missing from you code Z2M doesn't want to start. I'm no expert... but this is what have that seems to allow it to start. I don't know yet if its fully works at all. `const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_mgxy2d9f',
        },
    ],
    model: 'SP02-ZB',
    vendor: 'TuYa',
    description: 'Fill in a description of the device here',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [e.battery(),e.battery_low(),e.battery(),e.battery_voltage(),e.presence(),e.illuminance_lux(), e.occupancy(), e.enum('sensitivity', ea.ALL, ['low', 'medium', 'high']).withDescription('PIR sensor sensitivity'),
                e.illuminance()
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
        ],
    },
};

module.exports = definition;
`

ADavidBarros, 2023-08-09:
I was able to get to work using this settings.  The only thing that detects is presence.

`const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const legacy  = require('zigbee-herdsman-converters/lib/legacy');

const definition = {

	model: 'TS0601_human_presence_sensor',
	vendor: 'TuYa',
	description: 'Human presence sensor Zigbee',
	fromZigbee: [legacy.fromZigbee.hpsz],
	toZigbee: [legacy.toZigbee.hpsz],
	onEvent: tuya.onEventSetLocalTime,
	fingerprint: [
		{
			// The model ID from: Device with modelID 'TS0601' is not supported
			// You may need to add \u0000 at the end of the name in some cases
			modelID: 'TS0601',
			// The manufacturer name from: Device with modelID 'TS0601' is not supported.
			manufacturerName: '_TZE200_mgxy2d9f',
		},
	],
	exposes: [e.presence(),
		e.numeric('duration_of_attendance', ea.STATE).withUnit('min')
		.withDescription('Shows the presence duration in minutes'),
		e.numeric('duration_of_absence', ea.STATE).withUnit('min')
		.withDescription('Shows the duration of the absence in minutes'),
		e.binary('led_state', ea.STATE_SET, true, false)
		.withDescription('Turns the onboard LED on or off'),
	],
	description: 'Fill in a description of the device here',
	fromZigbee: [tuya.fz.datapoints, legacy.fz.tuya_smart_human_presense_sensor],
	toZigbee: [tuya.tz.datapoints, legacy.tz.tuya_smart_human_presense_sensor],
	//configure: tuya.configureMagicPacket,
	configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
            await tuya.sendDataPointEnum(endpoint, legacy.dataPoints.trsfTumbleSwitch, 0);
            await tuya.configureMagicPacket(device, coordinatorEndpoint, logger);
        },
	meta: {
		// All datapoints go in here
		tuyaDatapoints: [
			[1, 'presence', tuya.valueConverter.trueFalse1],
			[102, 'occupancy', tuya.valueConverter.trueFalse1],
			[103, 'illuminance_lux', tuya.valueConverter.raw],
		],
	},
};`

closed 2023-09-17 by github-actions[bot]

lezruk, 2023-09-25:
My config for this iAlarm device

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_mgxy2d9f',
        },
    ],
    model: 'SP02-ZB001',
    vendor: 'iAlarm',
    description: 'iAlarm Zigbee Motion Sensor',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.tamper(), e.battery(), e.occupancy(), e.linkquality()
    ],
    meta: {
        // All datapoints go in here
        tuyaDatapoints: [
          [1, 'occupancy', tuya.valueConverter.trueFalse0],
          [4, 'battery', tuya.valueConverter.raw],
          [5, 'tamper', tuya.valueConverter.trueFalse0],
        ],
    },
};

module.exports = definition;
```