Koenkk/zigbee2mqtt #18538: [New device support]: Schneider Electric - temperature & humidity sensor (550B1024)

Ramshackles, 2023-08-06:
### Link

https://www1.lk.dk/katalog/nmkatalog.ph7?vnr=550B1024

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0xd87a3bfffecdb49f","nwkAddr":19607,"manufId":4190,"manufName":"Schneider Electric","powerSource":"Battery","modelId":"CCT593011_AS","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,1,3,32,1026,1029,2821],"outClusterList":[0,3,25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xe0798dfffea8625d","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":6,"hwVersion":0,"dateCode":"20210611","swBuildId":"2106111","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691301392130,"defaultSendRequestWhen":"active","checkinInterval":3600}

### Comments

After initially using the below converter, the device just shows up as supported - it doesn't do any reporting besides initially:

`info  2023-08-04 18:52:24: MQTT publish: topic 'homeassistant/sensor/0xd87a3bfffecdb49f/battery/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xd87a3bfffecdb49f"],"manufacturer":"Schneider Electric","model":"Schneider Electric temperature & humidity sensor (550B1024)","name":"0xd87a3bfffecdb49f","sw_version":"2106111"},"device_class":"battery","enabled_by_default":true,"entity_category":"diagnostic","name":"0xd87a3bfffecdb49f battery","state_class":"measurement","state_topic":"zigbee2mqtt/0xd87a3bfffecdb49f","unique_id":"0xd87a3bfffecdb49f_battery_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.battery }}"}'
info  2023-08-04 18:52:24: MQTT publish: topic 'homeassistant/sensor/0xd87a3bfffecdb49f/temperature/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xd87a3bfffecdb49f"],"manufacturer":"Schneider Electric","model":"Schneider Electric temperature & humidity sensor (550B1024)","name":"0xd87a3bfffecdb49f","sw_version":"2106111"},"device_class":"temperature","enabled_by_default":true,"name":"0xd87a3bfffecdb49f temperature","state_class":"measurement","state_topic":"zigbee2mqtt/0xd87a3bfffecdb49f","unique_id":"0xd87a3bfffecdb49f_temperature_zigbee2mqtt","unit_of_measurement":"°C","value_template":"{{ value_json.temperature }}"}'
info  2023-08-04 18:52:24: MQTT publish: topic 'homeassistant/sensor/0xd87a3bfffecdb49f/humidity/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xd87a3bfffecdb49f"],"manufacturer":"Schneider Electric","model":"Schneider Electric temperature & humidity sensor (550B1024)","name":"0xd87a3bfffecdb49f","sw_version":"2106111"},"device_class":"humidity","enabled_by_default":true,"name":"0xd87a3bfffecdb49f humidity","state_class":"measurement","state_topic":"zigbee2mqtt/0xd87a3bfffecdb49f","unique_id":"0xd87a3bfffecdb49f_humidity_zigbee2mqtt","unit_of_measurement":"%","value_template":"{{ value_json.humidity }}"}'
info  2023-08-04 18:52:24: MQTT publish: topic 'homeassistant/sensor/0xd87a3bfffecdb49f/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"device":{"identifiers":["zigbee2mqtt_0xd87a3bfffecdb49f"],"manufacturer":"Schneider Electric","model":"Schneider Electric temperature & humidity sensor (550B1024)","name":"0xd87a3bfffecdb49f","sw_version":"2106111"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"0xd87a3bfffecdb49f linkquality","state_class":"measurement","state_topic":"zigbee2mqtt/0xd87a3bfffecdb49f","unique_id":"0xd87a3bfffecdb49f_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'`

And then ever so often;
`debug 2023-08-04 20:56:07: Received Zigbee message from 'varme', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":17171778,"imageType":39,"manufacturerCode":4190}' from endpoint 1 with groupID 0
debug 2023-08-04 20:56:07: Device 'varme' requested OTA
debug 2023-08-04 20:56:07: Responded to OTA request of 'varme' with 'NO_IMAGE_AVAILABLE'`

Sadly not knowledgeable enough to progress any further on my own

### External converter

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['CCT593011_AS'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '550B1024', // Vendor model number, look on the device for a model number
    vendor: 'Schneider Electric', // Vendor of the device (only used for documentation and startup logging)
    description: 'Schneider Electric temperature & humidity sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

tomasaberg, 2023-08-08:
I've created an issue for this device as well.
https://github.com/Koenkk/zigbee2mqtt/issues/18423

There is some additional information here
https://github.com/Koenkk/zigbee-herdsman/issues/472 by @willu, not sure if he managed to get it to work though.

Ramshackles, 2023-08-16:
> I've created an issue for this device as well. #18423
> 
> There is some additional information here [Koenkk/zigbee-herdsman#472](https://github.com/Koenkk/zigbee-herdsman/issues/472) by @willu, not sure if he managed to get it to work though.

Ah my bad, didn't realise there was already an issue.

I got a bit further by copying bits and pieces from existing devices, it reads the voltage wrong (200 %) :) but I seem to get a correct temperature and humidity.

This was more luck than anything though, but might help someone nail it perfectly.

`const fz = {...require('zigbee-herdsman-converters/converters/fromZigbee'), legacy: require('zigbee-herdsman-converters/lib/legacy').fromZigbee};
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const constants = require('zigbee-herdsman-converters/lib/constants');
const e = exposes.presets;
const ea = exposes.access;

const definition  = {
    zigbeeModel: ['CCTFR6100'],
    model: 'CCTFR6100Z3',
    vendor: 'Schneider Electric',
    description: 'Wiser radiator thermostat',
    fromZigbee: [fz.ignore_basic_report, fz.ignore_haDiagnostic, fz.ignore_genOta, fz.ignore_zclversion_read,
        fz.legacy.wiser_thermostat, fz.legacy.wiser_itrv_battery, fz.hvac_user_interface, fz.wiser_device_info],
    toZigbee: [tz.thermostat_occupied_heating_setpoint, tz.thermostat_keypad_lockout],
    exposes: [exposes.climate().withSetpoint('occupied_heating_setpoint', 7, 30, 1).withLocalTemperature(ea.STATE)
        .withRunningState(['idle', 'heat'], ea.STATE).withPiHeatingDemand()],
    configure: async (device, coordinatorEndpoint, logger) => {
        const endpoint = device.getEndpoint(1);
        const binds = ['genBasic', 'genPowerCfg', 'hvacThermostat', 'haDiagnostic'];
        await reporting.bind(endpoint, coordinatorEndpoint, binds);
        await reporting.batteryVoltage(endpoint);
        await reporting.thermostatTemperature(endpoint);
        await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
        await reporting.thermostatPIHeatingDemand(endpoint);
        // bind of hvacUserInterfaceCfg fails with 'Table Full', does this have any effect?
        await endpoint.configureReporting('hvacUserInterfaceCfg', [{attribute: 'keypadLockout', reportableChange: 1,
            minimumReportInterval: constants.repInterval.MINUTE, maximumReportInterval: constants.repInterval.HOUR}]);
    },
};

module.exports = definition;`

tomasaberg, 2023-08-16:
Thank you, I will try this on my devices

Ramshackles, 2023-08-16:
> Thank you, I will try this on my devices

Hang on, I seem to have copied the wrong external files code 🫣

tomasaberg, 2023-08-18:
@Ramshackles What coordinator do you use?
My sensors takes several attempts on joining the network, but after a while it seems to be able to join.
It's ~40cm to my coordinator (Sonoff ABDongle-P). However, no data is reported.

Worth mentioning is that I'm doing this in a test HA instance in Virtualbox

Ramshackles, 2023-08-19:
@tomasaberg I'm using ZBDongle-E on rPi with HA and Z2M

These devices joined instantly the first time, reporting took a bit of time though.
They're both reporting battery, humidity and linkquality but one of them hasn't reported temperatur yet - and the temperature does seem off.

Hoping someone with more knowledge gets one of these - or I can sponsor one for someone with the needed knowledge - perhaps I can ship one to @Koenkk ? 😄

Koenkk, 2023-08-20:
Can you check with the following ext converter: https://gist.github.com/Koenkk/222ed16f2abc72c1b360af10b0536394 , please re-pair the device after this.

Ramshackles, 2023-08-20:
> Can you check with the following ext converter: https://gist.github.com/Koenkk/222ed16f2abc72c1b360af10b0536394 , please re-pair the device after this.

That enables pairing, and it configures it probably it seems, but the battery, temperature, humidity and linkquality doesnt populate yet, only just re-addded it 10 mins ago.

`Debug 2023-08-20 11:54:50Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'genPowerCfg', data '{"256":0}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:54:50Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":8021}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:54:51Received Zigbee message from '0xd87a3bfffecdb555', type 'readResponse', cluster 'genPowerCfg', data '{"batteryPercentageRemaining":200}' from endpoint 1 with groupID 0
Info 2023-08-20 11:54:53Successfully configured '0xd87a3bfffecdb555'
Debug 2023-08-20 11:55:00Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6538}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:55:10Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":6433}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:55:18Received Zigbee message from '0xd87a3bfffecdb555', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":17171778,"imageType":39,"manufacturerCode":4190}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:55:18Device '0xd87a3bfffecdb555' requested OTA
Debug 2023-08-20 11:55:18Responded to OTA request of '0xd87a3bfffecdb555' with 'NO_IMAGE_AVAILABLE'
Debug 2023-08-20 11:55:20Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5863}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:55:33Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":3018}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:55:33Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5743}' from endpoint 1 with groupID 0
Debug 2023-08-20 11:55:53Received Zigbee message from '0xd87a3bfffecdb555', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5576}' from endpoint 1 with groupID 0`

Koenkk, 2023-08-20:
The converters are missing, updated https://gist.github.com/Koenkk/222ed16f2abc72c1b360af10b0536394

tomasaberg, 2023-08-26:
> The converters are missing, updated https://gist.github.com/Koenkk/222ed16f2abc72c1b360af10b0536394

This works! \o/
I have another sensor (Aqara temperature) to match with and they seem to report the same values.
Thank you a lot!

closed 2023-10-04 by github-actions[bot]

Ramshackles, 2023-10-08:
> The converters are missing, updated https://gist.github.com/Koenkk/222ed16f2abc72c1b360af10b0536394

Sorry was away for a while.

The external converter works perfectly thanks, issue went stale - just tagging in case it's needed for the main project to get update eventually.

Koenkk, 2023-10-08:
@Ramshackles thanks, you can remove the ext converter after updating to the next release.