Koenkk/zigbee2mqtt #18540: EcoDim Eco-Dim.05 not longer supported

ZabbixGuru, 2023-08-06:
### What happened?

Added Eco-Dim.05 to Zigbee2MQTT (tried version 1.32.2-1 and Edge) however device remains unsupported. 
Added device several times (>10) however without succes. 
Output from  log:

``` Zigbee2MQTT:warn  2023-08-06 21:45:05: Device '0x84b4dbfffe089fc4' with Zigbee model 'EcoDim-Zigbee 3.0' and manufacturer name 'EcoDim BV' is NOT supported, please follow https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html
Zigbee2MQTT:info  2023-08-06 21:45:05: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":null,"friendly_name":"0x84b4dbfffe089fc4","ieee_address":"0x84b4dbfffe089fc4","status":"successful","supported":false},"type":"device_interview"}' ```

Output from database.db:

``` {
  "id": 4,
  "type": "Router",
  "ieeeAddr": "0x84b4dbfffe089fc4",
  "nwkAddr": 58448,
  "manufId": 4098,
  "manufName": "EcoDim BV",
  "powerSource": "Mains (single phase)",
  "modelId": "EcoDim-Zigbee 3.0",
  "epList": [
    1,
    2,
    242
  ],
  "endpoints": {
    "1": {
      "profId": 260,
      "epId": 1,
      "devId": 257,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        4096
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "modelId": "EcoDim-Zigbee 3.0",
            "manufacturerName": "EcoDim BV",
            "appVersion": 0,
            "stackVersion": 0,
            "hwVersion": 0
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "2": {
      "profId": 260,
      "epId": 2,
      "devId": 257,
      "inClusterList": [
        0,
        3,
        4,
        5,
        6,
        8,
        4096
      ],
      "outClusterList": [
        25
      ],
      "clusters": {
        "genBasic": {
          "attributes": {
            "appVersion": 0,
            "stackVersion": 0,
            "hwVersion": 0
          }
        }
      },
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    },
    "242": {
      "profId": 41440,
      "epId": 242,
      "devId": 97,
      "inClusterList": [],
      "outClusterList": [
        33
      ],
      "clusters": {},
      "binds": [],
      "configuredReportings": [],
      "meta": {}
    }
  },
  "appVersion": 0,
  "stackVersion": 0,
  "hwVersion": 0,
  "dateCode": "20230227",
  "swBuildId": "1.00",
  "zclVersion": 8,
  "interviewCompleted": true,
  "meta": {},
  "lastSeen": 1691351105606,
  "defaultSendRequestWhen": "immediate"
}
``` 

My other devices added correctly. 

### What did you expect to happen?

That the device was correctly added. 

### How to reproduce it (minimal and precise)

Remove device from Zigbee2MQTT, reset device and add again. 

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20230507

### Adapter

sonoff zigbee 3.0 usb dongle plus

Koenkk, 2023-08-07:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

ZabbixGuru, 2023-08-07:
Bedankt! Thank you for the quick response and fix, confirm that it's discovered correct.