Koenkk/zigbee2mqtt #18545: MQTT name for HA

RubenKelevra, 2023-08-07:
### What happened?

I've got two error messages by HA today:

[image]

[image]

Learn more link goes in both cases to [this page](https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/) which is no help at all.

How can I continue to use Z2M after 2024.2?

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20221226

### Adapter

Slaesh's CC2652RB stick

olafz, 2023-08-07:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18445

RubenKelevra, 2023-08-07:
Not a duplicate, as that one is closed as fixed. 🤷‍♂️

I couldn't find an open bug report for that.

olafz, 2023-08-07:
> Not a duplicate, as that one is closed as fixed. 🤷‍♂️

It is fixed in the image tagged "latest-dev". If you don't want to run that version, you will have to wait a bit: the fix will be included in the release of next month.

RubenKelevra, 2023-08-07:
Okay? In this ticket 1.32.2-dev was mentioned, which is the Version I'm running. So the fix is not in 1.32.2?

olafz, 2023-08-07:
> So the fix is not in 1.32.2?

No. See also this comment (and the ones after that) in the issue mentioned: https://github.com/Koenkk/zigbee2mqtt/issues/18445#issuecomment-1666801532

Koenkk, 2023-08-07:
Duplicate of https://github.com/Koenkk/zigbee2mqtt/issues/18445 and already fixed in -dev.

closed 2023-08-07 by Koenkk