Koenkk/zigbee2mqtt #18559: [New device support]: Eglo 900093

Cayn183, 2023-08-08:
### Link

https://www.eglo.com/de/deckenleuchte-rovito-z-900093.html

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c138fb88650e22","nwkAddr":17255,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096,64599],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":1,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":25,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":269,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1690665009731,"defaultSendRequestWhen":"immediate"}

### Comments

Device is detected as Awox 33943/33944/33946.
Acutally the device isnt doing anything when i try to change anything.
There is the similiar device in round and not sqare: Eglo 900091, so i hope that is not much work.
https://www.eglo.com/de/deckenleuchte-rovito-z-900091.html

I have tried a few hours to set up a new device, but i dont get it to work. 
I hope you can help me.

Best regards
Cayn183

closed 2023-09-15 by github-actions[bot]

Cayn183, 2023-09-15:
reopen please

dexterionut, 2024-01-12:
> reopen please

Hey @Cayn183! Did you managed to get it working? I'm interested in the same lamp but I'm not sure if I can connect it to zigbee2mqtt, otherwise I think I will just go with the round one you were mentioning.

Cayn183, 2024-01-12:
Hey, the lamp will be detected as another, but it’s working.