Koenkk/zigbee2mqtt #18560: Devices only pair through coordinator, not routers

rhvs, 2023-08-08:
### What happened?

After upgrading the firmware of the Slaesh stick (or at least that's what I think happened), devices cannot join the Zigbee network from any of the routers (which are mostly Hue bulbs). Only when I bring the device close to the coordinator, the device can join the network and I can move it around the house.

### What did you expect to happen?

Devices to join the network from a router (far away from the coordinator).

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

Slaesh

francisp2, 2023-08-10:
Try what is shown here : 

https://github.com/Koenkk/zigbee2mqtt/issues/10339#issuecomment-1179710676

rhvs, 2023-08-16:
Tried to do that - but couldn't get it working from HA OS. Hopefully the 1.33.0 update will allow me to get to a fix.

Koenkk, 2023-08-17:
Let's continue in https://github.com/Koenkk/zigbee2mqtt/issues/18426

closed 2023-08-17 by Koenkk