Koenkk/zigbee2mqtt #18561: Zemismart Tuya Zigbee 3 Phase Electric Energy SPM02-D2TZ

xstefa12, 2023-08-08:
### Link

https://www.zemismart.com/products/spm02-d2tz

### Database entry

{"id":19,"type":"Router","ieeeAddr":"0x187a3efffe357a64","nwkAddr":33371,"manufId":4098,"manufName":"_TZE200_ves1ycwx","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_ves1ycwx","stackVersion":0,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691524249775,"defaultSendRequestWhen":"immediate"}

### Comments

Visually looks like Schneider Electric A9MEM1570 https://www.zigbee2mqtt.io/devices/A9MEM1570.html ...

neosloter, 2023-08-14:
This device works with an external convector.
Added like this:
In addon settings:
```
external_converters:
   - TS0601_new.js
```
Create TS0601_new.js file in z2m folder with contents:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    // Since a lot of TuYa devices use the same modelID, but use different datapoints
    // it's necessary to provide a fingerprint instead of a zigbeeModel
    fingerprint: [
        {
            // The model ID from: Device with modelID 'TS0601' is not supported
            // You may need to add \u0000 at the end of the name in some cases
            modelID: 'TS0601',
            // The manufacturer name from: Device with modelID 'TS0601' is not supported.
            manufacturerName: '_TZE200_ves1ycwx',
        },
    ],
        model: 'SPM02',
        vendor: 'TuYa',
        description: 'Smart energy monitor for 3P+N system',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [tuya.exposes.voltageWithPhase('X'), tuya.exposes.voltageWithPhase('Y'), tuya.exposes.voltageWithPhase('Z'),
            tuya.exposes.powerWithPhase('X'), tuya.exposes.powerWithPhase('Y'), tuya.exposes.powerWithPhase('Z'),
            tuya.exposes.currentWithPhase('X'), tuya.exposes.currentWithPhase('Y'), tuya.exposes.currentWithPhase('Z'),
            // Change the description according to the specifications of the device
            e.energy().withDescription('Total forward active energy'),
            e.produced_energy().withDescription('Total reverse active energy'),
        ],
    meta: {
        tuyaDatapoints: [
            [1, 'energy', tuya.valueConverter.divideBy100],
            [2, 'produced_energy', tuya.valueConverter.divideBy100],
            [6, null, tuya.valueConverter.phaseVariant2WithPhase('X')],
            [7, null, tuya.valueConverter.phaseVariant2WithPhase('Y')],
            [8, null, tuya.valueConverter.phaseVariant2WithPhase('Z')],
            ],
    },
};

module.exports = definition;
```

xstefa12, 2023-08-14:
Hello neosloter, thank you, it's working.! Is there any chance to read other parameters like on Schneider one?

Koenkk, 2023-08-14:
This device is supported now: https://github.com/Koenkk/zigbee-herdsman-converters/pull/6076

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-14 by Koenkk

almirus, 2023-10-10:
Hi! @Koenkk Please add error code display, especially for overheating. Thank you.
```
{
                        "abilityId": 9,
                        "accessMode": "ro",
                        "code": "fault",
                        "description": "1，告警\\n2，大端模式，HEX,4字节\\n3，报文格式\\n按照完整4个字节,对应的告警位上报告警；bit位，1表示有告警，0表示无告警；如果没用的告警位，补0。高位无定义，也补0；\\n举例：0X00 00 00 09,表示当前发生了漏电自检不正常告警和短路告警,无其它告警。\\n4，通信逻辑\\n只要有1个告警发生，就完整4个字节上报给面板；只要有1个告警关闭，就完整4个字节上报给面板，更新面板显示。",
                        "extensions": {
                            "iconName": "icon-baojing",
                            "attribute": "128"
                        },
                        "name": "故障告警",
                        "typeSpec": {
                            "label": ["short_circuit_alarm", "surge_alarm", "overload_alarm", "leakagecurr_alarm", "temp_dif_fault", "fire_alarm", "high_power_alarm", "self_test_alarm", "ov_cr", "unbalance_alarm", "ov_vol", "undervoltage_alarm", "miss_phase_alarm", "outage_alarm", "magnetism_alarm", "credit_alarm", "no_balance_alarm"],
                            "maxlen": 17,
                            "type": "bitmap",
                            "typeDefaultValue": 0
                        }
                    },
```

[image]

⬇️⬇️⬇️
<details>
  <summary>all DPs</summary>
```
{
        "modelId": "000002yenr",
        "services": [{
                "actions": [],
                "code": "",
                "description": "",
                "events": [],
                "name": "默认服务",
                "properties": [{
                        "abilityId": 1,
                        "accessMode": "ro",
                        "code": "total_forward_energy",
                        "description": "",
                        "extensions": {
                            "iconName": "icon-dp_lightning",
                            "attribute": "128"
                        },
                        "name": "累计正向有功电能",
                        "typeSpec": {
                            "max": 99999999,
                            "min": 0,
                            "scale": 2,
                            "step": 1,
                            "type": "value",
                            "typeDefaultValue": 0,
                            "unit": "kW·h"
                        }
                    }, {
                        "abilityId": 2,
                        "accessMode": "ro",
                        "code": "cur_neutral",
                        "description": "",
                        "extensions": {
                            "iconName": "icon-power1",
                            "attribute": "128"
                        },
                        "name": "累计反向有功电能",
                        "typeSpec": {
                            "max": 99999999,
                            "min": 0,
                            "scale": 2,
                            "step": 1,
                            "type": "value",
[... 138 lines cut ...]
                            "maxlen": 128,
                            "type": "raw"
                        }
                    }, {
                        "abilityId": 18,
                        "accessMode": "rw",
                        "code": "alarm_set_2",
                        "description": "1，告警阈值使能控制及设置2  \\n2，HEX格式，大端模式\\n3，报文格式\\n每个告警由4个字节组成，第1个字节（表示该告警存在）；第2个字节（发生此告警时，断路器控制模式（0X01拉闸，0X00不动作只告警）；第3、4字节：设置告警阈值\\n举例：0X01 01 00 3C 03 01 00 F5,表示该断路器支持过流告警，发生该告警时，拉闸动作，动作阀值60A。支持过压告警，发生该告警时，拉闸动作，动作阀值245V\\n4，通信逻辑\\nA，配网成功后，按照上面的格式，主动上报给云平台，面板知道哪些告警是断路器当前支持的。设置不成功，重新配网再上报。\\nB，断路器上电时，也上报一次。\\nC，面板根据协议下发告警设置。",
                        "extensions": {
                            "iconName": "icon-dp_warming",
                            "attribute": "128"
                        },
                        "name": "告警设置2",
                        "typeSpec": {
                            "maxlen": 128,
                            "type": "raw"
                        }
                    }
                ]
            }
        ]
    }
```

</details>

almirus, 2023-10-10:
@Koenkk I'm sorry, I don't know TypeScript very well.

elektrinis, 2023-12-07:
Thank you for your work.
Currently it is reporting values every 5 seconds. Is it possible to change to 1s, etc?

ishehov, 2024-01-19:
Is it possible to make it report values not that frequent though?

almirus, 2024-01-19:
@ishehov 
[image]

Debounces messages of this device. When setting e.g. debounce: 1 and a message from a device is received, Zigbee2MQTT will not immediately publish this message. But it will combine it with other messages that are received max 1 seconds apart from each other. So there needs to be "N second of silence" from the device before the combined message is sent out.

For example debounce: 5 means that there needs to be "5 seconds of silence" from the device, before combined messages is sent out.