Koenkk/zigbee2mqtt #18562: Can't connect to MQTT broker on latest-dev branch.

adorobis, 2023-08-08:
### What happened?

After updating to `latest-dev` container image z2m fails to start with the following error:
```
info  2023-08-08 21:52:00: Connecting to MQTT server at mqtt://10.144.1.103:1883
debug 2023-08-08 21:52:00: Using MQTT keepalive: 60
debug 2023-08-08 21:52:00: Using MQTT login with username: username
debug 2023-08-08 21:52:00: MQTT reject_unauthorized set false, ignoring certificate warnings.
error 2023-08-08 21:52:00: MQTT failed to connect, exiting...
info  2023-08-08 21:52:00: Stopping zigbee-herdsman...
info  2023-08-08 21:52:01: Stopped zigbee-herdsman
```
Works again ok after returning to `latest` tag.

### How to reproduce it (minimal and precise)

Pull latest-dev image with the following id:
sha256:460765ca0a8d7dfe49af3b125340fa0cd3317f0f5c27693e8f073aa32e99b140

### Zigbee2MQTT version

1.32.2-dev (commit #7f05f31)

### Adapter firmware version

0x26780700

### Adapter

Conbee II

### Debug log

debug 2023-08-08 21:51:59: Loaded state from file /app/data/state.json
info  2023-08-08 21:51:59: Logging to console and directory: '/app/data/log/2023-08-08.21-51-59' filename: log.txt
debug 2023-08-08 21:51:59: Removing old log directory '/app/data/log/2023-08-08.21-49-39'
info  2023-08-08 21:51:59: Starting Zigbee2MQTT version 1.32.2-dev (commit #7f05f31)
info  2023-08-08 21:51:59: Starting zigbee-herdsman (0.17.3)
debug 2023-08-08 21:51:59: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null,"disableLED":false},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"adapter":"deconz","baudRate":38400,"path":"/dev/ttyACM0"}}'
info  2023-08-08 21:52:00: zigbee-herdsman started (resumed)
info  2023-08-08 21:52:00: Coordinator firmware version: '{"meta":{"maintrel":0,"majorrel":38,"minorrel":120,"product":0,"revision":"0x26780700","transportrev":0},"type":"ConBee2/RaspBee2"}'
debug 2023-08-08 21:52:00: Zigbee network parameters: {"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754}
info  2023-08-08 21:52:00: Currently 48 devices are joined:
info  2023-08-08 21:52:00: Zigbee: disabling joining new devices.
info  2023-08-08 21:52:00: Connecting to MQTT server at mqtt://10.144.1.103:1883
debug 2023-08-08 21:52:00: Using MQTT keepalive: 60
debug 2023-08-08 21:52:00: Using MQTT login with username: tasmota
debug 2023-08-08 21:52:00: MQTT reject_unauthorized set false, ignoring certificate warnings.
error 2023-08-08 21:52:00: MQTT failed to connect, exiting...
info  2023-08-08 21:52:00: Stopping zigbee-herdsman...
info  2023-08-08 21:52:01: Stopped zigbee-herdsman

VonOx, 2023-08-08:
I confirm

kaysan13, 2023-08-08:
Same problem
while waiting for the correction I went back to: koenkk/zigbee2mqtt:latest

Gnarfoz, 2023-08-09:
Probably this commit? https://github.com/Koenkk/zigbee2mqtt/pull/18459/commits/fc822ba4627500753969e57c24383a39b2327073

Koenkk, 2023-08-09:
Fixed!

closed 2023-08-09 by Koenkk

adorobis, 2023-08-09:
Many thanks for quick fix! Works again :)