Koenkk/zigbee2mqtt #18563: [New device support]: EGLO 31747 (Motion sensor)

tonnyongit, 2023-08-09:
### Link

https://cdn.hornbach.de/data/shop/D04/001/780/492/701/380/10555774_Doc_02_AT_20230511141659.pdf

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c1387b264ca45c","nwkAddr":28872,"manufId":4417,"epList":[1,3,4],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"epId":3,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"epId":4,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1691537248303,"defaultSendRequestWhen":"immediate"}

### Comments

 Hello,

I'd like to help add this device but I'm stuck on adding support for new devices to zigbee-herdsman-converters. Can someone explain to me what to do this is the first time I do this.

I have extracted the following from the debug log:

Debug 2023-08-09 01:53:07Received Zigbee message from '0xa4c1387b264ca45c', type 'commandMoveToLevelWithOnOff', cluster 'genLevelCtrl', data '{"level":254,"transtime":0}' from endpoint 1 with groupID 32777
Debug 2023-08-09 01:53:07Received Zigbee message from '0xa4c1387b264ca45c', type 'commandMoveToColorTemp', cluster 'lightingColorCtrl', data '{"colortemp":438,"transtime":0}' from endpoint 1 with groupID 32777
Debug 2023-08-09 01:53:07Received Zigbee message from '0xa4c1387b264ca45c', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32777
Debug 2023-08-09 01:53:37Received Zigbee message from '0xa4c1387b264ca45c', type 'commandOff', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 32777

I get stuck trying to create the file for zigbee-herdsman converters. I do not know what to enter. Or do I have to contact the supplier to find out? I find it suspiciously similar to the 99106, which has already been cited https://github.com/Koenkk/zigbee2mqtt/issues/18519. Mine is not recognized like this but it is also a different item number

Here's an English version of the product sheet:
[31747.pdf](https://github.com/Koenkk/zigbee2mqtt/files/12297224/31747.pdf)

tonnyongit, 2023-08-09:
Well today I continued trying to get a configuration done so I did some debugging and suddenly it is recognized as an AwoX 33951/33948. I have included a copy of the database

Database entry:
`{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c1387b264ca45c","nwkAddr":28872,"manufId":4417,"manufName":"AwoX","powerSource":"Battery","modelId":"TLSR82xx","epList":[1,3,4],"endpoints":{"1":{"profId":260,"epId":1,"devId":263,"inClusterList":[0,3,4,4096],"outClusterList":[0,3,4,5,6,8,768,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TLSR82xx","manufacturerName":"AwoX","powerSource":3,"zclVersion":3,"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017"}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":263,"inClusterList":[65360,65361],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":263,"inClusterList":[0,3,4,1024,1030,4096,32],"outClusterList":[0,3,4,1024,1030,4096,18778,16967],"clusters":{"genBasic":{"attributes":{"appVersion":0,"hwVersion":0}},"genPollCtrl":{"attributes":{"checkinInterval":14400}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0026b87fc4","endpointID":1}],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1691583536395,"defaultSendRequestWhen":"active","checkinInterval":3600}
`

Today I'm going to see if I can get further with the external converter file. I've looked at a few others and I suspect I can get further with this one

External converter:
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TLSR82xx'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '31747', // Vendor model number, look on the device for a model number
    vendor: 'EGLO', // Vendor of the device (only used for documentation and startup logging)
    description: 'CROSSLINK.Z Motion sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.occupancy(), e.illuminance(), e.illuminance_lux()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

tonnyongit, 2023-08-09:
yes that's right first comes commandOn when motion is detected and after x time after that comes commandOff. 

But I also have the idea that he only gives a command when the lux falls within a certain threshold. If I put it in the daylight it won't work but if I put my hand in front of it and make sure no light can reach it it will send messages or I see them in the debug log

tonnyongit, 2023-08-09:
yes, it does look like that. Are these settings that you can send to the device? By the way, it's great that you responded so quickly, but can you advise me on how I can move forward or can you help me. I don't know how I can get further, my knowledge in this area needs to be refined a bit haha

Koenkk, 2023-08-10:
> Are these settings that you can send to the device?

I don't expect you can unless it is explained in the manual of the device. The IKEA motion sensors works the same and has a physical button to change the light sensitivity.

tonnyongit, 2023-08-18:
Well, I've been testing for a few evenings and you are indeed right that this is not an ideal sensor. I returned the sensor to the store and exchanged it for a known working model from a different supplier. Thank you very much for the quick responses, but I don't have time to experiment much with this sensor anymore, unfortunately

Koenkk, 2023-08-18:
No problem, I will close this.

closed 2023-08-18 by Koenkk

gielfeldt, 2023-10-07:
Sorry for re-opening this. I have this sensor, and I trying to add it, but without any luck.

Wrt only working in the dark, isn't this an outdoor dusk sensor to begin with? In my mind, it would make sense that it only works when it's dark. If you cannot adjust the lighting threshold, isn't that just a "feature" of this specific device?

> We can of course support the occupancy true/false when it is dark but this is of course not ideal.

This is precisely what I need. Would it not make sense to support this device, even though its feature-set might no be as broad as other sensors?

gielfeldt, 2023-10-10:
I got it somewhat working with:

```javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['EPIR_Zm'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
    model: '31747', // Vendor model number, look on the device for a model number
    vendor: 'EGLO', // Vendor of the device (only used for documentation and startup logging)
    description: 'Connect.Z Motion sensor', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    fromZigbee: [fz.command_on, fz.command_off, fz.command_move_to_level, fz.command_move_to_color_temp],
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.occupancy()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;
```

But I'm getting way too many false events with on/off.

I'll investigate some more.

Neuroquila-n8fall, 2024-07-04:
I think we can't fix this shoddy piece of hardware... I got one too today and it is being discovered as a light bulb. So far any overrides didn't work. Also the fingerprint method taken from a different issue isn't working either.
The pairing process itself is wonky as hell and i only succeeded to pair it properly two times out of dozens of attempts. It also craps out after a few seconds after being configured by the app via bluetooth but stays online on zigbee. I'm not the kind of person which throws in the towel early but these things just won't work as they should and are a bad example of "zigbee certified" devices.
I don't want to spoil the fun either but as it stands now I can only recommend staying away from AwoX things at all...