Koenkk/zigbee2mqtt #18567: [New device support]: Sonoff ZBmini (neutral, older version)

gyenesadam, 2023-08-09:
### Link

https://sonoff.tech/product/diy-smart-switches/zbmini/

### Database entry

{"id":15,"type":"Router","ieeeAddr":"0x00124b0024c5d069","nwkAddr":64749,"manufId":4742,"epList":[1,242],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"epId":242,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":false,"meta":{},"lastSeen":1691606626957,"defaultSendRequestWhen":"immediate"}

### Comments

I tried to connect to the first zbmini, but i always get fail to interview message, because the device not supported.
Unfortunatelly i haven't got any knowladge for make an external converter, please, somebody could help me?

May the screenshots can help.
[image]
[image]
[image]

gyenesadam, 2023-08-10:
Current version: 1.32.2-1

Koenkk, 2023-08-10:
The device did not pair correctly (interview completed = false). Try force removing the device, restart z2m and re-pair the device. If that doesn't help, re-pair while having the device close to the coordinator.

gyenesadam, 2023-08-13:
I going to try it tomorrow.

gyenesadam, 2023-08-14:
> The device did not pair correctly (interview completed = false). Try force removing the device, restart z2m and re-pair the device. If that doesn't help, re-pair while having the device close to the coordinator.

Well, I tried.
The device next to the coordinator, after enable connect, instantly start interview, but interview still failed. 
Have you got any idea?

Koenkk, 2023-08-14:
Try to other steps from https://www.zigbee2mqtt.io/guide/faq/#why-does-my-device-not-or-fail-to-pair

gyenesadam, 2023-08-14:
Got it!

The solution if anybody got  the same problem:

Enter "pairing mode" on ZBMINI.
Enable Join in Z2M.
The same time push ZBMINI's button every one sec.

After that the interview complete succesfully!

Thanks for help!!!

closed 2023-08-14 by gyenesadam