Koenkk/zigbee2mqtt #18570: Zigbee2MQTT cannot support new Tuya Zigbee VRF Air Conditioning Controller

Bpazy, 2023-08-10:
### What happened?

In the Tuya APP, adding that VRF device to the network first reports the device itself (which Tuya calls the `VRF Gateway`), and then the `VRF Gateway` discovers several other connected devices.

Preview video：
https://github.com/Koenkk/zigbee2mqtt/assets/9838749/4a054f67-e63e-40f3-b77d-72af5f2bfd67

But in the Zigbee2MQTT frontend, only the `VRF gateway` can be discovered, which prevents me from adding a new Tuya device.

And in Tuya IoT Platform, I can only find Data Points for `VRF gateway`, I can't find other connected devices.

I'm not sure why this is, but I'm guessing that the VRF Air Conditioning Controller is a physical device, and the other devices connected to it are what it virtualizes and reports to the Tuya platform.

Do you have any solution for this please?

### What did you expect to happen?

Zigbee2MQTT can discover the VRF Air Conditioning Controller and **the devices connected to it**.

### How to reproduce it (minimal and precise)

It's possible to reproduce it as I described above. but I think it's device related, so it should be difficult for others to reproduce it.

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

Not sure

### Adapter

ezsp

closed 2023-09-17 by github-actions[bot]

mayongze, 2023-10-24:
这个vrf网关设备添加成功了么, 我也打算买一个

Bpazy, 2023-10-24:
> 这个vrf网关设备添加成功了么, 我也打算买一个

@mayongze 没有添加成功，目前还在用 Tuya 集成顶着。

mayongze, 2023-10-24:
@Bpazy 有兴趣的话可以一起分析下这个vrf的zigbee协议

Bpazy, 2023-10-24:
@mayongze 没问题，但是你怎么确定你的 VRF 网关和我是相同型号的呢？

latel, 2023-11-27:
我用tuya-local可以接入涂鸦网关下的zigbee空调，试了下断了外网也可用。  但这本身没啥意义，我还不如直接用tuya wifi色vrf。[image]

whiteinho, 2024-02-22:
楼主是用的多模网关2做Z2M的网关吗？

Bpazy, 2024-02-22:
> 楼主是用的多模网关2做Z2M的网关吗？

是的

Bpazy, 2024-02-23:
@latel 没搞成。本来打算搞 z2m 适配器来仔细研究下的，但是太忙没时间搞就搁置了，临时先买了 tuya 网关+集成顶着