Koenkk/zigbee2mqtt #18574: [Feature request]: Additional workarounds to unfreeze IKEA lights

lbschenkel, 2023-08-10:
### Is your feature request related to a problem? Please describe

IKEA trådfri lights have a firmware bug in which if they are in the middle of a color/temperature transition, they "freeze" and will not respond to any other Zigbee commands while the transition is ongoing.

An example:
1. set light to 2000K
2. set light to 4000K with a transition of 10s
3. during these 10s the light will stop processing commands (for example: turning off the light)
4. Z2M light state becomes out of sync with the actual light state (for example: Z2M will report light off while the light is still on)
5. after 10s light completes the transition and starts responding to new commands
6. Z2M light state will stay out of sync

Notes:
- This also happens during scene transitions, if they trigger a color/temperature change.
- This *does not* happen for brightness transitions, only color/temperature.
- The light may start responding sooner if it has reached the target color/temperature before the transition time elapses (if it was already at the target color/temperature)

This bug may result in very undesirable behavior. A smart switch may be unable to turn off the light, for example, if the light is in the middle of a long transition (wake up light, flux / circadian / adaptive lighting, etc.) while Z2M will optimistically report that the light is off.

Extreme example: imagine a long transition, such as 30m: the light will simply freeze for the whole duration and you can't even turn it off, except by cutting the power.

### Workaround

There is actually a way to "unfreeze" the light while it is transitioning: sending a  `color_temp_move: stop` command (`lightingColorCtrl.moveColorTemp: 0` at the Zigbee layer). This stops the color/temperature transition and consistently makes the light start responding again.

### Describe the solution you'd like

## Best option

Ideally Z2M would implement the following workaround in [hedsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters) for IKEA lights:

1. it notices that it is about to send a command that will (or could) change the light color/temperature
2. it looks at the transition time for this command, and remembers that the light will potentially be frozen until `now + transition` (let's assume that the time is stored in a per-light state called `frozen_until`)
3. when a new command needs to be sent to the light, it checks if  the light is frozen (`frozen_util` is set and if `now <= frozen_until`). if it is:
  a. it sends `lightingColorCtrl.moveColorTemp: 0` to unfreeze the light
  b. clears `frozen_until`
  c. then it sends the new command (by going back to step 1, which might set `frozen_until` again)

This implies that any new command sent to the light will make it stop transitioning. But I argue that this is much better than the light just silently ignoring all commands, potentially refusing to turn off, and causing the state to get out of sync.

For groups I imagine that the same would apply, as long as one ore more IKEA lights are members of the group. However, setting `frozen_until` in the group *would also set it under the individual IKEA lights*, so subsequent individual commands to the lights would still successfully unfroze them.

I'm not a fan of premature generalization, but if it must then perhaps it could be generalized for all devices in the following way:
- all devices have a `frozen_until` state
- when sending any Zigbee commands to a device, first check `frozen_until` and if applicable, send an `unfreeze()` command before sending the actual command
- only IKEA converters will set `frozen_until`, and only when they detect a potential color/brightness change with a transition
- only IKEA converters will override `unfreeze()` to send `lightingColorCtrl.moveColorTemp: 0`, others do nothing

## Alternative

A simpler but much less effective alternative could be simply to send an unconditional `lightingColorCtrl.moveColorTemp: 0` command at critical points for IKEA lights, at minimum for on/off to make sure that the light can always be turned on/off even if it is frozen.

### Describe alternatives you've considered

I have done the above via custom scripts that I wrote in Home Assistant. But it's a very awkward solution that requires my scripts to be called instead of the regular `light.turn_on/turn_off` services, or wrapping the actual Z2M light under a template light.

Naturally these workarounds won't even apply when interfacing when not using Home Assistant. Doing this in Z2M is a much cleaner solution which benefits everybody.

### Additional context

This is a well-known problem with IKEA lights:
- https://github.com/Koenkk/zigbee2mqtt/issues/1810
- https://github.com/Koenkk/zigbee2mqtt/issues/4316
- https://github.com/dresden-elektronik/deconz-rest-plugin/issues/894
- https://www.reddit.com/r/tradfri/comments/au903n/firmware_bugs_in_ikea_bulbs/

Koenkk, 2023-08-10:
Nice analysis! Before fixing something, I first want to see how many people are affected by this (so please, if you are affected please put a 👍 on the OP)

lbschenkel, 2023-08-10:
A correction: it's better to use `brightness_move: stop` then `color_temp_move: stop` to unfreeze the lights. Both work but the latter is not recognized by non-CT (RGB) lights. I wrote the initial issue from memory but I re-checked my scripts and I'm actually sending the former.

lbschenkel, 2023-08-16:
@Koenkk: My knowledge of JS and the Z2M codebase is limited, but I was able to patch the source code to do a PoC of what I'm proposing and documenting what I could figure out with my IKEA lights.

First I've done it directly in the `converters` to check that it works but then I refactored it so I can run it in "production" via an external converter without touching anything else. The result is a bit crude and not very elegant and requires a lot of decoration, but it does work and it's the best way I could find to illustrate the logic. Naturally a "proper" solution will look different. My gut feeling is that putting this logic in the generic command codepath and opted in by `options` in the device declaration in `module exports` should be very clean.

I also think this should be gated by an user-facing setting (as `color_sync` or `state_action`) , so users can opt out or opt in to this behavior. Not sure which default is best, but for sure I know what I will use :-).

Anyway, I have this in my house and it is working beautifully so far. I will report any issues that I encounter. 

```js
////////////////////////////////////////////////////////////////////////////////
// copied from ikea.js:

const extend = require('zigbee-herdsman-converters/lib/extend');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const utils = require('zigbee-herdsman-converters/lib/utils');

const bulbOnEvent = async (type, data, device, options, state) => {
    /**
     * IKEA bulbs lose their configured reportings when losing power.
     * A deviceAnnounce indicates they are powered on again.
     * Reconfigure the configured reoprting here.
     *
     * Additionally some other information is lost like
     *   color_options.execute_if_off. We also restore these.
     *
     * NOTE: binds are not lost so rebinding is not needed!
     */
    if (type === 'deviceAnnounce') {
        for (const endpoint of device.endpoints) {
            for (const c of endpoint.configuredReportings) {
                await endpoint.configureReporting(c.cluster.name, [{
                    attribute: c.attribute.name, minimumReportInterval: c.minimumReportInterval,
                    maximumReportInterval: c.maximumReportInterval, reportableChange: c.reportableChange,
                }]);
            }
        }

        // NOTE: execute_if_off default is false
        //       we only restore if true, to save unneeded network writes
        if (state !== undefined && state.color_options !== undefined && state.color_options.execute_if_off === true) {
            device.endpoints[0].write('lightingColorCtrl', {'options': 1});
        }
        if (state !== undefined && state.level_config !== undefined && state.level_config.execute_if_off === true) {
            device.endpoints[0].write('genLevelCtrl', {'options': 1});
        }
        if (state !== undefined && state.level_config !== undefined && state.level_config.on_level !== undefined) {
            let onLevel = state.level_config.on_level;
            if (typeof onLevel === 'string' && onLevel.toLowerCase() == 'previous') {
                onLevel = 255;
            } else {
                onLevel = Number(onLevel);
            }
            if (onLevel > 255) onLevel = 254;
            if (onLevel < 1) onLevel = 1;

            device.endpoints[0].write('genLevelCtrl', {onLevel});
        }
    }
};

const tradfriExtend = {
    light_onoff_brightness: (options = {}) => ({
        ...extend.light_onoff_brightness(options),
        ota: ota.tradfri,
        onEvent: bulbOnEvent,
    }),
    light_onoff_brightness_colortemp: (options = {colorTempRange: [250, 454]}) => ({
        ...extend.light_onoff_brightness_colortemp(options),
[... 114 lines cut ...]
        zigbeeModel: ['TRADFRI bulb E27 WS opal 1000lm', 'TRADFRI bulb E26 WS opal 1000lm'],
        model: 'LED1732G11',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E27 1000 lumen, dimmable, white spectrum, opal white',
        extend: tradfriExtend.light_onoff_brightness_colortemp({ colorTempRange: [250, 454] }),
        toZigbee: unfreezeSupport(tradfriExtend.light_onoff_brightness_colortemp({ colorTempRange: [250, 454] }).toZigbee, unfreezeMechanisms.genLevelCtrl)
    },
    {
        zigbeeModel: ['TRADFRI bulb E27 WS opal 980lm', 'TRADFRI bulb E26 WS opal 980lm', 'TRADFRI bulb E27 WS\uFFFDopal 980lm'],
        model: 'LED1545G12',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb E26/E27 980 lumen, dimmable, white spectrum, opal white',
        extend: tradfriExtend.light_onoff_brightness_colortemp(),
        toZigbee: unfreezeSupport(tradfriExtend.light_onoff_brightness_colortemp().toZigbee, unfreezeMechanisms.genLevelCtrl)
    },
    {
        zigbeeModel: ['TRADFRI bulb GU10 WS 400lm'],
        model: 'LED1537R6/LED1739R5',
        vendor: 'IKEA',
        description: 'TRADFRI LED bulb GU10 400 lumen, dimmable, white spectrum',
        extend: tradfriExtend.light_onoff_brightness_colortemp(),
        toZigbee: unfreezeSupport(tradfriExtend.light_onoff_brightness_colortemp().toZigbee, unfreezeMechanisms.genLevelCtrl)
    }
];
```

lbschenkel, 2023-08-16:
This is a crude test case:
```sh
#!/bin/bash

pub() {
    mosquitto_pub -h HOST -u USER -P PASSWORD \
    -t zigbee2mqtt/LIGHT/set -m "$1"
}

pub '{"state": "on", "brightness": 2, "color_temp": 150, "transition": 0}'
sleep 2
pub '{"brightness": 255, "color_temp": 500, "transition": 5}'
sleep 3
pub '{"state": "off"}'
```
Without the unfreeze code, the light is frozen for 5s and will ignore any commands in that timeframe -- so it will not actually turn off even though Z2M thinks it did and updates the state. With the unfreeze code, the light will stop transitioning and turn off.

lbschenkel, 2023-08-17:
An update: I realized that I've made a mistake when evaluating the outcome of the "unfreezing mechanisms". **All** lights support `lightingColorCtrl.moveColorTemp: 0`, including the color lights that do not natively support temperature. What was actually producing the error I've seen was this read here: https://github.com/Koenkk/zigbee-herdsman-converters/blob/3a948abc5cc94c0a4be2ede464834431d4cd4148/src/converters/toZigbee.js#L762, which was happening with an early implementation of my workaround.

So, as far as I can see, both commands are equivalent and will successfully unfreeze any IKEA light that I have (a mixture of GU10/E14/E27 WS and color):
- `lightingColorCtrl.moveColorTemp: 0`
- `genLevelCtrl.stop`

protyposis, 2023-08-24:
Fixing this would be advantageous for users of Adaptive Lighting, which uses very long color transitions, e.g., 45 seconds. The currently recommended workaround for long color transitions on Ikea bulbs is basically simulating it by "flooding" the Zigbee network with sequences of frequent instant color changes (https://github.com/basnijholt/adaptive-lighting#bulb-bulb-specific-issues).

lbschenkel, 2023-09-24:
Please don't close this; @Koenkk what are your thoughts on the above?

Koenkk, 2023-09-24:
given the thumbs up on the OP, more people are interested in this. Could you make a PR? Then we can iterate from there.

lbschenkel, 2023-09-27:
All right, I will create a proper PR when able (but not with the above code, which was a roundabout way designed to work purely as an external converter to allow others to test it).

francisp2, 2023-10-28:
?

PetziAt, 2023-12-24:
any news on this topic? I'm in favour ;)

lbschenkel, 2023-12-24:
I  am going to submit a PR, but haven't had the time yet. Some other things came up that made me busy. But it'll come.

boesing, 2024-01-02:
I am kinda affected. I did try to get color temp + brightness be modified with transitions (2 sec) but it seems that sometimes, only temperature is applied while brightness remains the old value. I do set 3 values when modifying: 
1. transition time (2 sec)
2. color temp value
3. brightness level

I already realized that this happens and I tried to workaround with node-red "repeat message" where I re-send all 3 values in total for 4 times with 3 seconds delay but that does not change anything. Sometimes it works but most of the time, it does not.

lbschenkel, 2024-01-02:
> I already realized that this happens and I tried to workaround with node-red "repeat message" where I re-send all 3 values in total for 4 times with 3 seconds delay but that does not change anything. Sometimes it works but most of the time, it does not.

It will never work when you send all values at the same time. The lights don't support that, actually. What really happens is that Z2M splits your request into two Zigbee messages: (1) it sets the brightness level first, without any transition (the lack of transition here is an explicit IKEA workaround), and then (2) sets the color temperature with the transition. During (2) the lights will be "frozen".

If you change your automation to set only the brightness first with transition/2, then you wait for that time to elapse (+ 200 ms, which I found by experiment to be the time that lets the firmware "settle"), then you set the color with transition/2 (keeping in mind that light will be frozen during this period), it should consistently "work". At least it does for me.

bobrippling, 2024-12-03:
@lbschenkel I'd been about to propose a similar fix myself when I came across this issue - I like your approach, I'm happy to pick this up and work on a PR if you wouldn't mind?

lbschenkel, 2024-12-03:
Please go ahead. Unfortunately I have been far more busy than I anticipated, and I didn't have enough free time.

I also have been gradually replacing my IKEA lights because I'm getting tired of their firmware bugs, so I may end up solving the problem by different means.

But I still have a bunch of them lights, so feel free to ping me if you want to discuss ideas, review or test your fix.

bobrippling, 2024-12-03:
Great, thank you, shall bear that in mind. I'll start from the current herdsman `master` unless you had any work-in-progress branches you'd like me to go off?

lbschenkel, 2024-12-03:
No, basically just the code you can find here.

bobrippling, 2025-01-18:
I've pushed up a WIP PR - I'm not familiar with the codebase, so before I go down the current implementation, are there any pointers / does this look like the correct approach?

bobrippling, 2025-04-09:
Massaged the code into place and I'm pretty happy with it. Your proof-of-concept did the trick perfectly @lbschenkel, I've tested with some bulbs and the unfreeze works - I'm able to turn them off during the transition period.

@Koenkk this issue will be fixed by Koenkk/zigbee-herdsman-converters#8637 if you're happy with it

closed 2025-04-14 by Koenkk

lbschenkel, 2025-05-02:
I tested the new release and the new unfreeze workaround works well (good job @bobrippling), however the lights still freeze on color changes (XY/CIE/RGB) as the workaround did not take those into account. I submitted a new PR: https://github.com/Koenkk/zigbee-herdsman-converters/pull/9272

bobrippling, 2025-05-02:
Ah thanks for checking it out - I don't have any colour bulbs (just `color_temp`) so good catch!

sgtwilko, 2025-05-04:
Hi @lbschenkel 

Since this was released my IKEA lamps are slowly going dimmer with each colour transition.

I'm using a plugin in home assistant to change the colour, but not the brightness, every 30 seconds over the course of the evening to slowly remove the blue.

Since this update I've had to reset the brightness every so often.

Zigbee2mqtt shows the brightness hasn't changed, sometimes if I hit the refresh button it'll show that it's been reduced.
If I command it back to full it does return to full brightness.

Do you need anything from me to help debug?

lbschenkel, 2025-05-04:
> Do you need anything from me to help debug?

I'll try to reproduce it with my lamps. Are you seeing this in a WS or CWS? Can you share the exact script?

Based on what you explained, the theory is that the lights are changing color with a long transition and then freeze, some other command comes that kicks in the unfreezing logic, and a "stop brightness transition" is sent to unfreeze the light. This command is obviously not supposed to change brightness, only cut short the color transition and unfreeze the light, but IKEA firmware being IKEA firmware it's not completely unlikely that when cutting the transition short the actual brightness of the lamp does change. If that's the case it would be an unforeseen side effect of the unfreezing operation — perhaps a different choice of command will work better.

lbschenkel, 2025-05-05:
> Since this was released my IKEA lamps are slowly going dimmer with each colour transition.
> 
> Since this update I've had to reset the brightness every so often.
> 
> Zigbee2mqtt shows the brightness hasn't changed, sometimes if I hit the refresh button it'll show that it's been reduced. If I command it back to full it does return to full brightness.

I can't reproduce. I have made a script that simulates what you have explained: it tries to change only the color of the light and then waits a bit and sends another command (which kicks in the "unfreeze" command). I repeated that 256 times, and the brightness level never changed (either the perceived one or what's Z2M reports, even after a refresh). I've done that with a WS and a CWS light and both never went dimmer.

sgtwilko, 2025-05-07:
Hi @lbschenkel sorry didn't get notified you'd replied.

I'm using the home assistant flux plugin configured like this:
```
- platform: flux
    mode: mired
    disable_brightness_adjust: true
    stop_time: "22:00"
    lights:
...
```

It's configured to adjust the colour of two Lidl and two IKEA lights.

I tested with both types on at the same time and the Lidl lights still work as expected, retaining their brightness, the IKEA lights slowly fade.

Also as zigbee2mqtt shows that the lamps should be at 100% (until I hit refresh in z2m) it could very well be as you suspect another firmware bug that the new code is now hitting.

lbschenkel, 2025-05-07:
> I'm using the home assistant flux plugin configured like this:

OK. According to the `flux` [implementation](https://github.com/home-assistant/core/blob/dev/homeassistant/components/flux/switch.py) it will try to change the light color using a transition of 30 seconds by default (you don't seem to have change it). The adjustment will happen every 30 seconds too (also left as default). I'm not quite sure if this means that the lights would adjust for 30 seconds, then flux would wait 30 seconds, then adjust again; or if lights would adjust for 30 seconds, flux would wait 30 seconds *in parallel*, and then adjust again, which means they'll be adjusting the whole time.

The situation before is that your IKEA lights would be frozen while they were transitioning color (the freezing bug). You wouldn't be able to even turn them off. The light state in Z2M changes because it optimistically assumes the light changed but the light in practice doesn't do anything as it acknowledges but ignores most commands while it's frozen. I would say that at least half the time your lights were in this state, possibly the whole time. But you should have noticed they were not responding in that case? (It would be a good idea to downgrade Z2M to double check this.) 

With this new version, if a new MQTT command is sent to the light while they're in this 30 second freeze then Z2M will send another command first to unfreeze the light. This command is a "stop brightness move", it is not supposed to change the light state at all, but it has a nice side effect for our purposes that it will work cut the transition short (the light will instantly move to the final state) and the light will become "alive" again and able to process incoming commands.

Since you started to see a fade now, the only logical explanation I have is that this additional unfreezing command is actually changing the light brightness somehow. 

But two things must be happening simultaneously for this to make sense: this command is indeed doing this (big if), *and* you are actively trying to send other commands to the lights while they're on. That's because if no other command is being sent apart from the ones `flux` sends, then no additional unfreeze commands should have been sent either. *But*: it could be that the flux timing is so tight (30 seconds transition, 30 seconds parallel wait) that when flux send a new change the timing is still overlapping with the period that Z2M considers the light being frozen and does send the unfreeze command on each new adjustment.

> I tested with both types on at the same time and the Lidl lights still work as expected, retaining their brightness, the IKEA lights slowly fade.

It's puzzling. I could not reproduce this. As I mentioned in my other comment: I have created a script that does mostly the same what flux is doing (color transition without a brightness change), I send another command midway to trigger the unfreeze logic, and I repeated that 256 times. I never saw any brightness change, they remained visibly at the same level. Tested with both WS and CWS lights. Since there are 255 brightness levels, if the level was decreasing by 1 (minimum amount) each time the light should have slowly faded to minimum level.

Does that happen consistently for you? How much time does it take for you to notice a visible fade? Minutes, hours? I guess I need to try to set up `flux` with the same lights and see what happens.

So I have a little bit of homework for you for now:
- can you tell me the exact models of your IKEA lights?
- can you downgrade Z2M and check if the lights are frozen for at least 50% of the time, possibly 100%? (try to turn them off, if they're frozen they will stay on)
- now, with the latest Z2M, can you change flux transition and/or interval so interval is at least 2 seconds more than the transition (this makes sure there's no timing overlap) and report if the fading problem has disappeared?