Koenkk/zigbee2mqtt #18582: Wrong device identification

MikhailBaranov, 2023-08-11:
### What happened?

EARU EAKCB-T-M-Z is showed as TUYA TS011F_plug_1 so no options power_threshold, power_breaker, over_current_threshold, over_current_breaker, over_voltage_threshold, over_voltage_breaker, under_voltage_threshold, under_voltage_breaker are available.

[image]

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

EZSP v9

albertoneno, 2023-08-20:
Estoy en la misma situación. Me sucede con el modelo:
EAKCB-Z-M
Alguna idea?

I'm in the same situation. It happens to me with the model:
EAKCB-Z-M
Any ideas?

MikhailBaranov, 2023-09-08:
1.33.0 - problem solved.

closed 2023-09-08 by MikhailBaranov