Koenkk/zigbee2mqtt #18587: Router firmware not flashing on SONOFF Zigbee 3.0 USB Dongle Plus

ntlgr, 2023-08-11:
### What happened?

Unable to flash SONOFF Zigbee 3.0 USB Dongle Plus by ITead with the latest router firmware.
Used TI FlashProgrammer, the latest firmware 20221102 does not flash on the device. 
Luckily had the previous version and was able to reflash back to 20220125.

### What did you expect to happen?

Flashing to succeed

### How to reproduce it (minimal and precise)

Boot SONOFF Zigbee 3.0 USB Dongle Plus in bootloader mode 

### Zigbee2MQTT version

1.32.2-1

### Adapter firmware version

20220125

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Debug log

N/A

Tonio16, 2023-08-11:
Hello

https://github.com/Koenkk/Z-Stack-firmware/issues/397

Antoine

ntlgr, 2023-08-14:
Thank you and sorry for the duplicate.

closed 2023-09-21 by github-actions[bot]