Koenkk/zigbee2mqtt #18589: [New device support]:

thomasb86, 2023-08-11:
### Link

https://www.amazon.fr/Thermostat-intelligent-aliment%C3%A9-t%C3%A9l%C3%A9commande-programmabilit%C3%A9/dp/B0CCXRC7VK/ref=sr_1_1?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=3JD03H4JC6SXP&keywords=Tuya+Thermostat+intelligent+Tuya+Zigbee+aliment%C3%A9+par+batterie+t%C3%A9l%C3%A9commande+commande+vocale+programmabilit%C3%A9&qid=1691758598&sprefix=tuya+thermostat+intelligent+tuya+zigbee+aliment%C3%A9+par+batterie+t%C3%A9l%C3%A9commande+commande+vocale+programmabilit%C3%A9%2Caps%2C70&sr=8-1

### Database entry

{"id":19,"type":"EndDevice","ieeeAddr":"0xb43a31fffe84a84a","nwkAddr":52226,"manufId":4098,"manufName":"_TZE200_viy9ihs7","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"i>

### Comments

Hello, 
i bought this tuya zigbee thermostat and it is recognized by zigbee2mqtt but not supported.
i'd like to use it with home assistant but it's impossible.
this is the 1st time i'm posting for a peripheral i hope i did the right thing.
thank you

ALPF5555, 2023-09-08:
Hello thomasb86 did you ever get any feedback on the support of your Tuya thermostat and how it is supported in zigbee2mqtt. I have the same device and it can be found but no support.

cerme0, 2023-09-11:
Just got the device today. I tried to follow "support new devices" but I'm not this good and I'm unable to make anything work.
Is it possible to receive any assistance?

This device is one of the few easily available wich is battery powered and support zigbee

cerme0, 2023-09-14:
Today I received a Tuya hub, after pairing I was able to read tuya datapoints.
Taking example/inspiration from another thermostat js and checking the guide I was able to write down an external converter. Due to what I needed and my "coding" ability and knowledge, I was able to only make climate function work and expose.

I'm attaching the .js file I've created. 
[TS0601.zip](https://github.com/Koenkk/zigbee2mqtt/files/12612824/TS0601.zip)
For make it work you also have to add this lines into configuration.yaml:
external_converters:
  - TS0601.js

If anyone is willing to put some extra work here is the full datapoints list:
{"1":"Switch",
"2":"Set temperature",
"3":"Current temperature",
"4":"Mode",
"9":"Child lock",
"11":"Fault alarm",
"15":"Set temperature upper limit",
"16":"Lower Temp",
"19":"Temperature correction",
"101":"Working status",
"102":"Frost protection function",
"103":"Reset",
"104":"Workday setting",
"107":"Output startup temperature difference",
"109":"Weekly procedure",
"110":"Backlight brightness"}

Hope to help someone, also is someone better then me can ultimate the work it would be great!

Liionboy, 2023-10-02:
any news? when will be supported by Z2M ?

cerme0, 2023-10-02:
> any news? when will be supported by Z2M ?

quirk should be already up and ready. I removed my external converter and now thermostat is reconfigured natively from z2m. It's not 100% implemented but core functions works

Liionboy, 2023-10-02:
i dont know what exactly to do with that quirk, i make what write in Readme but not work after restart

Liionboy, 2023-10-03:
I saw now is working in Z2M but there are a problem with Running State is NULL:

hvac_modes:
  - 'off'
  - heat
min_temp: 5
max_temp: 35
target_temp_step: 0.5
preset_modes:
  - none
  - auto
  - manual
current_temperature: 24.3
temperature: 22
preset_mode: auto
child_lock: UNLOCK
current_heating_setpoint: 22
deadzone_temperature: 10
device:
  applicationVersion: 65
  dateCode: ''
  friendlyName: Termostat Test
  hardwareVersion: 1
  ieeeAddr: '0x3425b4fffe4c52d1'
  manufacturerID: 4098
  manufacturerName: _TZE200_viy9ihs7
  model: ZWT198
  networkAddress: 52768
  powerSource: Battery
  stackVersion: 0
  type: EndDevice
  zclVersion: 3
frost_protection: 'OFF'
last_seen: '2023-10-03T21:56:43+03:00'
linkquality: 149
local_temperature: 24.3
local_temperature_calibration: null
preset: auto
running_state: null
system_mode: heat
upper_temp: null
friendly_name: Termostat Test
supported_features: 17

From my research i saw this state need to be on idle or heat:

https://www.zigbee2mqtt.io/devices/ZWT198.html#tuya-zwt198

running_state: The current running state. Possible values are: idle, heat. Reading (/get) this attribute is not possible.

almazikv, 2023-11-03:
I have also bought this thermostat and now wait for supper in z2m

joji90, 2024-02-17:
i have the same issue , any help is appreciated.

closed 2025-01-14 by github-actions[bot]