Koenkk/zigbee2mqtt #18591: [Wrong device]: Eglo Fueva-Z 98847 detected as AwoX 33957

Micr0mega, 2023-08-11:
### Link

https://www.eglo.com/en/surface-mounted-light-fueva-z-98847.html

### Model

Eglo Fueva-Z

### Description

Ceiling light

### Vendor

Eglo

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/98847.jpg

### Database entry

{"id":4,"type":"Router","ieeeAddr":"0xa4c138a22ad1bd7c","nwkAddr":40182,"manufId":4417,"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx","epList":[1,3,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096,64599,10],"outClusterList":[6],"clusters":{"genBasic":{"attributes":{}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":153,"colorTempPhysicalMax":370}}},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":4751,"epId":3,"devId":268,"inClusterList":[65360,65361,4],"outClusterList":[65360,65361],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":2,"stackVersion":2,"hwVersion":60,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":88764544},"lastSeen":1691772377865,"defaultSendRequestWhen":"immediate"}

### Notes

I have updated the device firmware as described in discussion https://github.com/Koenkk/zigbee2mqtt/discussions/18366#discussion-5419242. When that did not help, I have re-paired the device twice once without removing it from Z2M, and the second time I did remove it. Both times it still got detected as the AwoX 33957.

I am on Zigbee2MQTT version 1.32.2.
Let me know if any additional information is needed, thanks in advance!

Koenkk, 2023-08-12:
This device is a whitelabel with exactly the same manufacturer name/modelid as the real AwoX bulbs, therefore we cannot distinguish it.

```
"manufName":"AwoX","powerSource":"Mains (single phase)","modelId":"TLSR82xx",
```

closed 2023-08-12 by Koenkk

Micr0mega, 2023-08-12:
Understood, thank you for your fast response.
It looks like they are functionally almost equal anyway, except that the `colorTempRange` max is 370 for this device instead of 454. Is there a way to override this, or maybe to force the device model?

Koenkk, 2023-08-12:
No, because we cannot distinguish this device from the other models.