Koenkk/zigbee2mqtt #18594: CO2 sensor detected with support for temperature and humidity which it hasn't.

gdave321, 2023-08-12:
### What happened?

I have this CO2 sensor from aliexpress  https://www.aliexpress.com/item/1005004058860966.html, it works just fine. However, it is detected with support for features temperature and humidity, but this sensor only has co2 support.

In the database it has this entry:
{"id":12,"type":"Router","ieeeAddr":"0xa4c13888.......","nwkAddr":9184,"manufId":4417,"manufName":"_TZE200_ogkdpgy2","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"��^,i","65506":54,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_ogkdpgy2","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1691829167663,"defaultSendRequestWhen":"immediate"}

### What did you expect to happen?

Only support indicated for co2.

### How to reproduce it (minimal and precise)

Add this sensor to zigbee network.

### Zigbee2MQTT version

1.31.2

### Adapter firmware version

20220219

### Adapter

Slaesh' stick

Koenkk, 2023-08-12:
@yuriybabiy since you added this device in https://github.com/Koenkk/zigbee2mqtt/issues/11581, can you confirm this?

yuriybabiy, 2023-08-13:
1. Good afternoon. The CO2 sensor I use does not analyze temperature and
humidity.
Only carbon dioxide.
2. Big request: This sensor spams data every second. It would be good to
limit the arrival of data from it (for example, one arrival for 30 seconds
is enough for me).
Thank you!

yuriybabiy, 2023-08-13:
Yes, the sensor has been working for so long that I don't remember. Thank you!

gdave321, 2023-09-15:
As originator agrees this unit only supplies co2, I think this patch should handle this:

--- tuya.js.ORIG        2023-06-24 13:19:46.507444402 +0200
+++ tuya.js     2023-09-15 10:45:35.231479522 +0200
@@ -1118,7 +1118,7 @@
         description: 'NDIR co2 sensor',
         fromZigbee: [fz.legacy.tuya_air_quality],
         toZigbee: [],
-        exposes: [e.temperature(), e.humidity(), e.co2()],
+        exposes: [e.co2()],
     },
     {
         fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_7bztmfm1' }],

gdave321, 2023-09-15:
```
--- tuya.js.ORIG        2023-06-24 13:19:46.507444402 +0200
+++ tuya.js     2023-09-15 10:45:35.231479522 +0200
@@ -1118,7 +1118,7 @@
         description: 'NDIR co2 sensor',
         fromZigbee: [fz.legacy.tuya_air_quality],
         toZigbee: [],
-        exposes: [e.temperature(), e.humidity(), e.co2()],
+        exposes: [e.co2()],
     },
     {
         fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_7bztmfm1' }],
```

closed 2023-10-24 by github-actions[bot]