Koenkk/zigbee2mqtt #18597: [Wrong device]: Xiaomi JTQJ-BF-01LM/BW

vit-um, 2023-08-12:
### Link

https://www.aliexpress.com/item/32829080106.html#nav-specification

### Model

YTC4019RT

### Description

Xiaomi MiJia Honeywell Natural Gas Detector (JTQJ-BF-01LM/BW) Zigbee Apple HomeKit

### Vendor

Xiaomi

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/JTQJ-BF-01LM-BW.jpg

### Database entry

{"id":22,"type":"EndDevice","ieeeAddr":"0x00158d0006d5275e","nwkAddr":57014,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.sensor_natgas","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,4,3,1,2,10,1280],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":45,"5":51,"8":8456,"9":4097,"100":0,"150":0},"modelId":"lumi.sensor_natgas","appVersion":1}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":33,"dateCode":"10-19-2016","zclVersion":0,"interviewCompleted":true,"meta":{},"lastSeen":1691844469850,"defaultSendRequestWhen":"immediate"}

### Notes

This device type is Router. It hasn't a battarey and has "powerSource":"Mains (single phase)" not "powerSource":"Battery"
All other parameters are transmitted by the converter correctly.

[image]

Koenkk, 2023-08-13:
Fixed!

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

closed 2023-08-13 by Koenkk

vit-um, 2023-08-13:
Hi, @Koenkk! 
But why you add this code? 
```
        configure: async (device, coordinatorEndpoint, logger) => {
            device.powerSource = 'Battery';
            device.type = 'EndDevice';
            device.save();
        },
```
This device type is **Router**. 
Perhaps the following code should be applied here:
```
        configure: async (device, coordinatorEndpoint, logger) => {
            device.powerSource = 'Mains (single phase)';
            device.type = 'Router';
            device.save();
        },
```

Koenkk, 2023-08-14:
Isn't it battery powered? I found this product picture:

[image]

vit-um, 2023-08-14:
All these devices are sold without battery in my country:
https://da1.com.ua/uk/p/1335884989-datchik-utechki-prirodnogo-gaza-xiaomi-honeywell-gas-detector-jtqj-bf-01lm-bw/
https://skirda.net/datchik-gaza-xiaomi-honeywell-gas-detector-jtqj-bf-01lmbw-1548420283
https://prom.ua/p789070083-datchik-utechki-gaza.html
In your picture, the smoke detector (JTYJ-GD-03MI/BB) I wrote to you about the CH4 gas detector (JTQJ-BF-01LM/BW)

[image]

[image]

[image]

VivantSenior, 2024-01-22:
Looks like this particular bug hasn't been fixed for [Xiaomi JTQJ-BF-01LM/BW](https://www.zigbee2mqtt.io/devices/JTQJ-BF-01LM_BW.html). It is powered by mains therefore it ought to be a router for a zigbee network.

[image]

vit-um, 2024-01-23:
Hello! I used this external converter, but I'm not sure if it will work on the current version of zigbee2mqtt

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const xiaomi = require('zigbee-herdsman-converters/lib/xiaomi');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{ modelID: 'lumi.sensor_natgas', manufacturerName: 'LUMI'}],
    model: 'JTQJ-BF-01LM/BW',
    vendor: 'Xiaomi',
    description: 'MiJia gas leak detector',
	fromZigbee: [fz.ias_gas_alarm_1, fz.JTQJBF01LMBW_sensitivity, fz.JTQJBF01LMBW_gas_density],
	toZigbee: [tz.JTQJBF01LMBW_JTYJGD01LMBW_sensitivity, tz.JTQJBF01LMBW_JTYJGD01LMBW_selfest],
	exposes: [
		e.gas(), e.tamper(), 
		e.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('Set sensitivity'),
		e.numeric('gas_density', ea.STATE).withDescription('Gas density'), 
		e.enum('selftest', ea.SET, ['SelfTest']).withDescription('Selftest'),
	],
	configure: async (device, coordinatorEndpoint, logger) => {
		device.powerSource = 'Mains (single phase)';
		device.type = 'Router';
		device.save();
	},
};

module.exports = definition;
```

VivantSenior, 2024-01-24:
@vit-um thank you. Now my gas sensor is recognized as router. But still it is a workaround. Could you please reopen this ticket?

VivantSenior, 2024-02-03:
> Hello! I used this external converter, but I'm not sure if it will work on the current version of zigbee2mqtt

@vit-um, because of the name change in the converter file in PR https://github.com/Koenkk/zigbee-herdsman-converters/pull/6969 (`xiaomi.ts` --> `lumi.ts`) a fix is required for the external converter:

```js
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const xiaomi = require('zigbee-herdsman-converters/lib/lumi');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [{ modelID: 'lumi.sensor_natgas', manufacturerName: 'LUMI'}],
    model: 'JTQJ-BF-01LM/BW',
    vendor: 'Xiaomi',
    description: 'MiJia gas leak detector',
	fromZigbee: [fz.ias_gas_alarm_1, fz.JTQJBF01LMBW_sensitivity, fz.JTQJBF01LMBW_gas_density],
	toZigbee: [tz.JTQJBF01LMBW_JTYJGD01LMBW_sensitivity, tz.JTQJBF01LMBW_JTYJGD01LMBW_selfest],
	exposes: [
		e.gas(),
		e.tamper(), 
		e.enum('sensitivity', ea.STATE_SET, ['low', 'medium', 'high']).withDescription('Set sensitivity'),
		e.numeric('gas_density', ea.STATE).withDescription('Gas density'), 
		e.enum('selftest', ea.SET, ['SelfTest']).withDescription('Selftest'),
	],
	configure: async (device, coordinatorEndpoint, logger) => {
		device.powerSource = 'Mains (single phase)';
		device.type = 'Router';
		device.save();
	},
};

module.exports = definition;
```