Koenkk/zigbee2mqtt #18605: Group state incorrect when member is unavailable

andreasbrett, 2023-08-13:
### What happened?

I have a group that contains multiple lights incl. 1 light that is currently unavailable. The group state is always "ON" since the last state of the unavailable light was "ON".

Checking how group state is determined, I see that unavailable and disabled devices are not filtered out of the equation but rather their cached state is used to determine the group state. This IMHO is incorrect as disabled and unavailable devices should not be taken into account.

https://github.com/Koenkk/zigbee2mqtt/blob/1ec1e57244834cc6b9dea6dbd3612b01cd21fa65/lib/extension/groups.ts#L185

### What did you expect to happen?

I expected for either...

a) unavailable/disabled devices to lose their cached state
b) unavailable/disabled devices to not be considered when determining the group state

### How to reproduce it (minimal and precise)

- create a group with multiple lights
- turn on all lights => group state is "ON"
- pull power from one light => group state is still "ON"
- turn off all lights => group state is still "ON" but should be off

### Zigbee2MQTT version

1.32.2

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus ZBDongle-P

andreasbrett, 2023-09-13:
Still active

----------------------------------------

skytern, 2023-09-25:
Hello, I think then your issue is linked to this topic :
https://github.com/Koenkk/zigbee2mqtt/issues/10086#issuecomment-989003017

I had a similar issue and finally I remove my "group" in configuration.yaml.

andreasbrett, 2023-09-26:
> Hello, thank you for your post, It helped me to understand. I think then your issue is linked to this topic : [#10086 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/10086#issuecomment-989003017)
> 
> I had a similar issue and finally I remove my "group" in configuration.yaml.
> 
> Zigbee2MQTT version: 1.33.0

Thanks, but this is not the issue. It's really about the code logic. If one member of a group is unavailable its cached state is used to determine the group state. I think this is incorrect as unavailable group members should not be considered when evaluating group state.

closed 2023-11-04 by github-actions[bot]